2dom/PxMatrix #91: Double buffering - difference between buffers brightness

maciejmatczak, 2019-04-19:
Hi,

I am playing around with your, really nice hardware back engineering!

I modified few functions to achieve this model of work with two buffers from current master:
- display only active buffer (remove buffer switch from display)
- draw only on "other" buffer (drawPixel to use _active_buffer, not hard coded 0)
- explicitly swap buffers once done with drawing (swapBuffer to actually swap out active and selected)

(BTW, those changes, is it something you would accept as PR?)

In the end, works really nice despite one thing. I am drawing the **same** frame in both buffers and on switching those I can see difference in brightness (and it's visible, so more like some higher bit then lower). I guess it would be something fillMatrixBuffer, but I am literally getting lost there :D 

Would you be able to check if there is some obvious for you miscalculation applied to this function and double buffering? I will add exact examples if you can't find anything like that.

Thanks!

2dom, 2019-04-19:
I'm happy to accept PR...maybe you could share some code an some pictures what you did?

maciejmatczak, 2019-04-20:
Allrighty. I am currently working in PlatformIO environment in VS Code, but examples should be almost copy pastable to Arduino.

I am using auto formatting, so I am including current master, but reformatted. Please check my main.cpp (it's your modified example) and diff two PxMatrix.h, there are 4 places I touched. 

[code.zip](https://github.com/2dom/PxMatrix/files/3099898/code.zip)

This is how it looks like:
https://photos.app.goo.gl/rBAaY1vTSBgCf9oK7
With 1s interval it gets a little brighter and after next second a little darker, but I am drawing literally the same things on each buffer after clearing it. It's more obvious in real though. Mostly visible how glow changes on dark mat.

maciejmatczak, 2019-04-20:
Line 615, from:
```C++
total_offset_r -= buffer_size * selected_buffer;
```
to:
```C++
total_offset_r += buffer_size * selected_buffer;
```
?

2dom, 2019-05-09:
Did you solve this? If so a PR would be great :)

maciejmatczak, 2019-05-12:
Hey @2dom,

Sure, I think I can propose some solution. I already forked PxMatrix and played around to get my C/C++ knowledge a little "derust". I need to clean it up and move somewhere and fork your original one, one more time.

If you could add some standard for formatting the code (some generic .clang-format) and reformatted your code it could be easier to work with, it's cheap to have your code formatted automatically :) and if you would define a standard in your project, everyone else should honor this. Also, some contribution guidelines, maybe written down ideas (needPR tag for unsolved issues with those)...

About the issue: I would like to introduce more obvious dual buffering, something like this:
https://github.com/maciejmatczak/PxMatrix/blob/147d0ed9433403f66bc9a664877a48e641e88f2d/PxMatrix.h#L175-L181
This would allow us to drop some math with dual buffering (just use index) + some changes around "swapBuffers" and in constructor. Can't say when I will have a moment to work on it, but eventually I would really like to push something "upstream".

Rasmusfk, 2019-06-01:
I just posted a pull request (#103) that solves this issue.
Regards Rasmus

maciejmatczak, 2019-06-01:
Hi @Rasmusfk 
I did it as well :) #96

2dom, 2019-06-01:
Thanks to both of you...Sorry that I am so slow with merge requests....will try to review soon !

Rasmusfk, 2019-06-01:
Hi @maciejmatczak & @2dom 
I should've read through this thread before fixing it, I now see that @maciejmatczak already had done some of the same. My commit does however make it easier to use double-buffering, and all the Adafruit draw routines now works with double buffering.
Regards

maciejmatczak, 2019-06-01:
@Rasmusfk, did you check out the differences about our approaches? I would rather say we did the same and the usage pattern is identical - draw on one buffer while another one is being shown, then swap.

I did left the functions that can reach active/selected buffer if needed though. But it's not needed to use it in normal situation. 

About draw routines - I believe I am missing something, so which one would not work on #96 and what fixes this issues?

@2dom Sure, thanks. Pick one whichever fits best :)

Rasmusfk, 2019-06-01:
The Adafruit drawing routines all use the drawPixel(int16_t x, int16_t y, uint16_t color) method, so it needs to be double-buffering-aware.

maciejmatczak, 2019-06-01:
Good eye, thanks for noticing that one!

2dom, 2019-06-01:
Ok thanks again to the both of you! I merged @Rasmusfk version as they are very similar and it comes with an example on how to use it. If you guys could post a video of the latest master with double buffering in action I would add this to the Readme.

Rasmusfk, 2019-06-01:
Here you go:
https://youtu.be/cWYY4nl-Ecg
Regards

maciejmatczak, 2019-06-03:
@Rasmusfk 
Any chance for _with_ vs _without_ double buffering?

Rasmusfk, 2019-06-04:
@maciejmatczak 
Yes, but I'm a little busy the next couple of days, but I'll get to it in the weekend.

2dom, 2019-08-20:
seems to work fine now

closed 2019-08-20 by 2dom