2dom/PxMatrix #92: Add in readme information about how actually displaying works

maciejmatczak, 2019-04-20:
Hi

I was wrapping my mind around how actually display works. Can you please take a look at following charts.

[image]
By example in this library, let assume we have this 2ms _display_ticker_, every time it hits we displays 1/_color_depth_ part of the whole frame, so with default 8 we have 8 parts for full frame. 16ms for one frame, pretty much 60fps.

Now, zooming in:
[image]
Lib does some internal part and sends data (red part, should be actually before latching) and latches the shift registers and enable output for _show_time_ via hard _nop_ loop.

I am not sure if I am right in this one. But that's why by making it longer we are achieving brighter display, aren't we?

I was also thinking about dropping _hard-nop-loop_ and having output enable for whole period. No luck though... I am getting weeeird results. Is it something that makes sense for full brightness? 

If charts are fine and my logic as well, I would like to write this down and make normal charts for PR. What do you think?

maciejmatczak, 2019-04-20:
I omitted _row_pattern_. I have 64x32 display with 1/16 scan. So "one big shift register" at once operates on... 128 LEDs? The test pattern sketch just shuffles in data, but doesn't switch the mux?

So for 1/16 scan, there should be actually 16 periods of 10us within display_ticker?

2dom, 2019-05-09:
Well ... in principle the display is scanned once for every time you call display.display(). You correct on the show_time, this time determines how long we keep the leds on. Keeping the display on while clocking in data is already implemented - see setFastUpdate.

closed 2019-05-09 by 2dom