2dom/PxMatrix #93: Strange behaviour of the LED display, image strongly distorted

TyskieGronie, 2019-04-24:
Hello,

I have been trying for several days to get a LED display from Lysonled (identically constructed Skryled) with FM6126A chipset to work.

I get a picture displayed but the result is strongly falsified.
I already tried a different controller, power supply and cables. Unfortunately without success.

I checked the wiring several times and can't find any error.

**Mario:**
[http://sendvid.com/770migz5](url)
_Mama Mia ..._

**With controller switched off:**
http://sendvid.com/are4e79v

**Pixel Time:**
http://sendvid.com/jyg73ury

**Connecting GND from HUB75 (PIN bottom right):**
http://sendvid.com/dfs2kvck

I would be very happy about your support and advice on what else it could be.

TyskieGronie, 2019-04-27:
I've been experimenting. My cabling is basically the same as shown here but with a NodeMCU.

[image]

I connected the NodeMCU to the 5V power supply of the LED matrix.
As soon as I connect the GND pin as shown in the sketch (black wire), the signal breaks off.

If I power the NodeMCU via USB the result is as shown in the video.
As soon as I ground the LED panel against the NodeMCU, the signal breaks off or is distorted as seen in the last video.

2dom, 2019-05-09:
Hard to say whats wrong here - this funny effect is usually due to improper initialization of the chip. Try turning everything off, then power the LED panel, then the MCU.

TyskieGronie, 2019-05-14:
First of all I want to thank you for your great work and spare time you put into this library.

Unfortunately, I couldn't solve the problems and eventually switched to my Raspberry and another library.

For the Raspberry there is the Adafruit RGB Matrix Bonnet, with which it worked almost immediately. Probably this was the better choice for me, a amateur in electrical engineering.

It took me many frustrating hours to locate and solve the problems with the FM6126A chip. I finally failed with the wiring.

The result is still impressive with the Raspberry and Adafruit Bonnet:

[image]

2dom, 2019-05-14:
No worries man...I'm glad you got it working and a bit mad that it was not with PxMatrix. 

The FM6126A chip is just terrible and unfortunately I don't have the time to create proper support. 

Cheers !

closed 2019-08-20 by 2dom