2dom/PxMatrix #95: Pettern_Test works, but display.print("Pixel") is not working

JLoTech, 2019-05-08:
Hello and thanks for the library.

The pattern_test works fine on my 192x192mm 32x32px P6 with 1/16.
First red lines, then yellow and white.

But when I change the void display_updater() from 
   display.displayTestPattern(70);
to
   display.display(70)

the Matrix is black and the serial console from the Arduino IDE will show this very long message:

hello

Exception (9):
epc1=0x4020355a epc2=0x00000000 epc3=0x00000000 excvaddr=0x3ffee7fb depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffed00 end: 3fffffb0 offset: 01a0
3fffeea0:  00000000 3ffee7fb 00000018 402035c6  
3fffeeb0:  ffffffff ffffffff ffffffff 3fff18e0  
3fffeec0:  3ffee7e0 00000000 3fff17e0 40202b81  
3fffeed0:  00000046 ffffffff ffffffff ffffffff  
3fffeee0:  ffffffff ffffffff ffffffff 60000600  
3fffeef0:  005946b1 3ffee1c0 3fff22a4 40202c0f  
3fffef00:  40105096 3ffee198 000019af 40203f52  
3fffef10:  4022cf70 3fff22a4 3ffee1c0 40204246  
3fffef20:  60000600 3fff22a4 ffffffff 4022cf7d  
3fffef30:  4022cfc2 3fffdab0 00000000 3fffdcb0  
3fffef40:  3ffee1d0 3fffdad0 3fff1a4c 40203c6f  
3fffef50:  40000f49 40000f49 3fffdab0 40000f49  
3fffef60:  40000e19 40001878 00000002 00000000  
3fffef70:  3fffff10 aa55aa55 00000010 4010494c  
3fffef80:  40104952 00000002 00000000 a9e9ddca  
3fffef90:  4010000d 37658332 1326e787 2f4fe634  
3fffefa0:  40100b14 3fffef3c 40100ac1 3fffff18  
3fffefb0:  3fffffc0 00000000 00000000 feefeffe  
3fffefc0:  feefeffe feefeffe feefeffe feefeffe  
3fffefd0:  feefeffe feefeffe feefeffe feefeffe  
...
3ffff850:  feefeffe feefeffe feefeffe feefeffe  
3ffff860:  feefeffe feefeffe feefeffe feefeffe  
3ffff870:  feefeffe feefeffe feefeffe feefeffe  
3ffff880:  feefeffe feefeffe feefeffe feefeffe  
3ffff890:  feefeffe feefeffe feefeffe feefeffe  
3ffff8a0:  feefeffe feefeffe feefeffe feefefv⸮⸮⸮⸮

Could you please give me some help?

I already tried different WeMos D1 boards.

Thank you very much.

2dom, 2019-05-08:
Try display.display(10), run at 160MHz and set display.setFastUpdate(true)

JLoTech, 2019-05-08:
I just tried, but same problem.

Exception (9):
epc1=0x4020355a epc2=0x00000000 epc3=0x00000000 excvaddr=0x3ffee7e7 depc=0x00000000

and no pixel on. I tried three different ESP8266 boards with newest Arduino IDE (1.8.9)

Board Manager Link "https://arduino.esp8266.com/stable/package_esp8266com_index.json"

Settings:
Board WeMos D1 R1
Upload Speed 115200
CPU Frequency 80 or 160MHz
Flash Size 4M no SPIFFS
Debug port disabled
debug level none
IwIP variant v2 Lower Memory
VTables Flash
Exceptions disabled
erase flash only sketch
port COM6

2dom, 2019-05-09:
Can you post your code?

JLoTech, 2019-05-09:
Here is the code:

#include <PxMatrix.h>
#include <Ticker.h>
Ticker display_ticker;
#define P_LAT 16
#define P_A 5
#define P_B 4
#define P_C 15
#define P_D 12
#define P_E 0
#define P_OE 2

// Pins for LED MATRIX
//PxMATRIX display(32,16,P_LAT, P_OE,P_A,P_B,P_C);
PxMATRIX display(32,32,P_LAT, P_OE,P_A,P_B,P_C,P_D);
//PxMATRIX display(64,64,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E);

// ISR for display refresh
void display_updater()
{
  // display.displayTestPattern(70);
  // display.displayTestPixel(70);
  display.display(70);
}

uint16_t myCYAN = display.color565(0, 255, 255);
void setup() {
  // put your setup code here, to run once:
Serial.begin(9600);
  display.begin(16);
  display.flushDisplay();
  display.setTextColor(myCYAN);
  display.setCursor(2,0);
  display.print("Pixel");
  Serial.println("hello");
  display_ticker.attach(0.002, display_updater);
  delay(1000);
}

void loop() {
 delay(100);
}

2dom, 2019-05-09:
Try removing the delay from the loop. If that does not work increase ticker time, e.g. display_ticker.attach(0.004, display_updater);

JLoTech, 2019-05-14:
Same problem.

I think it is not a timing problem of the display, more a memory problem inside ESP?

Please check this post:
https://github.com/esp8266/Arduino/issues/869

But I am not so familar in programming. So could you please help me again?