2dom/PxMatrix #98: esp32 flicker with ticker

bribur, 2019-05-16:
This relates to #56 and #87 closed issues.  There is a workaround to the Guru Meditation error that almost works, that is to use the Ticker display timer intended for esp8266 instead of the esp32 interrupt approach in your examples.  I have wifi, BLE and SPIFFS all working that way. The problem is the display flickers, so I am wondering if there is some simpler bandaid to make Ticker/Pxmatrix and esp32 play together. I am able to turn off WIFI and only connect when needed to update a clock but BLE does not turn off/on cleanly yet. The display is stable when they are disconnected though. Any thoughts?

Wferr, 2019-05-23:
Im having this same issue, anyone got any ideas?

2dom, 2019-05-25:
You may want to try another library that uses DMA, e.g. SmartMatrix

closed 2019-08-20 by 2dom