2dom/PxMatrix #99: FM6126A SetBrightness

bastienvans, 2019-05-21:
As you probably noticed, the setBrightness function doesn’t work for the FM6126A chips.

I’ve been trying to mess with the source code for a week now but I am not able to get anything working as my knowledge in the field is limited.

Anyone with some spare time willing to take a look into it?

Thank you!

2dom, 2019-06-02:
I won't spend more time on FM6126A related issues as the manufacturers have stopped using them (FM6126A suck big time).

zakrava, 2019-09-23:
I just bought new p3 panel, and to my surprise, it has FM6126A chip. It looks like, that this is is still widely used. Please can you consider add brightness support for this chip? I am using esp8266.

2dom, 2019-10-19:
Brightness support only for esp8266 for now sorry. Don't have the time to port the ESP32 code to esp8266. Do you want to give it a shot ?

closed 2019-10-20 by 2dom

IAmOrion, 2019-11-24:
I'm hoping someone will eventually pick this up (I'm not skilled enough myself), however these damn FM6126A chips, whilst they're apparently not in production any more, the UK at least is FLOODED with these matrix panels! I've bought about 5 this year, all from different companies, different times, differing prices etc.... 4 out of 5 are FM6126A :(