2dom/PxMatrix #100: Cannot drive more than 2 displays

GilDev, 2019-05-22:
Hi,
I would like to drive up to 8 P3 64x32 displays with my ESP32 and PxMatrix v1.3.0. It works fine up to two displays, but when I set the display width to 192 for 3 panels, the watchdog triggers and I’m not able to display anything. When setting the width to 128 (along with `display.setPanelsWidth(3)`), it works fine but the display is broken and things are not in order.
Changing the display draw time from 70 to 10 doesn’t change a thing.
Any idea on why it desn’t work? Thanks.

2dom, 2019-05-25:
Yes ... you will have to increase the buffer size. You can specify the frame buffer size by putting

```
#define PxMATRIX_MAX_HEIGHT 32
#define PxMATRIX_MAX_WIDTH 512

```
before the PxMatrix include. I am not sure if PxMatrix can handle displays of this size. You may want to use another library, e.g. SmartMatrix.

GilDev, 2019-05-27:
Thanks. I hadn’t searched enough in the library itself for that macro.
Now, I made a small test program that makes a white pixel bounce around. It works great with one display but colors are funky with more as you can see [in this video](https://youtu.be/4cX3_Ziyklc). Still trying to figure out where it comes from?
I’ll look into other librairies as well, there is apparently [a port of SmartMatrix for ESP32](https://community.pixelmatix.com/t/smartmatrix-library-esp32-port/272), maybe that will help me to drive the 128x128 display I need to make (gotta make sure it works before making the PCB).

closed 2019-08-20 by 2dom

jaroslavas, 2020-01-05:
@GilDev Were you able to fix those strange colors?

GilDev, 2020-01-05:
@jaroslavas No sorry, I've switched to a Raspberry Pi to drive my 128x128 matrix.

adrianrv, 2020-04-29:
I solved the issue by modifying the value of `PxMATRIX_SPI_FREQUENCY`, you can do with

```
#define PxMATRIX_SPI_FREQUENCY 13000000
#include <PxMatrix.h>
```
I did not check still some other values.

The solution was proposed in this issue: https://github.com/2dom/PxMatrix/issues/94