2dom/PxMatrix #101: Hardware issue, wiring issue, or bug?

bseishen, 2019-05-30:
64x64 panel with 5 bit addressing. 1/32 scan rate.
Seems to be an off by one error on Y axis. Using setPixel method to make sure its not an issue with the Adafruit GFX library. Addressing Y = 0 lights up row 1 not row 0. Y=31 lights up row 0. Bottom half of the display does the same thing.

https://youtu.be/Fxj09cAWFhQ

bseishen, 2019-05-31:
Decremented the passed in value by 1 in the set_mux function. No idea why that works though.

bseishen, 2019-06-06:
Plot thickens, So at full brightness this is an issue, not an issue when brightness is less than 254.

2dom, 2019-08-20:
no idea ... any updates on this?

bseishen, 2019-08-20:
setBrightness needed to be called for this to resolve correctly. If brightness is at 255, this issue presents itself.

2dom, 2019-08-20:
Madness ... will keep this open as a reminder to see what's going on here. Thanks!

2dom, 2019-11-06:
Is this still an issue with the latest master?

bseishen, 2020-03-25:
yes it is

2dom, 2020-03-28:
Sorry man ... defeats me

xsrf, 2020-04-13:
@bseishen are u using fast update?
@2dom in fast_update you set latch low, enable the output, then latch high. 
```
        digitalWrite(_LATCH_PIN,HIGH);
        digitalWrite(_OE_PIN,0);
        start_time = micros();

        digitalWrite(_LATCH_PIN,LOW);
```
https://github.com/2dom/PxMatrix/blob/master/PxMatrix.h#L1063

In regular update using latch(), you latch low, latch high, then enable the output.
```
    digitalWrite(_LATCH_PIN,HIGH);
    //delayMicroseconds(10);
    digitalWrite(_LATCH_PIN,LOW);
```
https://github.com/2dom/PxMatrix/blob/master/PxMatrix.h#L990

Assuming the shift register actually latches on a falling edge and prevents a low signal on the latch if output is enabled (to prevent changing any outputs while they are enabled), they will latch the output after the data has been transferred in fast_update, which would result in an off-by-one error.

I think latching low after the output was enabled is generally a bad idea and might cause ghosing in other cases.

@bseishen
Try this instead
```
        digitalWrite(_LATCH_PIN,HIGH);
        digitalWrite(_LATCH_PIN,LOW);
        digitalWrite(_OE_PIN,LOW);
        start_time = micros();
```
https://github.com/2dom/PxMatrix/blob/master/PxMatrix.h#L1063

mylab4492, 2022-12-27:
@2dom @xsrf 
Hi guys I'm using exact same panel and panel where _E_PIN is connected to pin 10 or pin A0 of esp8266(tried both). Panel works till brightness is below 77 in case of pin 10 and it goes up to 80 in the case of A0. when I set brightness more than that then panel stuck and esp does not crash but nothing else happens. and panel keep stuck on displaying the value. any idea how this can be fixed.