2dom/PxMatrix #104: New chipset on the market?

chhofi, 2019-06-02:
Hello,

the last days I try to get my 64x32 panel to work. Unfortunately without great success. When looking closer at the board I realized, that it is not using any of the known chips. The are different chips on the board: DP5020B, RUC7258 and 74HC245C.
The supplier said it is using a normal HUB75 interface. Also Pin E is grounded. So probably not used.

Should the library work with those chips too? 
Has anyone got any experience with that specific LED board?

Link to the board on Aliexpress:
https://www.aliexpress.com/item/32902068286.html?spm=a2g0s.9042311.0.0.5cd94c4dzOgXcG

Cheers,
Christian

[image]
[image]
[image]
[image]

chhofi, 2019-06-02:
Here a picture of the "Pixel Time"
[image]
 test

2dom, 2019-06-02:
Could you run the pattern test and post the result?

chhofi, 2019-06-02:
Sure. I also ran the "Pixel Time" sketch again.

https://youtu.be/tBEZxhWevRs
https://youtu.be/eb2GSpwCbzU

In real it looks the same as in the videos.

2dom, 2019-06-09:
Strange ... Read the data sheets...seems ok. Anyone else having this panel?

HoLo85, 2019-06-23:
Hi, not the same panel, but a 64x64 with the same chipset and the same problem.
The E line on my panel is used and if i disconnect the E line the missing part of the text is sometimes visible but constantly flickering, so maybe its a timing issue?

I can use the panel with hzellers library for the raspberry, but sadly not with the esp8266 or esp32.

serestre, 2019-08-30:
i have same problem
[image]

2dom, 2019-10-19:
This is solved in the latest master ... the multiplxer chips on some panels are too slow. You can now configure the mux delay as follows:

```
  display.begin();
  display.setMuxDelay(0,1,0,0,0);
```

This would add some delay to the B channel.

2dom, 2019-10-19:
Just experiment a bit, e.g. start with 1us on all channels an then reduce until problems occur

closed 2019-10-19 by 2dom

Porokoura, 2019-12-18:
> Hello,
>
> the last days I try to get my 64x32 panel to work. Unfortunately without great success. When looking closer at the board I realized, that it is not using any of the known chips. The are [...]

Hello! Have you solved the problem?

chhofi, 2020-01-07:
Now I had some time for testing. It is working great! Thanks a lot for the new feature!!!

Porokoura, 2020-01-08:
Did you add mux delay to each channel or just to one individual?
What else did you do?

Porokoura, 2020-02-08:
> 
> 
> Now I had some time for testing. It is working great! Thanks a lot for the new feature!!!

I would be interrested to hear what was your solution for a problem.

chhofi, 2020-02-10:
-> Just experiment a bit, e.g. start with 1us on all channels an then reduce until problems occur
I tried some different values. But now I switched to the RaspberryPi, because for my project I needed more computing power.