2dom/PxMatrix #107: I found a trick by adding the line :

tammaaziz, 2019-06-09:
_Originally posted by @pyzimmer in https://github.com/2dom/PxMatrix/issues/94#issuecomment-491263764_

Hello!
I have a P10 rgb full color 32x16 display : I tried all available options (display(A, B, {C}, {D}), setScanPattern, setMuxPattern) on the subject using the PxMatrix library with a NodeMCU (ESP12).

beside, I tried the display.setPanelsWidth(4) as suggested by pyzimmer but does not work .. I am really stuck !! Can anyone help please !

2dom, 2019-06-09:
Hi...could you post a picture / gif of what's going on?

tammaaziz, 2019-06-09:
Hi..
Thank you 2dom..

I just uploaded a youtube video of the display result of the "pattern test" sketch : I am using a NodeMCU ESP 12N.
[https://www.youtube.com/watch?v=dQBQuWF_Xp4]

Just to mention that I have already tried using Wemos D1 (with same PxMatrix Library) but also RGB Matrix library using Arduino Uno : all the results are not ok.

tammaaziz, 2019-06-11:
hello Dominic,

I hope you are doing well,

Did you get a chance to check my response ?
if so, is there any guidance, anything I can check/do please ?

Thank you

2dom, 2019-06-12:
did you try display.setMuxPattern(STRAIGHT) ?

tammaaziz, 2019-06-13:
Hi,

The same result : I have also tried the different scan patterns + ratios : (almost) same result.

really stuck !!!
Any idea please.

tammaaziz, 2019-06-13:
Hi,

I tried the following and the result is now different..but still cannot display numbers or words..
I did like this :
- remove the B, C and D pins from Panel
- set scan rate to 2 .. as only A pin is connected which gives 2 states.
- use displayTestPixel instead of displayTestPattern

the result is in the link [https://youtu.be/Z3agz-ws9vw](url)

Here is the complete sketch used 

#include <PxMatrix.h>

#ifdef ESP32

#define P_LAT 22
#define P_A 19
#define P_B 23
#define P_C 18
#define P_D 5
#define P_E 15
#define P_OE 2
hw_timer_t * timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

#endif

#ifdef ESP8266

#include <Ticker.h>
Ticker display_ticker;
#define P_LAT 16
#define P_A 5
#define P_B 4
#define P_C 15
#define P_D 12
#define P_E 0
#define P_OE 2

#endif
// Pins for LED MATRIX

PxMATRIX display(32,16,P_LAT, P_OE,P_A,P_B);

#ifdef ESP8266
// ISR for display refresh
void display_updater()
{
//  display.displayTestPattern(70);
   display.displayTestPixel(70);
//  display.display(70);
}
#endif

uint16_t myCYAN = display.color565(0, 255, 255);
void setup() {
  // put your setup code here, to run once:
Serial.begin(9600);
  display.begin(2);

  #ifdef ESP8266
    display_ticker.attach(0.002, display_updater);
  #endif

  delay(1000);
}

void loop() {
 delay(100);
}

2dom, 2019-06-15:
This display appears to have some new strange scanning pattern. You could try setScanPatter(VZAG) but I don't think this will do the trick. This panel is most likely unsupported by PxMatrix (sorry)

tammaaziz, 2019-06-16:
Hi,

I have already tried that one too, no success !!!
What about giving me hints to adapt this new pattern to the library  :do
you think this is possible ?

thanks

2dom, 2019-08-24:
Apart from reading the Readme and the code not much unfortunately - it's a bit of a mind bender.

jakaisna, 2020-05-27:
its work use p10 outdor

// This is how many color levels the display shows - the more the slower the update
//#define PxMATRIX_COLOR_DEPTH 8

// Defines the buffer height / the maximum height of the matrix
//#define PxMATRIX_MAX_HEIGHT 64

// Defines the buffer width / the maximum width of the matrix
//#define PxMATRIX_MAX_WIDTH 64

// Defines how long we display things by default
//#define PxMATRIX_DEFAULT_SHOWTIME 30

// Defines the speed of the SPI bus (reducing this may help if you experience noisy images)
//#define PxMATRIX_SPI_FREQUENCY 20000000

// Creates a second buffer for backround drawing (doubles the required RAM)
//#define PxMATRIX_double_buffer true

#include <PxMatrix.h>

// Pins for LED MATRIX
#ifdef ESP32

#define P_LAT 22
#define P_A 19
#define P_B 23
#define P_C 18
#define P_D 5
#define P_E 15
#define P_OE 2
hw_timer_t * timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

#endif

#ifdef ESP8266

#include <Ticker.h>
Ticker display_ticker;
#define P_LAT 16
#define P_A 5
#define P_B 4
#define P_C 15
#define P_D 12
#define P_E 0
#define P_OE 2

#endif

#define matrix_width 32
#define matrix_height 16

// This defines the 'on' time of the display is us. The larger this number,
// the brighter the display. If too large the ESP will crash
uint8_t display_draw_time=60; //30-70 is usually fine

//PxMATRIX display(32,16,P_LAT, P_OE,P_A,P_B,P_C);
PxMATRIX display(32,16,P_LAT, P_OE,P_A,P_B,P_C,P_D);
//PxMATRIX display(64,64,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E);

// Some standard colors
uint16_t myRED = display.color565(255, 0, 0);
uint16_t myGREEN = display.color565(0, 255, 0);
uint16_t myBLUE = display.color565(0, 0, 255);
uint16_t myWHITE = display.color565(255, 255, 255);
uint16_t myYELLOW = display.color565(255, 255, 0);
uint16_t myCYAN = display.color565(0, 255, 255);
uint16_t myMAGENTA = display.color565(255, 0, 255);
[... 170 lines cut ...]

  draw_weather_icon(icon_index);
  icon_index++;
  if (icon_index>10)
    icon_index=0;

  for (int xx=0; xx<16;xx++)
  {
    display.drawLine(xx+16,0,xx+16,5,display.color565(xx*16,0,0));
    display.drawLine(xx+16,6,xx+16,10,display.color565(0,xx*16,0));
    display.drawLine(xx+16,11,xx+16,15,display.color565(0,0,xx*16));
  }
  delay(1000);
  for (uint8_t dimm=255; dimm>0; dimm--)
  {
    display.setBrightness(dimm);
    delay(5);
  }
  for (uint8_t dimm=0; dimm<255; dimm++)
  {
    display.setBrightness(dimm);
    delay(5);
  }

}