2dom/PxMatrix #108: P10 32x16 incorrect row display

kaung29, 2019-06-15:
[image]
[image]

my very first time led project. I use Nodemcu esp8266 board.
I attach the photo .. test with pattern_test

2dom, 2019-06-15:
Could you post a video of gif?

AeroX2, 2019-08-05:
This P10 matrix looks very similar to mine and I think I'm having the same issue.
I've tried different scanning and muxing patterns but it always results in the same behaviour.

Here is a video of the effect: https://youtu.be/jZTgdUTgVeE

AeroX2, 2019-08-07:
Had a look into this myself and managed to get my display working but I had to make changes to the PxMatrix.h. It took a while to troubleshoot and probably would have been nice to have known that displayTestPattern doesn't use the scanning patterns.

But in the end, I used the ZIGZAG pattern and swapped lines 563 and 565 in PxMatrix.h:
`total_offset_r=base_offset-row_sector__offset*row_sector-(_scan_pattern==ZAGGIZ ? 1: 0);` 
`total_offset_r=base_offset-row_sector__offset*row_sector-(_scan_pattern==ZIGZAG ? 1: 0);`

Not sure if there was a set of parameters that I missed that would do this but this worked for me.

2dom, 2019-08-24:
Well yes ... `display.setScanPattern(ZAGGIZ)` would do the same. The whole point of the pattern test is to  just shifts in bytes to reveal the internal structure of the matrix. The pattern test would, however, not show reversed byte patterns. You could then use displayTestPixel.

closed 2019-08-24 by 2dom