2dom/PxMatrix #110: Latest version breaks my  2x  64x64 matrix setup

Ep101kur, 2019-06-19:
Thanks for your great work!  I had hours of fun with your library <3

**tl;dr :** 
My Esp32 + 2x 64x64 Matrix  setup crashes with the new SPI.setFrequency(10000000);. If I change it back to 20000000 everything runs smooth. 

When I tried to adjust my current project to the new "double_buffer" API I got this error:

> Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
> PC: 0x400d2641: spiWriteNL at  ..\Arduino15\packages\esp32\hardware\esp32\1.0.2\cores\esp32\esp32-hal-spi.c line 882
> EXCVADDR: 0x00000000
> 
> Decoding stack results
> 0x400d2641: spiWriteNL at  ..\Arduino15\packages\esp32\hardware\esp32\1.0.2\cores\esp32\esp32-hal-spi.c line 882
> 0x400d1d82: SPIClass::writeBytes(unsigned char*, unsigned int) at  ..\Arduino15\packages\esp32\hardware\esp32\1.0.2\libraries\SPI\src\SPI.cpp line 213
> 0x400d1120: PxMATRIX::display(unsigned short) at ..\\libraries\PxMatrix/PxMatrix.h line 870
> 0x40080e55: display_updater() at ..\\sketch_jun19a/sketch_jun19a.ino line 32
> 0x40081201: __timerISR at  ..\Arduino15\packages\esp32\hardware\esp32\1.0.2\cores\esp32\esp32-hal-timer.c line 88
> 0x40081209: __timerISR at  ..\Arduino15\packages\esp32\hardware\esp32\1.0.2\cores\esp32\esp32-hal-timer.c line 86
> 0x400d0fa6: PxMATRIX::drawPixel(short, short, unsigned short) at ..\\libraries\PxMatrix/PxMatrix.h line 624
> 0x400ea9d6: Adafruit_GFX::writePixel(short, short, unsigned short) at ..\\libraries\Adafruit_GFX_Library\Adafruit_GFX.cpp line 18

After fiddling around a view hours I found the source of the problem: is was the SPI.setFrequency(10000000) setting.  It should set to 20000000 like in the older releases.

With only one 64x64 matrix the 10000000 frequency runs fine. 

The new double_buffer example only runs with my setups (1 or 2 Matrices) if the draw_time is set lower then 50. If not you get the same generic error as above.

ty

2dom, 2019-06-20:
OK ... put back to 20MHz ... can you check out the latest master and see if it works. Also, could you post a gif / video with the 2 x 64 matrices (rather cool that this works :)

Ep101kur, 2019-06-24:
Latest release works fine.
You only have to set the not so well documented  constant "PxMATRIX_MAX_WIDTH" to 128 or you get random errors.

The Video is crafted from the double buffer example with the following changes:
1. #define PxMATRIX_COLOR_DEPTH 1
2. #define PxMATRIX_MAX_HEIGHT 64
3. #define PxMATRIX_MAX_WIDTH 128
4. PxMATRIX display(128,64,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E); 
4. display.display(10);
5. Delete delay(10); in the loop;

(With the default color depth of 8 (Bit) the frame rate drops to 7 fps)

[image]

closed 2019-06-26 by Ep101kur