2dom/PxMatrix #111: 128x64 HUB75E panel fails pixeltime example

rorosaurus, 2019-06-27:
Hello, I started tinkering with this library the other day with a 64x32 panel I had.  Thanks for the hard work! :)  I got a new panel today and tried the pattern_test and pixeltime examples.  pattern_test completed successfully, but I have some artifacts on pixeltime, using the same lines for setup.

````PxMATRIX display(128,64,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E);````
````display.begin(32);````

[image]

The panel is 128x64, HUB75E, 1/32.  This might be too large for this library, I'm realizing.  I don't see anything in serial monitor.  For the controller I'm using [TTGO-T8-ESP32](https://github.com/LilyGO/TTGO-T8-ESP32).

2dom, 2019-08-20:
You have to tell the library that you have a larger panel. set

```
#define PxMATRIX_MAX_HEIGHT 64
#define PxMATRIX_MAX_WIDTH 128
```

before you include PxMatrix

Leo1907, 2019-08-23:
Hello 2Dom,
After i set 
PxMATRIX display(128,64,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E);

i get the errors
Please help
ESP32, 2 P3 panel
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40087fb3  PS      : 0x00060130  A0      : 0x80086546  A1      : 0x3ffe3b40  
A2      : 0x3ffc32ac  A3      : 0x3ffba784  A4      : 0x00000001  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffe3b20  
A10     : 0x3ffba700  A11     : 0x3f40b130  A12     : 0x00000000  A13     : 0x3ffba700  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000017  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000008  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

2dom, 2019-08-24:
Well ... read my comment ...

helmarw, 2019-10-28:
was wondering if I should buy one of those. Did anyone get it to work ?
also aren't those displays 1/64 and not 1/32?!

closed 2020-03-28 by 2dom

dvmourik, 2021-04-11:
I have the same problem. Even when i define the parts that 2dom says before PxMatrix was included.
When i put 

PxMATRIX display(64,64,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E);

It works but ofcourse the display is not correct. (half screen)

when i go to 

PxMATRIX display(128,64,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E);

The exception comes up:

Backtrace: 0x4008ac84:0x3ffbfe20 0x400d0b7e:0x3ffbfe60 0x400d1510:0x3ffbfe80 0x400810b8:0x3ffbfec0 0x40081419:0x3ffbfee0 0x400873e5:0x3ffbff00 0x40081421:0x3ffb⸮Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Core 1 register dump:
PC      : 0x400d0bda  PS      : 0x00060b34  A0      : 0x800d1521  A1      : 0x3ffbfe60  
A2      : 0x3ffbdbd8  A3      : 0x3ffc3034  A4      : 0x0000001f  A5      : 0x00000040  
A6      : 0x00000010  A7      : 0x00000001  A8      : 0x00000008  A9      : 0x000000f7  
A10     : 0x00000010  A11     : 0x3ff65000  A12     : 0x3ff65000  A13     : 0x00ffffff  
A14     : 0xff000000  A15     : 0x0000001f  SAR     : 0x00000005  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400d10d5  LEND    : 0x400d1119  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x4008ad42  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400d0bda

ELF file SHA256: 0000000000000000

Backtrace: 0x400d0bda:0x3ffbfe60 0x400d151e:0x3ffbfe80 0x400810b8:0x3ffbfec0 0x40081419:0x3ffbfee0 0x400873e5:0x3ffbff00 0x40081421:0x3ffb1da0 0x400d11f2:0x3ffb⸮

Which is a translation for:

PC: 0x400d0bda: spiWriteNL at \AppData\Local\Temp\arduino_build_402428\sketch\esp32-hal-spi.c line 871
EXCVADDR: 0x00000000

Decoding stack results
0x4008ac84: vTaskEnterCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 4188
0x400d0b7e: spiSimpleTransaction at \AppData\Local\Temp\arduino_build_402428\sketch\esp32-hal-spi.c line 754
0x400d1510: PxMATRIX::display(unsigned short) at \Documents\Arduino\libraries\PxMatrix_LED_MATRIX_library/PxMatrix.h line 1116
0x400810b8: display_updater() at Beta1.0.0.ino line 117
0x40081419: __timerISR at Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.6\cores\esp32\esp32-hal-timer.c line 88
0x40081421: __timerISR at Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.6\cores\esp32\esp32-hal-timer.c line 86

Did anyone ever got a 128x64 (FM6124 chipset) working with a ESP32 (i use a Lolin32)

2dom, 2021-04-11:
Can you try reducing the update rate, e.g. using timerAlarmWrite(timer, 8000, true);

dvmourik, 2021-04-11:
Allready tried to reduce further but nothing helps. Looks like it's the display.display routine in ISR that's causing the problem on the ESP32. 

I am now trying the solution from
[https://github.com/2dom/PxMatrix/issues/225](https://github.com/2dom/PxMatrix/issues/225)

The software continues with this solutions, only now my screen stays black... so into coding again ;-)