2dom/PxMatrix #114: Putting the display task into second core on ESP

RobertHolscher, 2019-07-03:
Hi 2dom,

Instead of messing around with the interupts in handling the display, would it not be an idea to put the display handling in core 0 of the ESP32 as task ?
This is then always securing the refreshrate and will not have to do with any other handling of code. How to do this efficiently and if it will work is something I can not judge at the moment but it would be a good use of the second core in the ESP32 module, don't you think ?

2dom, 2019-08-20:
I tried this before not with much success. Feel free to try and send a pull request if you get it to word :)

closed 2019-08-20 by 2dom

witnessmenow, 2019-08-24:
I've had good success with the opposite, run PxMatrix on the regular Arduino core and run some other stuff (i2c and maths calculations for the falling sand project) on core 0. I will share my code the next time I'm on my PC