2dom/PxMatrix #116: chaining 8 number of P5 led matrix 64*32, as 2*4=8, by declaring display1(   ), dispaly2()  etc

sudheernizwa, 2019-07-10:
Hi Sir
I have just now succeeded chaining of 8 P5 matrix with esp32 by declaring 4 dipalyX(128,32, latchX,O_EX,A,B,C,D)
I am getting displays all over the led matrix, but the problem is the brightness level is falling too low and the flickering effect is more, 
is there a soluttion for this problem. 
here is the part of the declaration,
dipaly1(128,32, latch1,O_E1,A,B,C,D)
dipaly2(128,32, latch2,O_E2,A,B,C,D)
dipaly3(128,32, latch3,O_E3,A,B,C,D)
dipaly4(128,32, latch4,O_E4,A,B,C,D)

inside setup()
[
display1.begin(16)
display1.begin(16)
display1.begin(16)

sudheernizwa, 2019-07-10:
inside setup()
[
display1.begin(16)
display2.begin(16)
display3.begin(16)
display4.begin(16)
}

2dom, 2019-08-20:
No ... sorry. The SPI bus just cannot keep up with all the data. If you intend to drive a larger number of matrices I suggest another library. PxMatrix is optimized for low pin count and not made to drive lot's of panels.

closed 2019-08-20 by 2dom