2dom/PxMatrix #117: Chipset ICN2037 / Driver 5958

pyzimmer, 2019-07-26:
Hello, I bought this panel with SMD 2525 : https://fr.aliexpress.com/item/32836331716.html?spm=a2g0s.9042311.0.0.232f6c37cu1lnO

I am able to diplay something (1 line on 2) but unable to diplay test_pattern !

The seller said me to use 5958 decoding, i think it is that : http://www.ti.com/lit/ds/symlink/tlc5958.pdf

I send you picture the chips :
[image]
[image]
[image]

pyzimmer, 2019-07-26:
When I unplugg A, B and C pins, and I touch them with my fingers, the test_pattern appears sometimes. Strange...

pyzimmer, 2019-07-26:
Here is a video of unplug A pin : https://photos.app.goo.gl/4xqDg5gXukx6tXMZA

pyzimmer, 2019-07-26:
I try to connect A, B, C to GND and to VCC, no way... The only way to get the test pattern is touching the wire with my fingers..

pyzimmer, 2019-07-27:
Could you help me to understand datasheet of ICN2037 : http://en.chiponeic.com/content/details36_245.html
and state graphs : https://www.mikrocontroller.net/attachment/392914/ICN2037.pdf
For 74HC245 chip : https://assets.nexperia.com/documents/data-sheet/74HC_HCT245.pdf
And RT5957 : http://www.dlfushi.com/uploads/201805/5b03e7bc77f3a.pdf
thank you !

pyzimmer, 2019-07-31:
The datasheet of 74HC245 specify that it has 3 states output (High Impedance) . I modify the code like that :

```
 if (_mux_pattern==BINARY)
  {
    if (value & 0x01){
	  pinMode(_A_PIN, OUTPUT);
      digitalWrite(_A_PIN,LOW);
    }else{
	  pinMode(_A_PIN, INPUT);
	  digitalWrite(_A_PIN, HIGH);
	}
	  
    if (value & 0x02){
	  pinMode(_B_PIN, OUTPUT);
      digitalWrite(_B_PIN,LOW);
    }else{
	  pinMode(_B_PIN, INPUT);
	  digitalWrite(_B_PIN, HIGH);
	}

    if (_row_pattern>=8)
    {
	  if (value & 0x04){
			pinMode(_C_PIN, OUTPUT);
			digitalWrite(_C_PIN,LOW);
		}else{
		  pinMode(_C_PIN, INPUT);
		  digitalWrite(_C_PIN, HIGH);
		}
    }
```

With this code I can display Test_pattern, but nothing else...

pyzimmer, 2019-08-02:
I think it is a problem of voltage level. I will order 3,3->5v level shifter.

pyzimmer, 2019-08-05:
No difference with the level shifter.
Any idea ?

2dom, 2019-08-05:
If it works if you touch the wire it sounds like a grounding issue. Make sure that esp32 and panel ground are connected property.

pyzimmer, 2019-08-05:
Thanks you, I was my first idea, but no way. I grounded, all unused pins and ESP32 to power supply ground, no way ! The only pins that I don't ground are Hub75 outputs... I will try !

pyzimmer, 2019-08-05:
it does not work

pyzimmer, 2019-08-07:
I think the issue is about the mux. Could you explain me in which order the chips are linked :  ICN2037 , 74HC245, RT5957. Thank you.

2dom, 2019-08-20:
Sorry ... very busy atm ....

pyzimmer, 2019-09-20:
I read this in the datasheet :

> 9.3 Device Functional Modes
> After power on, all OUTXn of TLC5957 are turned off. All the internal counters and function control registers are
> initialized. Below is a brief summary of the sequence to operate TLC5957, just give users a general idea how this
> part works. After that, the function block related to each step will be detailed in following sections.
> 1. According to required LED current, choose BC and CC code, select the current programming resistor RIREF.
> 2. Send WRTFC command to set FC register value if the default value need be changed.
> 3. Write GS data of line 1 into GS data latch. Using LATGS command for the last group of 48bit GS data
> loading, the GS data written just now will be displayed.
> 4. Input GCLK continuously, 2
> N GCLK (N>=9) as a segment. Between the interval of two segments, supply
> voltage should be switched from one line to next line accordingly.
> 5. During the same period of step4, GS data for next line should be written into GS data latch. Using LATGS
> command for the last group of 48bit GS data loading.
> 6. Repeat step 4-5 until it comes to the last line for a multiplexing panel. Input 2
> N GCLK (N>=9) as a segment,
> at the same time, GS data for 1
> st line should be written into GS data latch. Using LINERESET command for
> the last group of 48bit GS data loading.
> Repeat step 4 through 6

the datasheet : http://www.ti.com/lit/ds/symlink/tlc5957.pdf

Do you know if there is any difference with your library ?

pyzimmer, 2019-09-21:
Chip ICN2037 is a 16 bits output register, your library use 8 bits , that's right ? 
What should I modify in the code for 16 bits ? thanks

texynz, 2019-09-22:
Seems to be related to issue 119 and the ICN 2037 chip although these should be fully compatible according to data sheets and don't require the triple clocking like the fm6126 and icn2038s. I have trying to work through the code and see where it fails. If I look at the comments by Brian in his article the issue relates to pin E required even though 16 scan, but I have set that and still not working. Can happily populate the even numbered rows 0,2,4,6 but not the odd 1,3,5.

pyzimmer, 2019-09-22:
Thank you, I set 1/8 scan , because with 1/16 a white pix because 3 pix red, blue, screen. I connect ABC pins, E si grouded and D is usefull...

pyzimmer, 2019-09-30:
hello, here is the result of :
display.fillScreen(WHITE);

[image]

@texynz : as you say  : Can happily populate the even numbered rows 0,2,4,6 but not the odd 1,3,5.

2dom, 2019-09-30:
Hi ..maybe this is related to https://github.com/2dom/PxMatrix/issues/119 ? You could try adding some delay...

pyzimmer, 2019-09-30:
thanks, i read it. I try adding 1/2 µs,  before/after, A and C mux, no way

texynz, 2019-09-30:
Instead of using the fillscreen function can you try just populating the rows manually in a for loop. As it is a timing issue, the fillscreen maybe adding additional timing.

pyzimmer, 2019-09-30:
@texynz  thank you, I try :
```
  for(int i=0; i<32; i++)
  {
  display.drawPixel(0, i, BLANC); 
  delay(500);
  }
```
strange result for a vertical line populating : https://photos.app.goo.gl/7nxpjen948Jk1UgK8
I try the same with a horizontal line populating and I is OK.

2dom, 2019-10-19:
You can try the latest master. I added an option to control the delay in every stage of the multiplexing.

Try adding display.setMuxDelay(0,1,0,0,0); after display.begin() .

pyzimmer, 2019-10-20:
Thank you for your work.
I try it, but no way,  it has no effect, the same for a vertical line populating : https://photos.app.goo.gl/7nxpjen948Jk1UgK8

I try with :
display.setMuxDelay(1,0,0,0,0); 
display.setMuxDelay(0,1,0,0,0); 
display.setMuxDelay(0,0,1,0,0); 
display.setMuxDelay(1,1,1,0,0); 
display.setMuxDelay(0,5,0,0,0);

pyzimmer, 2019-10-22:
The mario screen : some lines are missing, others are clone :
[image]

pyzimmer, 2019-10-23:
hello, when I unplug Bmux wire, some new lines appears 3 times :
[image]

pyzimmer, 2019-10-29:
I try to modify code of mux and latch, no way ! Any idea to modify the code of scan pattern ? thanks

2dom, 2019-10-29:
Ok...try to re-assign the B Wire to a differen pin

pyzimmer, 2019-10-29:
thank you for your help, same issue with pins 21, 22 or 23

pyzimmer, 2019-11-06:
Hello, 

Some new test, with this code :
```
for(int y=0; y<32; y++)
{
  display.drawLine(0,y,y,y, BLANC);
  delay(500);
}
```
line 1 should have 1 pixel,
line 2 should have 2 pixels,
...
However, i get this, with pin Bmux connected :
[image]

same code with pin Bmux unplug :
[image]

any idea ?

pyzimmer, 2019-11-06:
expected result (did with another panel model) :
[image]

pyzimmer, 2019-11-09:
I think the issue don't provide from ICN2037 chip, but from RT5957 like in this post: https://community.pixelmatix.com/t/led-module-p2-128-64-1-32-driver/414/5

yefeng0, 2019-11-26:
I have A new driver chip screen with ICND2038S, but I use driverchip “shift”, it works fine. but there is no detail about the driverchip register info.
[ICN2038S_datasheet_CN_2017_V1.3.pdf](https://github.com/2dom/PxMatrix/files/3893605/ICN2038S_datasheet_CN_2017_V1.3.pdf)
[image]

2dom, 2019-11-26:
@pyzimmer ... Do you have a data sheet?

pyzimmer, 2019-11-26:
Thank you, I compare the two datasheets, the only differnence I seen is OE time = 60ns for ICN 2037 instead of 40ns for ICN2038 . I will try to increase OE_pin activation.

pyzimmer, 2019-11-26:
Yes @2dom : https://github.com/2dom/PxMatrix/issues/117#issuecomment-515672955

yefeng0, 2019-11-26:
> Thank you, I compare the two datasheets, the only differnence I seen is OE time = 60ns for ICN 2037 instead of 40ns for ICN2038 . I will try to increase activation time of OE_pin.

ohmm, i have visited the chip's official website,  ICN2038 is a little different with ICN2038S, with suffix "S", have LAT registers, ICN2038 none.

pyzimmer, 2019-12-04:
I try another test :
```
for(int y=0; y<32; y++)
  {
   i++;
   if(i==1) color= RED;
   if(i==2) color= GREEN;
   if(i==3) color= BLUE;
   if(i==4) {color= WHITE; i=0;}
   display.drawLine(0,y,y,y, color);
  delay(500);
  }
```
the result :
[image]

when I unplug Bmux :
[image]

when I unplug Amux :
[image]

pyzimmer, 2019-12-12:
Hello, I found something : the chip RT5957 replace classic 74HC138 chip for multiplexing. This chip use a DATA/CLOCK pin instead of ABC pins. But I have no idea how to modify the code...
I send you an english translation of the chinese datasheet :
[RT5957.zh-CN.en.pdf](https://github.com/2dom/PxMatrix/files/3953383/RT5957.zh-CN.en.pdf)

DanielRTRD, 2020-01-08:
I have the same issue. And will happily help with testing.

xsrf, 2020-04-13:
I think A, B and C are actually DATA, CLK and LATCH of an shift register used for muxing.
I guess A is DATA, because leaving it floating actually shifts out random noise activating all lines randomly,
Can you trace A, B, and C to actual chip pins with a multimeter / by inspection?

pyzimmer, 2020-04-15:
Hello, you are right :
A,B,C pins are connected to A0,A1,A2 of the 74HC245.
Then B0,B1,B2 pins of 74HC245 are connected to LCK, BK, DIN pins of RT5957.
LCK and BK are connected to one RT5957 chip, and DIN to another RT5957 chip. (there are 4 RT5957 chips)

> I send you an english translation of the chinese datasheet :
> [RT5957.zh-CN.en.pdf](https://github.com/2dom/PxMatrix/files/3953383/RT5957.zh-CN.en.pdf)

xsrf, 2020-04-15:
@pyzimmer you may try https://github.com/xsrf/PxMatrix/tree/shift-mux , I've added SHIFTREG as additional mux pattern via https://github.com/xsrf/PxMatrix/commit/b91bbd4d4f55346e30fa8cb3320ea82f6b91ed32

Just use this within pattern_test etc:
```
  display.begin(16); // or 32 ...
  display.setMuxPattern(SHIFTREG); // must be set after begin()!
```
Performance is pretty bad, it takes 45µs to shift out 16 bits for the mux...
The RT5957 datasheet didn't mention if data is clocked in / latched on rising or falling edge so I just guessed.

xsrf, 2020-04-16:
Btw. if this works, rewiring could speed up performance and allow us to use SPI hardware for the muxing too.

Option 1) Joining CLK+A and DATA+C together on the first panel would allow using SPI in parallel for muxing and pixels. We would send out up to 4 Bytes of data via SPI for mux (before or after Pixeldata is sent and latched), then latch B. This would also free up 2 Pins on the ESP (only one needed instead of A,B,C)

Option 2) Connect CLK(OUT) of the last panel to A(IN) on the first panel, DATA(OUT) of the last panel to C(IN) on the first panel and LATCH(OUT) of the last panel to B(IN) on the first panel.
This would require us to send up to 4 Bytes for muxing prepended to the Pixeldata. There MAY be a small issue because the propagation delay of the data from the Input of the first panel to the output of the last panel MAY require a tiny guard delay between latching and enabling LED outputs.
This would however free up all muxing pins on the ESP (A,B,C, ... not needed)

pyzimmer, 2020-04-16:
it seems good, but flashing (some frames are missing on the video)  : https://photos.app.goo.gl/DWrPRg9NXvByzgSV6

pyzimmer, 2020-04-16:
Very good job : https://photos.app.goo.gl/UQMMxsuhDiBzuu8AA
I am going to do more test...

xsrf, 2020-04-16:
Thx, at least a steady row selection :) But that's only the first mux selection. What happens when all four lines turned white? Is it repeating the same but with all four lines offset by one?
The display clearly is an 1/8 mux btw, so try to use `begin(8)`... not sure what kind of ZIGZAG pattern this is though.

pyzimmer, 2020-04-16:
the result :
[image]

Do you need video of complete pattern_test ?

I have to emulate 4 panels for pattern :
display.begin(8);
display.setMuxPattern(SHIFTREG);
display.setScanPattern(LINE);
display.setPanelsWidth(4);

I'll try with 3 panels...
Thanks a lot !!!

xsrf, 2020-04-16:
I think we can skip the full pattern test :)
If you have some performance measurements let me know... I'll try to implement a faster method.

xsrf, 2020-04-16:
@pyzimmer I've moved to Hardware SPI now for muxing via https://github.com/xsrf/PxMatrix/commit/c34b1b38439277456ab8eeeff63597bf4cdf1c99 , you may try the latest version of https://github.com/xsrf/PxMatrix/tree/shift-mux

Be aware, this one **requires rewiring**!!
Pin A of the Panel input must be connected to CLK.
Pin B of the Panel input still is connected to the ESP Pin labeled B in the code.
Pin C of the Panel input must be connected to DATA.

Now Muxing and Pixel-Data share the same SPI bus. This frees two pins on the ESP.

pyzimmer, 2020-04-16:
Performances are quite the same, I can refresh a complete panel of 192x32 pixels every 100ms :
[image]

pyzimmer, 2020-04-16:
Thank you , I'll try this tomorrow :
> @pyzimmer I've moved to Hardware SPI now for muxing via [xsrf@c34b1b3](https://github.com/xsrf/PxMatrix/commit/c34b1b38439277456ab8eeeff63597bf4cdf1c99) , you may try the latest version of https://github.com/xsrf/PxMatrix/tree/shift-mux
> 
> Be aware, this one **requires rewiring**!!
> Pin A of the Panel input must be connected to CLK.
> Pin B of the Panel input still is connected to the ESP Pin labeled B in the code.
> Pin C of the Panel input must be connected to DATA.
> 
> Now Muxing and Pixel-Data share the same SPI bus. This frees two pins on the ESP.

pyzimmer, 2020-04-17:
Hello,
It does not work, the video of test_pattern : https://photos.app.goo.gl/zs7azTvtNrFv5QVA7
Furthermore, rewiring is not very convenient when using dupont cable. Or is it possible to bridge 2 output pins on ESP ?

xsrf, 2020-04-18:
Hmm, the video shows that the mux data actually got clocked into the pixeldata and the mux is not working... not sure why this happened. Have you swapped the latch pins?

Please check the wiring? It should look like this:

[image]

DATA+C share the same DATA Output of the ESP
CLK+A share the same CLK Output of the ESP
B is still connected as usual.

I know that connecting dupont cables to two pins is harder, a breadboard might help :)

This is the actual timing measured in regular update mode on my ESP8266:

[image]

It should clock in the data for the mux and latch the mux. Then it should clock in all data for the pixels and latch these...

One difference between ShiftOut() and HSPI is that ShiftOut had the data active on rising and falling edge. HSPI only on rising edge:

ShiftOut:
[image]

HSPI:
[image]

But that should only create a line shift in worst case...

xsrf, 2020-04-18:
I guess I'll implement both methods anyway to be compatible with existing wiring/boards for those who can accept the lower refresh rate caused by shiftout. But I'd be interested if it actually would work with shared SPI.

xsrf, 2020-04-18:
Just FYI, here is a comparison between ShiftOut and SPI speed @ 20MHz. It takes more than twice as long to just select the mux, than it takes to shift out RGB Data for 128 Pixels.

[image]

xsrf, 2020-04-18:
In terms of wiring, the sequential muxing approach would actually be easier and more compatible... It would require three additional connections from the output of the last panel to the input of the first panel, as mentioned two days ago:

[image]

xsrf, 2020-04-18:
Actually, both wiring should work with exactly the same code... The only code difference is that PIN_B is not needed in the sequential one 🤷‍♂️

DanielRTRD, 2020-04-18:
@xsrf Amazing work! I will hopefully get time this weekend to test! 👍

2dom, 2020-04-18:
Great work @xsrf ... Thanks for looking into this

pyzimmer, 2020-04-18:
I check my wiring, It is good.
Maybe it is because :
- I use an ESP32 ?
- I use 3 panels of 32*64 for total 32*128
- my config is : display.setPanelsWidth(12); (pattern to emulate 4 little panels per panel)

I'll try your sequential wiring this afternoon

[25 more replies]

closed 2020-05-03 by pyzimmer