2dom/PxMatrix #118: False pixels

wavewolf, 2019-07-31:
I'm using an Adafruit 16x32 1/8 RGB rgb array and I'm having unwanted behaviors.

When I command to write anything with the "print" function or any other function, there are several pixels on the screen unwanted and it looks very bad visually.

[image]

I don't know how to solve it, any idea?

robjen, 2019-08-14:
this is not an error in the code library   it is electrical noise ... tidy up your wiring and make sure your ground connections are good !

2dom, 2019-08-24:
@robjen is probably right ... also make sure that your power supply is at least 3A and that you use cables that can handle the current. Please re-open if this does not help.

closed 2019-08-24 by 2dom