2dom/PxMatrix #119: New P3 64x32 Issue

556duckvader, 2019-08-12:


2dom, 2019-08-20:
can you post the pattern test, post the result as gif/video and share your code ?

556duckvader, 2019-08-25:
Here is the code I used for the Mario Image.  I am working on getting the video of the test pattern uploaded.
[CodeUsed.txt](https://github.com/2dom/PxMatrix/files/3538843/CodeUsed.txt)

maditnerd, 2019-08-26:
I have a similar issue, 4 lines works and then the next 4 doesn't works.
I also followed Brian's guide (I bought the matrix on aliexpress with the link, he provided on his tetris clock repo)

https://github.com/witnessmenow/WiFi-Tetris-Clock

maditnerd, 2019-08-26:
I contacted the manufacturer, who confirmed other people has issues with ESP32/ESP8266 and suggested to try it with a Raspberry Pi.

I try the matrix using https://github.com/hzeller/rpi-rgb-led-matrix/ and it works fine with default settings.

Time to find what is going wrong with the ESP32, i'll report what I try here :-)

maditnerd, 2019-08-27:
It seems related to #104

Chipset on my matrix

* RuiChips RUC7258 (Led Gate Driver) : http://www.ruichips.com/uploads/file/20180816/20180816053859223.pdf
* Chipone ICN2037 (16-Channel Constant Current LED Sink Driver with Dual Latch) http://en.chiponeic.com/content/details36_245.html
* Chipone ICN74HC245 (Logic Control) http://en.chiponeic.com/content/details38_95.html

serestre, 2019-08-30:
I have same problem

serestre, 2019-08-31:
This was the response of the provider:

"hello friend, please connect it to raspberry pi controller. ESP8266 controller is updated, not support our led panels recently. Any questions, please feel free to contact me."

scarecrow2k6, 2019-08-31:
I too have the exact same problem, tried with RPi works great.

2dom, 2019-08-31:
Did you try with and without fastUpdate ... might be a timing issue ....

2dom, 2019-08-31:
Cannot get hold of that RUC7258 pdf ... does any one of you happen to have it?

2dom, 2019-08-31:
@556duckvader I would assume your display is 8 scan. 

Set ` display.begin(8);`

serestre, 2019-09-03:
This is the Chipset on my matrix

[image]

[image]
[image]
[image]

texynz, 2019-09-14:
Same issue, same chipset. Removing E makes no difference, neither does fastupdate. Obviously how the library latches to the next row

texynz, 2019-09-15:
RUC7258 pdf
http://www.ruichips.com/uploads/file/20180819/20180819083248457.pdf

texynz, 2019-09-15:
Did some testing and used the following:

  display.drawPixel(0,0,myRED);
  display.drawPixel(0,1,myRED);
  display.drawPixel(0,2,myRED);
  display.drawPixel(0,3,myRED);
  //
  display.drawPixel(0,4,myGREEN);
  display.drawPixel(0,5,myGREEN);
  display.drawPixel(0,6,myGREEN);
  display.drawPixel(0,7,myGREEN);
  //
  display.drawPixel(0,8,myBLUE);
  display.drawPixel(0,9,myBLUE);
  display.drawPixel(0,10,myBLUE);
  display.drawPixel(0,11,myBLUE);

It should give three vertical lines of red, green,blue. But only shows RED and BLUE. If I remove the D pin leave it floating I can see the green show up (flicking) in position 2, but also in position 4 of which I have not programmed. Depending on the state of the "flicker:" if I earth the D, then I can get the position 2 GREEN to stay on. But the other states will now not light up. 
So in conclusion it its leading me to think that on a row change, the D pin needs to be constantly toggled. Funnyily the D pin on the ESP is sitting at 2.5V... and even if I just ground the D pin then I only get the GREEN POSITION.

Will have to have more of a play, but hopefully that info will help someone investigate further

2dom, 2019-09-16:
Did you set the right scan in 

display.begin(x)?

texynz, 2019-09-16:
Yes set it to 16. Tried others also just in case it was some odd setting but no difference. I am running an older pxmatrix.h file that doesn't have an compile issue with the esp8266, but even on the latest build on an esp32 I get the same results.

zakrava, 2019-09-22:
Hi, I have P3 64x32 1/16 panel https://www.aliexpress.com/item/1615847398.html and it has another issue, when I drive it with esp8266 it turns all leds white or show two white stripes.
It has FM6126A, TC7258EN and 74HC245C chips and pin E not grounded.

zakrava, 2019-09-22:
Ok, false alarm, I needed just call display.setDriverChip(FM6126A); like it is in mario example for esp32 and now it works like a charm. Sorry and thanks for great library.

witnessmenow, 2019-09-26:
> 
> 
> I have a similar issue, 4 lines works and then the next 4 doesn't works.
> I also followed Brian's guide (I bought the matrix on aliexpress with the link, he provided on his tetris clock repo)
> 
> https://github.com/witnessmenow/WiFi-Tetris-Clock

Hey @maditnerd , can you confirm where you used the link and I'll at the very least update it with a note to this issue

Sorry that it didn't work, it was the link i used for myself!

maditnerd, 2019-09-26:
@witnessmenow No problem i'll used it with a Raspberry Pi.
Yes, it is the link from your youtube video : EMvjy3z from **SRYLED Display Store**
Maybe they changed the chipset ? 

I should have tell you first, I was hoping it was simple to fix, but It seems to be more complicated then I would have thought. 

Next time, i'll ask the seller if their matrix is compatible with ESP32. They responded pretty quickly so it's a good idea to do before buying.

2dom, 2019-09-26:
Does anyone of you live in central Europe and can send a spare matrix I can test with?

egrekov, 2019-09-27:
They also bought a panel 64x32 P3, tried a bunch of sketches and everywhere 4 stripes shine, 4 do not shine ...
[image]
[image]

witnessmenow, 2019-09-27:
I don't have a problem board, but I would be happy to contribute to you getting one from the same seller @2dom

2dom, 2019-09-27:
How about this ... tell me the seller and I will get one myself ... if it works you guys can buy me a beer :)

witnessmenow, 2019-09-27:
I have updated the readme on the WiFi clock with the warning about this display. Does anyone have a good source from Aliexpress at the moment.

zakrava, 2019-09-27:
Hi, I bought P3 64x32 panel, but there is FM6126A and with esp8266 brightness doesn't work, from here https://www.aliexpress.com/item/1615847398.html
I was looking for other P3 64x32 display, but it looks like it has the same "fm6126a layout" https://www.aliexpress.com/item/32728985432.html

texynz, 2019-09-27:
https://s.click.aliexpress.com/e/dj9WupK0
This is the panel I bought. The issue is that the chips may change depending on the batch. I didn't get the FM chip and have the issue with the every 4 rows. If you float the a b C pins in you hand like someone suggested the entire screen lights up but flickers. The data is correctly being shifted in but the timing for the scan is either out or requires a different of triggering

egrekov, 2019-09-28:
Conducted a little research, maybe it will somehow help
[image]

egrekov, 2019-09-28:
Presumably the impulse of line C is supposedly short and because of this OUT4-8 do not have time to grow to the desired amplitude, like A, B, D

2dom, 2019-09-28:
Did you try adding a few nops to the code yet?

egrekov, 2019-09-28:
Unfortunately not so strong as to add changes.

2dom, 2019-09-28:
Could you try adding delayMicroseconds(1); to the very end of the setMux function?

egrekov, 2019-09-28:
I tried to add delayMicroseconds(1); in set_mux. [commit](https://github.com/egrekov/PxMatrix/commit/508da87f44a56ec80d453a225cf9d8e01acc5179)
But I had to use an older version of the library [pxMatrix.zip](https://github.com/2dom/PxMatrix/files/3610615/PxMatrix.zip), otherwise I get an error SPI, how in [#issue129](https://github.com/2dom/PxMatrix/issues/126)

2dom, 2019-09-28:
No worries...will wait for the panel to arrive (Brian thankfully donated one) and I will try to figure out what the timing issues are. You scope plots are very helpful ! Thanks!

egrekov, 2019-09-29:
Add delayMicroseconds(1); only line C [commit](https://github.com/egrekov/PxMatrix/commit/33a8d49bea703b36597d78ac90e3380b2e341c47)
Everything started to display correctly.  There are small artifacts when a large picture is drawn, for example, mario.
[image]
[image]
1. [Youtube video - Mario](https://youtu.be/nwVtCHrKraE)
2. [Youtube video - Banana](https://youtu.be/PccuIavyugk)
3. [Youtube video - Circle](https://youtu.be/3IXsdAyPuL4)
4. [Youtube video - Clock](https://youtu.be/foNBCUxge6I)

2dom, 2019-09-29:
Can you try putting the delay after digitalWrite(_C_PIN,HIGH); YOu could also try increasing the delay if that does not do the trick.

texynz, 2019-09-29:
Nice work. I was so close but had the delay in a different place. If you increase the delay it will reduce the flicker and artifact.

2dom, 2019-09-29:
So what are you using now...could you post the snippet of code ?

texynz, 2019-09-29:
Same place as egrekov in the binary part of setmux for line C but delay of 2us. delay must before the pull high.

egrekov, 2019-09-30:
By advice @texynz increase delayMicroseconds to 2 and sent pull request to @2dom, to include support for the new display in the library code.

egrekov, 2019-10-03:
Nevertheless, the addition of microseconds 1-10 in delayMicroseconds, adds a flicker to the display, of course this fixes the problem temporarily, but the picture is not so pleasant for the eye.

2dom, 2019-10-03:
Will look into it as soon as I have a panel ...

2dom, 2019-10-18:
You can try the latest master. I added a option to control the delay in every stage of the multiplexing.

Try adding `display.setMuxDelay(0,1,0,0,0);` after `display.begin()`.

2dom, 2019-10-18:
This would a 1us after the B channel

2dom, 2019-10-18:
@witnessmenow Thanks for the panel and the matrix shield - working just fine !

scarecrow2k6, 2019-10-21:
@2dom Very sorry about this but I can't seem to figure out what is going on. Somewhat of a noob here... tried with both Arduino IDE and VSC- PlatformIO

[image]

egrekov, 2019-10-21:
@scarecrow2k6 You need to update the library
https://github.com/2dom/PxMatrix/blob/master/PxMatrix.h

egrekov, 2019-10-21:
Spent more testing. it turns out a working commit 4d8c6d4, but I had to comment out lines 654-657 otherwise it did not compile.
`PxMatrix.h:655:123: error: expected ';' before ')' token`
Beginning with the de2df96 commit, problems have already appeared, and at the last commit, not a single LED is working at all.
[image]

2dom, 2019-10-21:
Thanks for the feedback @egrekov . Left a stray return in the function :) Check again with the latest master please!

egrekov, 2019-10-22:
@2dom, thank you so much, everything works, you're just super)

2dom, 2019-10-22:
No worries ... Glad that it works :)

closed 2019-10-22 by 2dom

zakrava, 2019-11-25:
Hi, I have 64x32 P2.5 16S V1.0 panel with ICN2037 and RUC7258 and when I connect it, display is totally blank.
I tried to play with setMuxDelay, but it doesn't help at all. I have latest master.

tobobus, 2019-11-25:
Maybe this helps. see also issue 139.

Via the "discussions" from the site of Morphing Digital Clock by HariFun i tested a suggestion from "timz3818 " to disable "#define double_buffer " and voila this works.
No black panel-screen after PX 1.7.0 and ESP 8266.
It works for Hari's clock, the remix and further variations.

I have no ideer what double buffer does in those scetches, but disabling works somehow.

zakrava, 2019-11-25:
I disabled double_buffer a long time ago, otherwise my P3 and P4 panels didn't work.
But P2.5 with two ICN74HC245TS, four RUC7258 and a lot of ICN2037BP is not showing a single pixel.
It is from here https://www.aliexpress.com/item/32650905490.html but is has different chips and layout than on the picture there.

icepick7, 2019-12-03:
@zakrava
Hello, I have a P3 display witch the exact same ic's an the same behavior, no pixel lights up. Except the pattern_test and the double_buffer Demo works as expected. I cant understand that behavior :(

Porokoura, 2019-12-16:
> I disabled double_buffer a long time ago, otherwise my P3 and P4 panels didn't work.
> But P2.5 with two ICN74HC245TS, four RUC7258 and a lot of ICN2037BP is not showing a single pixel.
> It is from [...]

 I currently have 3 different 64 * 32 displays, all with different chip sets. One of them does not have a RUC7258 circuit. This is the only one that works seamlessly with ESP32. There are different problems with those with the RUC725 (eg issue # 152) All 3 of my monitors run smoothly on the Arduino Mega, unfortunately the Arduino's performance is not sufficient for my needs.

2dom, 2019-12-16:
Do you use PxMatrix on Arduino Mega?

2dom, 2019-12-16:
If so you could try reducing SPI speed

Porokoura, 2019-12-17:
I don't use PxMatrix in arduino. I just wanted to point out that displays were working.

[6 more replies]