2dom/PxMatrix #120: No data out with FM6126A

cQszasza, 2019-08-27:
Hi guys, 

I'm trying to use this lib with a nodemcu-32s and a 32x64 display with FM6126A chipset but if I add display.setDriverChip(FM6126A); to setup no data or clock is coming out from the ESP. ABCD, OE, and LATCH lines are doing their thing (confirmed with scope) But nothing on CLK and MOSI pins. If I comment out the setDriverChip line there is data coming but obviously the display isn't working. Any ideas?

cQszasza, 2019-08-28:
I've tried it with [this fork](https://github.com/kkaoo/PxMatrix) and it is working now. Its just weird that the original library isn't.

556duckvader, 2019-08-28:
> 
> 
> I've tried it with [this fork](https://github.com/kkaoo/PxMatrix) and it is working now. Its just weird that the original library isn't.

I take it this fork is only for ESP32's

2dom, 2019-09-01:
Hi guys ... just merged the changes by @kkaoo ... do you want to give it another try?

cQszasza, 2019-09-02:
Its working fine now with this lib, the brightness setting as well.  I'm using gpio 27 instead of gpio 2 for OE, I just couldn't upload with gpio 2 connected.

556duckvader, 2019-09-08:
> 
> 
> Its working fine now with this lib, the brightness setting as well. I'm using gpio 27 instead of gpio 2 for OE, I just couldn't upload with gpio 2 connected.

Can you post a pic of your wiring?  I am trying to make sure I have mine wired up correctly.

closed 2019-10-20 by 2dom