2dom/PxMatrix #121: Arduino Clock PIN

Robotech-e, 2019-09-03:
Hi everyone!

I´m working on a project just to show some static text and maybe scrolling text, I was wondering if I can use PxMatrix to do that with an Arduino UNO, (at the Readme appears "The driver is Adafruit GFX compatible and currently works with ESP8266 and ESP32 microcontrollers. However, it should be rather straightforward to port it to Atmel-based Arduinos."), I suppose I have to edit the PxMatrix.h file at:

// HW SPI PINS
#define SPI_BUS_CLK 14
#define SPI_BUS_MOSI 13
#define SPI_BUS_MISO 12
#define SPI_BUS_SS 4

To fit UNO hardware and set CLK to pin 8 like Adafruit recommends?

And then at the sketch declare the variables for the rest of the pins, something like this:

#define P_LAT 10
#define P_A A0
#define P_B A1
#define P_C A2
#define P_D A3  // if present
#define P_E A4   // if present
#define P_OE 9

?????

Or all this is just an alucination of my own and is better to use ESP8266 ?

Thanks for your comments and if is the case, for your recommendations

:-)

witnessmenow, 2019-09-27:
It would be a bigger change than that because those HW SPI PINS are only for the ESP32 iirc

The short answer is that it would be easier to just use an ESP8266

Longer answer would be to use Adafruits library: https://learn.adafruit.com/32x16-32x32-rgb-led-matrix/connecting-with-jumper-wires

Robotech-e, 2019-09-27:
Thank you bro, but with my panel the Adafruit´s option doesn´t work, I have a P10 panel,
the issue is the "y" axis is interchanged on both rows of the panel, for example, if you draw a pixel at x=0, y=0 it appears at x=0, y=4, on the other hand if you draw the pixel x=0, y=4 it appears at x=0, y=0 and is the same with the lower row if you draw a pixel at x=0, y=8, it appears at x=0, y=12

If you write something the letters are cutted by the half.

So for mental health I´ll work with the ESP8266 :-)

[image]

closed 2019-10-19 by 2dom