2dom/PxMatrix #126: Error compiling sample, SPI-related errors

rtek1000, 2019-09-10:
Hello,

SPI-related errors, but I tested other code that uses SPI and the Arduino IDE 1.8.9 was able to compile.

How can I solve this please?

Pattern_test:
```
In file included from C:\ESP8266_P5_32x64\pattern_test\pattern_test.ino:1:0:

C:\Arduino IDE\arduino-1.8.9\portable\sketchbook\libraries\PxMatrix/PxMatrix.h: In member function 'void PxMATRIX::fm612xWriteRegister(uint16_t, uint8_t)':

C:\Arduino IDE\arduino-1.8.9\portable\sketchbook\libraries\PxMatrix/PxMatrix.h:393:5: error: 'spi_t' was not declared in this scope

     spi_t * spi = SPI.bus();

     ^

C:\Arduino IDE\arduino-1.8.9\portable\sketchbook\libraries\PxMatrix/PxMatrix.h:393:13: error: 'spi' was not declared in this scope

     spi_t * spi = SPI.bus();

             ^

C:\Arduino IDE\arduino-1.8.9\portable\sketchbook\libraries\PxMatrix/PxMatrix.h:393:23: error: 'class SPIClass' has no member named 'bus'

     spi_t * spi = SPI.bus();

                       ^

C:\Arduino IDE\arduino-1.8.9\portable\sketchbook\libraries\PxMatrix/PxMatrix.h:399:29: error: 'spiSimpleTransaction' was not declared in this scope

     spiSimpleTransaction(spi);

                             ^

C:\Arduino IDE\arduino-1.8.9\portable\sketchbook\libraries\PxMatrix/PxMatrix.h:413:26: error: 'spiEndTransaction' was not declared in this scope

     spiEndTransaction(spi);

                          ^
```

Pixeltime:
```

In file included from C:\Arduino IDE\arduino-1.8.9\portable\sketchbook\libraries\PxMatrix\examples\pixeltime\pixeltime.ino:2:0:

C:\Arduino IDE\arduino-1.8.9\portable\sketchbook\libraries\PxMatrix/PxMatrix.h: In member function 'void PxMATRIX::fm612xWriteRegister(uint16_t, uint8_t)':

C:\Arduino IDE\arduino-1.8.9\portable\sketchbook\libraries\PxMatrix/PxMatrix.h:393:5: error: 'spi_t' was not declared in this scope

     spi_t * spi = SPI.bus();

     ^

C:\Arduino IDE\arduino-1.8.9\portable\sketchbook\libraries\PxMatrix/PxMatrix.h:393:13: error: 'spi' was not declared in this scope

     spi_t * spi = SPI.bus();

             ^

C:\Arduino IDE\arduino-1.8.9\portable\sketchbook\libraries\PxMatrix/PxMatrix.h:393:23: error: 'class SPIClass' has no member named 'bus'

     spi_t * spi = SPI.bus();

                       ^

C:\Arduino IDE\arduino-1.8.9\portable\sketchbook\libraries\PxMatrix/PxMatrix.h:399:29: error: 'spiSimpleTransaction' was not declared in this scope

     spiSimpleTransaction(spi);

                             ^

C:\Arduino IDE\arduino-1.8.9\portable\sketchbook\libraries\PxMatrix/PxMatrix.h:413:26: error: 'spiEndTransaction' was not declared in this scope

     spiEndTransaction(spi);

                          ^
```

rtek1000, 2019-09-10:
SPI Test Code (Can compile without errors):

```
#include<SPI.h>

char buff[]="Hello Slave\n";

void setup() {
 Serial.begin(9600); /* begin serial with 9600 baud */
 SPI.begin();  /* begin SPI */
}

void loop() {
 for(inti=0; i<sizeof buff; i++)  /* transfer buff data per second */
  SPI.transfer(buff[i]);
 delay(1000);  
}
```
Source: https://www.electronicwings.com/nodemcu/nodemcu-spi-with-arduino-ide

rtek1000, 2019-09-10:
Note: The examples are being compiled without errors for ESP32.

But for the ESP8266 is returning these errors.

rtek1000, 2019-09-13:
@yogeshwaran774 

I have no errors with ESP32. (IDE 1.8.9 ESP32 1.0.3)

Tip: Always test with the latest version of IDE and other supporting software.

rtek1000, 2019-09-13:
Library version 1.2.0 can be compiled with ESP8266, use library manager to regress version.

closed 2019-09-13 by rtek1000

witnessmenow, 2019-09-13:
Why did you close it? Seems like it's broken in main branch.

rtek1000, 2019-09-13:
@witnessmenow yep

rtek1000, 2019-09-13:
Version 1.2.0 has no brightness control, if anyone needs it immediately, follows the adapted file (PxMatrix.h):

[PxMatrix.zip](https://github.com/2dom/PxMatrix/files/3610615/PxMatrix.zip)

egrekov, 2019-09-30:
Added pull request which solves this problem. Fuction fm612xWriteRegister usage only ESP32 platform.