2dom/PxMatrix #127: My RGB LED Matrix shows only funny things.

Jetiman, 2019-09-28:
Hi,

I have a problem. My RGB LED Matrix shows only funny things. No matter what I hire.

Cable connected as described. A B C D

Have it already
"Display.begin (4);
display.setScanPattern (ZAGGIZ);
display.setMuxPattern (STRAIGHT); "
but it tries only as in the video

https://youtu.be/Nmd7f6M5Kak

https://de.aliexpress.com/item/1248297006.html?spm=a2g0s.9042311.0.0.88be4c4dxRtA3N

[image]

[image]

kenle, 2019-10-01:
@Jetiman This worked for me after some hacking around, search for the following code in PxMatrix.h and comment it out or delete it, it should be around line number 494 in the file.

```
  if ((_scan_pattern == ZAGGIZ) && ((y % (_row_pattern * 2)) < _row_pattern))
    bit_select = 7 - bit_select;
```

I have my modes set as
```
  display.begin(4);
  display.setScanPattern(ZAGGIZ);
  display.setMuxPattern(BINARY);
```

Jetiman, 2019-10-01:
same matrix?

makes the same mistake with the changes

kenle, 2019-10-01:
@Jetiman It looked like I had a similar problem with you on the "pattern test" video you have.  The Y-Axis pixels came out in the wrong order, however the X-axis was in the correct order.  

I'm using this panel: http://www.ylrled.net/product/75992050.html

After the fix I showed the Y-Axis ordering was corrected.

2dom, 2019-10-18:
Could you try the latest master with 

` display.setScanPattern(ZAGZIG);`

please?

Jetiman, 2019-10-19:
It does something different, but not yet the right one

plattern test
https://youtu.be/LISa_5ivWgw

pixelime
https://youtu.be/RLEhaaQUw8c

2dom, 2019-10-19:
OK ... I screwed up the boolean algebra bit ... can you try again with the latest master?

2dom, 2019-10-19:
with display.setScanPattern(ZAGZIG); of course

Jetiman, 2019-10-19:
gets better. But it is not right yet
what is better for testing, pixeltime or plates test

https://youtu.be/rbGpSaQU_7w

2dom, 2019-10-19:
just to get this straight .. you only commented 

```
  if ((_scan_pattern == ZAGGIZ) && ((y % (_row_pattern * 2)) < _row_pattern))
    bit_select = 7 - bit_select;
```

out and it worked with   `display.setScanPattern(ZAGGIZ);`  ?

2dom, 2019-10-19:
can you repeat/verify with the latest master?

Jetiman, 2019-10-19:
latest master

Okay, I commented that out in PxMatrix.h
`if ((_scan_pattern == ZAGGIZ) && ((y % (_row_pattern * 2)) < _row_pattern))
    bit_select = 7 - bit_select;`

My settings are:
```
PxMATRIX display(32,16,P_LAT, P_OE,P_A,P_B,P_C,P_D);
  display.begin(4);
  display.setScanPattern(ZAGGIZ);
  display.setMuxPattern(STRAIGHT);

```

added (at `(((_scan_pattern `----- because otherwise an error comes
`if ((yy<=y) && (y<yy+_row_pattern))
        total_offset_r=base_offset-row_sector__offset*row_sector-(((_scan_pattern==ZAGGIZ)||(_scan_pattern==ZAGZIG)) ? 1: 0);`

https://youtu.be/wYsXzSe81IE

2dom, 2019-10-19:
OK ... lets start from the beginning. Please run the pattern test and place

`display.displayTestPixel(70);`

in the display_updater() (and post the video)

Jetiman, 2019-10-19:
https://youtu.be/F9kLLQD2Cac

2dom, 2019-10-19:
Ca you post you display_update function....still looks like the pattern test

Jetiman, 2019-10-19:
sorry I have set the wrong setting
https://youtu.be/mxRiKaEZDpQ

```
#include <PxMatrix.h>

#ifdef ESP32

#define P_LAT 22
#define P_A 19
#define P_B 23
#define P_C 18
#define P_D 5
#define P_E 15
#define P_OE 2
hw_timer_t * timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

#endif

#ifdef ESP8266

#include <Ticker.h>
Ticker display_ticker;
#define P_LAT 16
#define P_A 5
#define P_B 4
#define P_C 15
#define P_D 12
#define P_E 0
#define P_OE 2

#endif
// Pins for LED MATRIX

//PxMATRIX display(32,16,P_LAT, P_OE,P_A,P_B,P_C);
PxMATRIX display(32,16,P_LAT, P_OE,P_A,P_B,P_C,P_D);
//PxMATRIX display(64,64,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E);

#ifdef ESP8266
// ISR for display refresh
void display_updater()
{
  //display.displayTestPattern(70);
   display.displayTestPixel(70);
  //display.display(70);
}
#endif

#ifdef ESP32
void IRAM_ATTR display_updater(){
  // Increment the counter and set the time of ISR
  portENTER_CRITICAL_ISR(&timerMux);
  //isplay.display(70);
  display.displayTestPixel(70);
  portEXIT_CRITICAL_ISR(&timerMux);
}
#endif

uint16_t myCYAN = display.color565(0, 255, 255);
void setup() {
  // put your setup code here, to run once:
Serial.begin(9600);
  display.begin(4);
  display.flushDisplay();
  display.setTextColor(myCYAN);
  display.setCursor(2,0);
  display.print("Pixel");
  Serial.println("hello");
  display.setMuxPattern(STRAIGHT);

  #ifdef ESP8266
    display_ticker.attach(0.002, display_updater);
  #endif

  #ifdef ESP32
    timer = timerBegin(0, 80, true);
    timerAttachInterrupt(timer, &display_updater, true);
    timerAlarmWrite(timer, 2000, true);
    timerAlarmEnable(timer);
  #endif

  delay(1000);
}

void loop() {

 delay(100);

}
```

2dom, 2019-10-19:
OK ... that is a really strange byte split pattern. I tried to implement it.

You can try the latest master, use pixeltime example with `display.setScanPattern(ZZAGG);  `

Jetiman, 2019-10-20:
It works. There are only a few single pixel errors in it. Above the font and top right

https://youtu.be/ZHA9_0a_gL0

2dom, 2019-10-20:
Ok...a few basic things to check. Do you have a power supply rated >3A and proper cables (beefy) between panel and power supply

2dom, 2019-10-20:
Verify the ground connection between panel and esp

Jetiman, 2019-10-20:
think it's up to my jumper cables. Power supply is a 5v -10A with big power cords

2dom, 2019-10-20:
Another thing is to make sure that the E wire is connected or grounded (try both)

2dom, 2019-10-20:
If you leave E floating on the panel it leads to artifacts on some panels

Jetiman, 2019-10-21:
I do not have an E pin. Have three GND. I have also connected. Will the days test other cables

kenle, 2019-10-21:
@Jetiman I'm not sure this will help in your case, but just throwing it out there in case it can solve your problem.

Things you can try:
1. Turn off fastUpdate and choose a "display_draw_time" other than zero (e.g. 25)
2. Reduce the SPI clock rate SPI.setFrequency();  I set mine to 7,500,000 [This could help with long lossy cables, lower frequencies will have less loss and less noise).

2dom, 2019-10-21:
Thanks @kenle !

jakaisna, 2020-05-27:
its work to p10 outdor

// This is how many color levels the display shows - the more the slower the update
//#define PxMATRIX_COLOR_DEPTH 8

// Defines the buffer height / the maximum height of the matrix
//#define PxMATRIX_MAX_HEIGHT 64

// Defines the buffer width / the maximum width of the matrix
//#define PxMATRIX_MAX_WIDTH 64

// Defines how long we display things by default
//#define PxMATRIX_DEFAULT_SHOWTIME 30

// Defines the speed of the SPI bus (reducing this may help if you experience noisy images)
//#define PxMATRIX_SPI_FREQUENCY 20000000

// Creates a second buffer for backround drawing (doubles the required RAM)
//#define PxMATRIX_double_buffer true

#include <PxMatrix.h>

// Pins for LED MATRIX
#ifdef ESP32

#define P_LAT 22
#define P_A 19
#define P_B 23
#define P_C 18
#define P_D 5
#define P_E 15
#define P_OE 2
hw_timer_t * timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

#endif

#ifdef ESP8266

#include <Ticker.h>
Ticker display_ticker;
#define P_LAT 16
#define P_A 5
#define P_B 4
#define P_C 15
#define P_D 12
#define P_E 0
#define P_OE 2

#endif

#define matrix_width 32
#define matrix_height 16

// This defines the 'on' time of the display is us. The larger this number,
// the brighter the display. If too large the ESP will crash
uint8_t display_draw_time=60; //30-70 is usually fine

//PxMATRIX display(32,16,P_LAT, P_OE,P_A,P_B,P_C);
PxMATRIX display(32,16,P_LAT, P_OE,P_A,P_B,P_C,P_D);
//PxMATRIX display(64,64,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E);

// Some standard colors
uint16_t myRED = display.color565(255, 0, 0);
uint16_t myGREEN = display.color565(0, 255, 0);
uint16_t myBLUE = display.color565(0, 0, 255);
uint16_t myWHITE = display.color565(255, 255, 255);
uint16_t myYELLOW = display.color565(255, 255, 0);
uint16_t myCYAN = display.color565(0, 255, 255);
uint16_t myMAGENTA = display.color565(255, 0, 255);
[... 170 lines cut ...]

  draw_weather_icon(icon_index);
  icon_index++;
  if (icon_index>10)
    icon_index=0;

  for (int xx=0; xx<16;xx++)
  {
    display.drawLine(xx+16,0,xx+16,5,display.color565(xx*16,0,0));
    display.drawLine(xx+16,6,xx+16,10,display.color565(0,xx*16,0));
    display.drawLine(xx+16,11,xx+16,15,display.color565(0,0,xx*16));
  }
  delay(1000);
  for (uint8_t dimm=255; dimm>0; dimm--)
  {
    display.setBrightness(dimm);
    delay(5);
  }
  for (uint8_t dimm=0; dimm<255; dimm++)
  {
    display.setBrightness(dimm);
    delay(5);
  }

}