2dom/PxMatrix #131: Wiring for a Matrix of displays

justsilviu, 2019-10-09:
Hello.

A simple picture with how the wiring for multiple panels in the readme would help.
i understand how 2 or 3 in a single row work, and i have tried and it works.
but how about creating a matrix of 3x3, how should that be wired?
and "display.setPanelsWidth();" function does not have a coresponding -  display.setPanelsHeight(); ?? 
or maybe the library is not design to handle multiple rows and columns?

thanks

2dom, 2019-10-18:
No vertical chaining supported yet ... will come soon

closed 2019-10-18 by 2dom

kenle, 2019-10-20:
@2dom how hard is this to implement?  Can you give me a hint where to start?  Any references on how these displays work so I can understand the code better? This is a critical feature for me now so I may need to try to implement myself.   Thanks 🙏

2dom, 2019-10-20:
There is a pull request implementing vertical chaining...had no time to review it...maybe start there

2dom, 2019-10-20:
The readme gives quite a bit of info how these displays work in general

justsilviu, 2019-10-20:
after i have played a lot with this library, the above settings might not work.
this setup sends the color data on only one wire, and this is somehow slow.
i have chained 2 panel in a row, and when i have scrolling text, which i mostly use,the matrix has a very visibile and strong flicker.
when the color of the text is onle red, or green or blue, it's ok, but if i use any other comibation, then it does not work anymore, it also depends on the size of the text.
but chaining even more i guess will be even worse.

@2dom  maybe an implementation of less bits color, like color333 would solve this?
have you considered adding this as a feature?

justsilviu, 2019-10-20:
> @2dom how hard is this to implement? Can you give me a hint where to start? Any references on how these displays work so I can understand the code better? This is a critical feature for me now so I may need to try to implement myself. Thanks 🙏

@kenle you can try another library which supports chaining of more matrices
https://github.com/pixelmatix/SmartMatrix/tree/teensylc , 
https://github.com/marcmerlin/SmartMatrix_GFX

i have tried them but found them very complicated for me.
it also depends on your aplication and controller.
these work only with ESP32, and teensy 3.x

2dom, 2019-10-20:
All correct ... PxMatrix is optimized for low bin count. If you want to use many panels with parallel color data lines you should rather use another library that operates that way.

2dom, 2019-10-20:
Regarding Colo depth ... Easy ... Just configure it in the header file

kenryukyu, 2019-10-21:
> @kenle you can try another library which supports chaining of more matrices
> https://github.com/pixelmatix/SmartMatrix/tree/teensylc ,
> https://github.com/marcmerlin/SmartMatrix_GFX
>
> i have tried them [...]

@justsilviu Thank you