2dom/PxMatrix #133: Need New Scan pattern

wardi1971, 2019-10-17:
Add need new scan patern for my led matrix Cai Liang 1/4 scan.

I add this code in PxMATRIX::fillMatrixBuffer in line number 447 

for (uint8_t yy = 0; yy<_height; yy+=2*_row_pattern)
    {
      if ((yy<=y) && (y<yy+_row_pattern))
        total_offset_r=base_offset-row_sector__offset*row_sector-(_scan_pattern==ZAGGIZ ? 1: 0);

//  ==== add new scan patter GAZGIZ =============== start
      if ((yy<=y) && (y<yy+_row_pattern))
        total_offset_r=base_offset-row_sector__offset*row_sector-(_scan_pattern==GAZGIZ ? 1: 0);
//  ==== add new scan patter GAZGIZ =============== end

      if ((yy+_row_pattern<=y) && (y<yy+2*_row_pattern))
        total_offset_r=base_offset-row_sector__offset*row_sector-(_scan_pattern==ZIGZAG ? 1: 0);
      row_sector++;
    }

And at line 476

  uint8_t bit_select = x%8;
  if ((_scan_pattern==ZAGGIZ) && ((y%(_row_pattern*2))<_row_pattern))
      bit_select = 7-bit_select;
//  ==== add new scan patter GAZGIZ =============== start
  if ((_scan_pattern==GAZGIZ) && ((y%(_row_pattern*2))<_row_pattern))
      bit_select = bit_select;
//  ==== add new scan patter GAZGIZ =============== end

also add patern Type in line 64
 enum scan_patterns {LINE, ZIGZAG, ZAGGIZ, WZAGZIG, VZAG, GAZGIZ};

wardi1971, 2019-10-17:
this is updated File 

[PxMatrix.zip](https://github.com/2dom/PxMatrix/files/3737353/PxMatrix.zip)

2dom, 2019-10-19:
If I read this correctly it is a ZAGGIZ pattern without the reversed bit order...hence it would be called ZAGZIG :)

Try the latest master with `display.setScanPatter(ZAGZIG) `

2dom, 2019-10-22:
Any update on this?

2dom, 2019-11-06:
No response ... Closing

closed 2019-11-06 by 2dom