2dom/PxMatrix #135: Pattern test partially works

gdombiak, 2019-10-21:
- Using latest from master commit 341d0331bccb98332097c0d95a2d00bb0a007d82 (Sun Oct 20 20:32:34 2019 +0200)
- Hardware [image]
- Adafruit-GFX-Library commit fc13f348504aea2dc7a626581a7418b959c055b3 (Fri Oct 18 14:30:21 2019 -0400)
- ESP32

I ran pattern_test and after changing display to 16 and display(64,32,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E) I see some lines appear and change color (I think this is expected). However, when it moves to another row the line seems to appear and then disappear. I then replaced displayTestPattern(70) for display(70) and no text appeared.

I then tried to run pixeltime and again nothing appeared. D and E do not go to ground based on continuity test so I hooked them up to 5 and 15 respectively. 

Here is a [video](https://youtu.be/0uEYAbLN5eQ) running pattern_test. And here are the modified examples and a high level picture of the cabling. I double and triple checked and I think I got it right.  RGB has its own 5V power supply. ESP32 powered from USB. Ground is shared between ESP and RGB panel.

```
#include <PxMatrix.h>

#ifdef ESP32

#define P_LAT 22
#define P_A 19
#define P_B 23
#define P_C 18
#define P_D 5
#define P_E 15
#define P_OE 2
hw_timer_t * timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

#endif

#ifdef ESP8266

#include <Ticker.h>
Ticker display_ticker;
#define P_LAT 16
#define P_A 5
#define P_B 4
#define P_C 15
#define P_D 12
#define P_E 0
#define P_OE 2

#endif
// Pins for LED MATRIX

//PxMATRIX display(32,16,P_LAT, P_OE,P_A,P_B,P_C);
PxMATRIX display(64,32,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E);
//PxMATRIX display(64,64,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E);

#ifdef ESP8266
// ISR for display refresh
void display_updater()
{
  display.displayTestPattern(70);
  // display.displayTestPixel(70);
  //display.display(70);
}
#endif

#ifdef ESP32
void IRAM_ATTR display_updater(){
  // Increment the counter and set the time of ISR
  portENTER_CRITICAL_ISR(&timerMux);
  display.display(70);
//  display.displayTestPattern(70);
  portEXIT_CRITICAL_ISR(&timerMux);
}
#endif

uint16_t myCYAN = display.color565(0, 255, 255);
void setup() {
  // put your setup code here, to run once:
[... 263 lines cut ...]
  {
    display.setBrightness(dimm);
    delay(5);
  }
  for (uint8_t dimm=0; dimm<255; dimm++)
  {
    display.setBrightness(dimm);
    delay(5);
  }

}
```

A [image]
A [image]
D [image]
E [image]
E [image]
STB [image]
STB [image]
Cabling [image]

Let me know if you need any other information.
Thanks,
Gaston

2dom, 2019-10-21:
Could you just leave everything as is and only revert the library back to a few days ago?

2dom, 2019-10-21:
The flickering in the pattern test ist Strange as I have not changed anything in that section

gdombiak, 2019-10-21:
Thanks for the fast response. At work now, I will run the same test with older code (Oct 18, Oct19 and Sept 1 if things still don't work) when I get back home tonight. 

After things didn't work last night, I unplugged cables from the ESP32 and used the RPi and the panel worked fine. That was one of my goals that I wanted to check before AliExpress dispute deadline expired. :) I will plug things back to the ESP32 now. Do you want me to use the ESP8266 instead or there is no difference for this test?

Thanks,
Gaston

2dom, 2019-10-21:
I also screwed up something ... The latest master should at least show something ...

gdombiak, 2019-10-21:
I guess running code from previous commits might help confirm that. Happy to run whatever test you can think ok. Or maybe it was something on my cabling ... so many of them ... :)

gdombiak, 2019-10-22:
Good news, I plugged everything back and updated library to include commit from Oct 21st and now all examples work perfect! double_buffer and pixeltime display things. I then reverted to commit of Oct 20th and both example show nothing. 

pattern_test now displays text when using display(70);. However, in all commits tested (>=  Sep 1, 2019) the line when using displayTestPattern(70) keeps flickering. No idea what is expected to happen but all other things work fine.

Let me know if you want me to run more tests or close this ticket.

Thanks,
Gaston

2dom, 2019-10-22:
Ok...great...will investigate the flicker in pattern test

closed 2019-10-22 by 2dom

2dom, 2019-10-22:
Can you check out the latest master if the flicker in pattern test is gone?

gdombiak, 2019-10-22:
Absolutely. Will check tonight and report back. Thanks for all your hard work and fast support. :)

gdombiak, 2019-10-22:
Updated to Oct 22 (commit 0312b1ef8f3c24201e941d9daf0c132cca2759e) and test still flickers. :(

I'm going to leave things plugged in so I can quickly run any tests that you might need. Let me know if you want me to debug anything. I assume you do not see this in your setup so hard to troubleshoot I guess.

Thanks,
Gaston

2dom, 2019-10-25:
Can you check out the latest master ... I think I fixed it

gdombiak, 2019-10-25:
Sure thing. Will report back results later tonight!

gdombiak, 2019-10-26:
You nailed it! There is zero flickering now and lines appear in all rows. Did a quick regression test for the other tests and they also work perfect.

Thanks so much. Great job!
Gaston

2dom, 2019-10-26:
A pleasure...will create a release today. Thanks for testing!