2dom/PxMatrix #136: MH-ET LIVE D1 mini wiring "issues" and P4 64x32 LED display

helmarw, 2019-10-26:
not per se an issue just more an information for ppl want to use MH-ET LIVE D1 mini.
Took me a couple of nights to figure it out and to get it to work. The general pin layout is the same but there is a tiny bit of information missing, the IOs are completely different to the WeMos.
Neither https://github.com/dragondaud/myClock nor the PxMatrix library https://github.com/2dom/PxMatrix are working out of the box. Even SPI interface is on different IOs. I tried to summarise my modification in the attached pdf. Nothing fancy and i had to modify PxMatrix.h but it works now. 
i use a p4 64x32 Matrix display e.g. https://de.aliexpress.com/item/32380311346.html
and those ESP32 https://de.aliexpress.com/item/32815530502.html
This display seems to have a slow muxer so i had to use setMuxDelay, which is not part of the
standard Arduino library v1.3.0 (installed form the library manager). The GitHub version v1.5.0
didnt work for me at all, so i added only the SPI and setMuxDelay-Modifications to the v1.3.0
also attached. then of course i needed to modify display.h and myclock.ino

[MH-ET Live Minikit pinout and mods for LED Matrix.pdf](https://github.com/hallard/WeMos-Matrix-Shield/files/3774625/MH-ET.Live.Minikit.pinout.and.mods.for.LED.Matrix.pdf)
[PxMatrix_mod.txt](https://github.com/hallard/WeMos-Matrix-Shield/files/3774630/PxMatrix_mod.txt)
You can use those files as reference. rename and replace PxMatrix.h with my mod, if you have
the same issue

2dom, 2019-10-26:
Could you try out the latest master of PxMatrix and check if it works for you?

2dom, 2019-10-26:
1.5 had a nasty big in it

2dom, 2019-10-26:
Just checked out your pdf...will make the SPI configurable from the sketch

2dom, 2019-10-26:
Correction ... is already exposed via begin ...

```
 // Define your display layout here, e.g. 1/8 step, and optional SPI pins begin(row_pattern, CLK, MOSI, MISO, SS)
  display.begin(8);
  //display.begin(8, 14, 13, 12, 4);
```

Updated Pixeltime example to reflect this

2dom, 2019-10-26:
So you should get it to work without any modification of PxMatrix whatsoever (latest master)

helmarw, 2019-10-26:
oki ill give it a try, but i tried v1.5.0 two days ago and it wasn't working at all not even a blip, but probably your big fix came later ?
I will set it up on an other computer and give it a try, for now my modified version is doing everything i want..
btw the Library Manager only shows 1.3.0, i don't know who is managing that but its no up to date
thnx
Helmar

helmarw, 2019-10-27:
tried it out, doesnt work fo me, nothing on the LCD
is this correct ?
display.begin(16, CLK, MOSI, MISO, SS);

2dom, 2019-10-31:
Can you try the latest master (not 1.6 via library manager)

closed 2019-11-06 by 2dom

kquinsland, 2020-11-19:
I was struggling last night to get my 64x64 matrix panel working with what i *thought* was a generic ESP32 module. After re-soldering all joints and beeping out each connection, I figured out that the `MH-ET LIVE MiniKit for ESP32` aka the esp32 version of the 'classic' WeMo D1 Mini is *not* a 'standard' module.

Using the board config / pins from [this](https://doc.riot-os.org/group__boards__esp32__mh-et-live-minikit.html) table and a continuity tester, i figured out that the correct pins to use are

```c++
#include <PxMatrix.h>

/*

ESP32 MiniKit uses non-standard pins from typical code examples
Values below are working for me w/ a 64x64 panel (2.5mm pitch)
Using the ESP8266 adapter board from Brian

Here's what i beeped out

 * PANEL  ESP 
 * R1     IO23
 * B1     N/C
 * R2     N/C
 * B2     N/C
 * A      IO22
 * C      IO5
 * CK     IO18  
 * OE     IO16
 * 
 * G1     NC
 * N      GND
 * G2     NC
 * E      IO17
 * B      IO21
 * D      IO19
 * ST     IO26
 * N      GND
 */

#define P_LAT 26
#define P_A 22
#define P_B 21
#define P_C 5
#define P_D 19
#define P_E 17
#define P_OE 16

#define P_CLK 18
#define P_MOSI 23
#define P_MISO 19
#define P_SS 5

// Pins for LED MATRIX
PxMATRIX display(64, 64, P_LAT, P_OE, P_A, P_B, P_C, P_D, P_E);

// In setup()
display.begin(32, P_CLK, P_MOSI, P_MISO, P_SS);

```

As soon as i made those changes to the WiFi Tetris code, it compiled and worked perfectly :). Now to begin adding MQTT support so i can flash messages and add a ambient light sensor to dim the panel when it's dark or turn it off all together when there's nobody in the room :)

**NOTE:** I am using the ESP8266 to HUB75 board from Brian / @witnessmenow which you can learn more about here: http://blough.ie/d1mat/

TL;DR: one of the most ubiquitous and cheapest ESP32 modules is *not* generic and requires special pin assignments as well as special `display.begin()` setup.

2dom, 2020-11-19:
Good to know - thanks for the heads up

witnessmenow, 2020-11-19:
Just an FYI, that module works with standard wiring with the esp32 version
of that matrix board

https://www.tindie.com/products/brianlough/esp32-matrix-shield-mini-32/