2dom/PxMatrix #137: Port library for atmega2560

kasperv79, 2019-10-28:
Hi. 
I got a bunch of P8 1/4 16x32 panel, which I really struggles to find a library where the pattern fits. 
Is it possible to have some help to port this library for a atmega2560 chip? I know that chip is not really that fast and lags of memory, but the panels are not to be used for any animations only  still pics.

/kasper.

the photo is from another library where the pattern does not fit. 
[image]

2dom, 2019-10-28:
Check out the latest master and run the example "pixeltime_AVR"

2dom, 2019-10-28:
You will have to attach SCK and R0 to the correct SPI pins on your board

2dom, 2019-10-28:
Let me know if it works

2dom, 2019-10-28:
I also updated the readme ...

kasperv79, 2019-10-28:
you Rock! 
at least the lower part of the display shows the scroling text perfectly. but the Pixel time text is both on the lower part of the display.  
it works with "ZAGZIG" pattern.

2dom, 2019-10-28:
Picture?

kasperv79, 2019-10-28:
picture of?   the gif shows the example. also when I use fillscreen, only the lower half works.

2dom, 2019-10-28:
Did you adapt the constructor to 32x16?

2dom, 2019-10-28:
Also check you cabling (especially the loop back)

kasperv79, 2019-10-28:
> Did you adapt the constructor to 32x16?

yes I did.

`#define PxMATRIX_COLOR_DEPTH 1
#define PxMATRIX_MAX_HEIGHT 16
#define PxMATRIX_MAX_WIDTH 32

#include <SPI.h>
#include <Adafruit_GFX.h>
#include <PxMatrix.h>
#include <avr/pgmspace.h>
// Pins for LED MATRIX
#define P_A 2
#define P_B 3
#define P_C 4
#define P_D 5
#define P_E 6
#define P_LAT 7
#define P_OE 8

PxMATRIX display(32,16,P_LAT, P_OE,P_A,P_B,P_C);`

2dom, 2019-10-29:
Can you post some more code ... Especially the initialisation?

kasperv79, 2019-10-29:
Sure. And again. THANKS!
btw. I did check the wiring again including the loop back. So unless its a broking wire it should be correct.

here's the init and setup part.

`#define PxMATRIX_COLOR_DEPTH 1
#define PxMATRIX_MAX_HEIGHT 16
#define PxMATRIX_MAX_WIDTH 32

#include <SPI.h>
#include <Adafruit_GFX.h>
#include <PxMatrix.h>
#include <avr/pgmspace.h>
// Pins for LED MATRIX
#define P_A 2
#define P_B 3
#define P_C 4
#define P_D 5
#define P_E 6
#define P_LAT 7
#define P_OE 8

PxMATRIX display(32,16,P_LAT, P_OE,P_A,P_B,P_C);
//PxMATRIX display(64,32,P_LAT, P_OE,P_A,P_B,P_C,P_D);

// Some standard colors
uint16_t myRED = display.color565(255, 0, 0);
uint16_t myGREEN = display.color565(0, 255, 0);
uint16_t myBLUE = display.color565(0, 0, 255);
uint16_t myWHITE = display.color565(255, 255, 255);
uint16_t myYELLOW = display.color565(255, 255, 0);
uint16_t myCYAN = display.color565(0, 255, 255);
uint16_t myMAGENTA = display.color565(255, 0, 255);
uint16_t myBLACK = display.color565(0, 0, 0);

unsigned long start_time=0;
uint8_t icon_index=0;

uint16_t myCOLORS[8]={myRED,myGREEN,myBLUE,myWHITE,myYELLOW,myCYAN,myMAGENTA,myBLACK};

const uint8_t weather_icons[] PROGMEM={0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x20,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x20,0x00,0x00,0xff,0xe0,0x00,0x00,0x00,0x00,0xff,0xe0,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x20,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x20,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x07,0xff,0x07,0xff,0x07,0xff,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x20,0x00,0x00,0x07,0xff,0x07,0xff,0x07,0xff,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x20,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0xff,0xe0,0xff,0xe0,0xff,0xe0,0xff,0xe0,0x00,0x00,0x00,0x20,0x00,0x00,0x00,0x00,0x00,0x00,0x07,0xff,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x20,0xff,0xe0,0xff,0xe0,0xff,0xe0,0xff,0xe0,0xff,0xe0,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0xff,0xe0,0xff,0xe0,0xff,0xe0,0xff,0xe0,0xff,0xe0,0x00,0x00,0x00,0x00,0x00,0x00
  ,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0xff,0xe0,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x20,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x07,0xff,0x07,0xff,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x07,0xdf,0x07,0xdf,0x07,0xdf,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x07,0xff,0x07,0xff,0x07,0xff,0x07,0xff,0x07,0xff,0x07,0xff,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x07,0xff,0x07,0xff,0x07,0xff,0x07,0xff,0x07,0xff,0x07,0xff,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0xff,0xe0,0xff,0xe0,0xff,0xe0,0xff,0xe0,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x20,0x00,0x20,0x07,0xff,0x07,0xff,0x07,0xff,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x07,0xff,0x00,0x00,0x00,0x00,0x07,0xff,0x07,0xff,0x07,0xdf,0x00,0x00,0x00,0x00,0x07,0xff,0x00,0x00,0x00,0x00,0xff,0xe0,0xff,0xe0,0xff,0xe0,0xff,0xe0,0xff,0xe0,0xff,0xe0,0xff,0xe0,0x00,0x00,0x00,0x00,0x00,0x00,0xff,0xe0,0xff,0xe0,0xff,0xe0,0xff,0xe0,0xff,0xe0,0xff,0xe0,0xff,0xe0,0x00,0x00,0x00,0x00
  ,0x00,0x00,0x00,0x00,0xff,0xe0,0x00,0x00,0x00,0x00,0x00,0x00,0xff,0xe0,0x00,0x00,0x00,0x00,0x00,0x20,0x00,0x00,0x00,0x00,0xff,0xe0,0x00,0x00,0xff,0xe0,0xff,0xe0,0xff,0xe0,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x07,0xff,0x07,0xdf,0x07,0xff,0x07,0xff,0x07,0xff,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x07,0xff,0x07,0xff,0x07,0xff,0x07,0xff,0x07,0xff,0x07,0xff,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x07,0xff,0x07,0xff,0x07,0xff,0x07,0xff,0x07,0xff,0x07,0xff,0x07,0xff,0x07,0xff,0x00,0x00,0x00,0x00,0x07,0xff,0x07,0xff,0x07,0xff,0x07,0xff,0x07,0xff,0x07,0xff,0x07,0xff,0x07,0xff,0x00,0x00,0x00,0x00,0x00,0x20,0x00,0x00,0xff,0xe0,0xff,0xe0,0xff,0xe0,0xff,0xe0,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x07,0xff,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x07,0xff,0x07,0xff,0x00,0x00,0x00,0x00,0x00,0x00,0x07,0xff,0x07,0xff,0x00,0x00,0x00,0x00,0xff,0xe0,0xff,0xe0,0xff,0xe0,0xff,0xe0,0xff,0xe0,0xff,0xe0,0xff,0xe0,0xff,0xe0,0xff,0xe0,0x00,0x00,0xff,0xe0,0xff,0xe0,0xff,0xe0,0xff,0xe0,0xff,0xe0,0xff,0xe0,0xff,0xe0,0xff,0xe0,0xff,0xe0,0x00,0x00
  ,0x00,0x20,0x00,0x00,0x00,0x00,0xff,0xe0,0xff,0xe0,0xff,0xe0,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x07,0xff,0x07,0xff,0x07,0xff,0xff,0xe0,0xff,0xe0,0xff,0xe0,0x00,0x00,0xff,0xe0,0x00,0x00,0x07,0xff,0x07,0xff,0x07,0xdf,0x07,0xff,0x07,0xdf,0x07,0xff,0x07,0xff,0x00,0x00,0x00,0x00,0x07,0xff,0x07,0xff,0x07,0xff,0x07,0xff,0x07,0xff,0x07,0xff,0x07,0xff,0x07,0xff,0x00,0x00,0x00,0x00,0x07,0xff,0x07,0xff,0x07,0xff,0x07,0xff,0x07,0xdf,0x07,0xff,0x07,0xff,0x07,0xff,0x07,0xff,0x00,0x00,0x07,0xff,0x07,0xff,0x07,0xff,0x07,0xff,0x07,0xdf,0x07,0xff,0x07,0xff,0x07,0xff,0x07,0xff,0x00,0x00,0x00,0x00,0x00,0x00,0xff,0xe0,0xff,0xe0,0xff,0xe0,0xff,0xe0,0x00,0x00,0x00,0x20,0x00,0x00,0x00,0x00,0x00,0x00,0x07,0xff,0x00,0x00,0x07,0xff,0x07,0xff,0x07,0xff,0x00,0x00,0x07,0xff,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0xff,0xe0,0xff,0xe0,0xff,0xe0,0xff,0xe0,0xff,0xe0,0xff,0xe0,0xff,0xe0,0xff,0xe0,0xff,0xe0,0x00,0x00,0xff,0xe0,0xf [... cut ...]
 #define GPIO_REG_SET(val) (val < 8) ? PORTD |= _BV(val) : **PORTB** |= _BV(val-8)
  #define GPIO_REG_CLEAR(val) (val < 8) ? PORTD &= ~_BV(val) : PORTD &= ~_BV(val-8)
#endif
`
should it be D?

and also the part where Hardware SPI pins are defined, they match the atmega 328:
`//HW SPI PINS
#define SPI_BUS_CLK 14
#define SPI_BUS_MOSI 13
#define SPI_BUS_MISO 12
#define SPI_BUS_SS 4`

should it match the 2556 pins?
`#define SPI_BUS_CLK 52
#define SPI_BUS_MOSI 51
#define SPI_BUS_MISO 50
#define SPI_BUS_SS 53`

/Kasper

2dom, 2019-10-30:
well spotted ... However These things don't take effect in this case

2dom, 2019-10-30:
Can you check your D wire ... It seems to be stuck on high

kasperv79, 2019-10-31:
Well. The D wire is not present on the board and the GND which is on the "D" location is grounded. What I have understood about the board is that the D is not used.?

2dom, 2019-10-31:
OK ... can you run the pattern test and post a gif/video of the result?

2dom, 2019-11-02:
There might be a bug in the offset calculation - can you try the latest master and give feedback?

Chsldz, 2019-11-03:
I was watching this issue, because i experienced the same issues. Just the lower half worked. 
With the latest master, it now works as expected with my Panel. Full Screen is used.
Great Job!

2dom, 2019-11-03:
Awesome - thanks! If you have 32x16 (or run the atmega2560 you could also try increasing the color depth a little bit. E.g. 

```
#define PxMATRIX_COLOR_DEPTH 2
#define PxMATRIX_MAX_HEIGHT 16
#define PxMATRIX_MAX_WIDTH 32
```

Could you try gradually increasing the color depth and give feed-back if it works?

Thanks!

Chsldz, 2019-11-03:
Hey, i ran the Pixeltime example, with diffrent color depths. Hope i was looking for the right differences. Where the red, green, and blue bars are on the right side. I was able to see diffrent brightness steps in each color depending on the color depth. 
With depth 1 were was only on or off, and with each depth increase, it increased the steps between on and off. As i got the depth increased to 14 the display started flickering so i stopped there.
So as far as i understand it, it works :)

2dom, 2019-11-03:
Great! Very happy that it works with Atmel now as well. Thanks for the feed-back! :sunglasses:

kasperv79, 2019-11-04:
Hi Guys.
Thanks so much for the effort! 
works great now!!

kasperv79, 2019-11-04:
Hi Again.
the colors seems to be swabbed at display. myRED = blue, myBLUE = RED, myGREEN = GREEN.

Is this related to this library?

2dom, 2019-11-05:
The latest master allows you to set the Color order. You could try

`display.setColorOrder(BBRRGG);`

or other combinations. Don't know anything about Fonts - sorry

kasperv79, 2019-11-05:
Again. Thanks for all your work. 
I'll close this Issue now as solved :O)

closed 2019-11-05 by kasperv79