2dom/PxMatrix #138: Same code is not working with the lib ver 1.6.0

jorgecis, 2019-10-29:
I have a panel  P3 63x32 from amazon, the code example works fine with my board if I use the version 1.4.0 of your library but doesn't work with the version 1.6.0

  With the version, 1.4.0 works with some flicking issues, but works, with the version 1.6.0 nothing, all LEDs are off. The same code just the lib is different.

2dom, 2019-10-29:
Ok...thanks for the feedback. Could you post your code

2dom, 2019-10-29:
Really ... 1.6 does not work but the current master does?

amcolash, 2019-10-31:
Similar here, 1.5 works great with code but 1.6 does not. Have not tested master yet... I was not even able to fully use the test program in 1.6. The test pattern and pixel works, but the "Pixel" text didn't show and I had a blank display (using `display.display(70)`).

More context for me: I am using an esp8266 (adafruit huzzah), arduino 1.8.9, 64x32 led panel. Happy to provide more info if needed.

2dom, 2019-10-31:
Please check the master and give feedback if it works ...

jorgecis, 2019-10-31:
Yes that is correct, 1.6 is not working but 1.5, 1.4 or master works.

  Using esp8266 mini d1, with plataformio

2dom, 2019-10-31:
OK thanks ... good to know ... will investigate  issue 137 and then push another release

amcolash, 2019-11-01:
Yeah looks like `master` is ok for me too

hulud, 2019-11-02:
I was unable to use the library and tried to debug it for many hours. It seems that only the test patters work properly but nothing else. I am not 100% sure, but I believe the PxMATRIX_buffer does not get written at all. I printed it and saw zeroes at the beggining where there should be non-dark pixels.

hulud, 2019-11-02:
ok, it seems that the current master is good, but somehow the version available on platformio is broken :(

2dom, 2019-11-03:
Ok ... noted ... new release soon

closed 2019-11-03 by 2dom