2dom/PxMatrix #139: Problem with P3 64x32 desame as issue 119

tobobus, 2019-10-29:
I received a new panel from sryled with the chips as decribed in issue 119.
I have read issue 119  multiple times and understand the upgrading the px matrix should be the solution. 
What i did , going from px matrix 1.3 .0 to 1.6 .0 unfortunate i dont see than no pixels at all.

With a older panel and px matrix 1.3.0 no problems.

Would you please give me some suggestions what iam doing wrong or misunderstood.
Unfortunate iam a newbe of this sort of problems.

regards Tobo.

2dom, 2019-10-29:
Can you try the latest master?

tobobus, 2019-10-29:
Is that the one from 9 days ago in issue 119

tobobus, 2019-10-29:
https://github.com/2dom/PxMatrix/blob/master/PxMatrix.h

tobobus, 2019-10-29:
sorry from the above, i used the file from your , see below.
Unfortunate still 4lines, nothing , 4lines etc.
Hope i used the right one !!!!!!!

 Popular repositories

    PxMatrix

    Adafruit GFX compatible graphics driver for LED matrix panels

tobobus, 2019-10-29:
some further info, if a take a look it displays version 1.6.0 installed.
There is  some change as before with 1.6.0 i got no pixels at all and now the 4 lines etc.

Sorry if i have been a little chaotic.

tobobus, 2019-10-30:
After some extra reading of issue 119 and issue 104, i put in  display.setMuxDelay(1,1,0,0,0); 
This is the solution, it Works ( for this panel anyway.
Thanks for your work and time.

2dom, 2019-10-30:
Great....Are you running the latest master or 1.6?

tobobus, 2019-10-30:
Iam running 1.6.0 with shows up after i uploaded from your repositories, so not via the scetch upgrade.
Where can i find the latest master and what is the name of it? Iam happy to test that too.

tobobus, 2019-10-30:
Your statement: I added an option to control the delay in every stage of the multiplexing. 
That would be very handy, but do you have to do that in the .H file ?

2dom, 2019-10-30:
This is precisely what display.setMuxDelay(1,1,0,0,0) does

2dom, 2019-10-30:
If you are downloading from github you are running the latest version of the master branch aka "master"

tobobus, 2019-10-30:
and that is 1.6.0?? is that correct ?

2dom, 2019-10-31:
Nope...1.6 is a few commits ago...you can see the different releases if you click master then tags on the github site

tobobus, 2019-10-31:
If i follow that i got 1.6.0 . do i something wrong?? whit a last change of 3 day ago( today 31-10)

2dom, 2019-10-31:
Correcy

tobobus, 2019-10-31:
Thanks, iam going to test now the clock from Hari and all the another with the 8266 and new panel with last px matrix.
For now Big Thank You

tobobus, 2019-11-01:
unfortunate all sketches from the clocks with esp8266 gives the 4 pixel, 4 nothig etc problem with the new panel.
Testing all with the latest px matrix gives no pixels at all.

2dom, 2019-11-06:
Did you add delay via setMuxDelay?

tobobus, 2019-11-07:
Tested with IDE 1.8.10. Latest libraries except Json 5.13.3, Ntp clientlib 2.5.1. Used mux delay from 1.1.1.0.0 till 5.5.5.0.0 and everething in between.
 I used Morphing RGB Hari Clock and get ,see picture , with Px 1.3.0 and nothing of pixels with 1.7.0 +mux-delay.
All the ESP 8266 scetches gives the same result except one. See below
The picture is from the original Hari-clock .

[image]

Now the strange part of it, The zip-file is from "SnowHead"whom modified the Hari-file, please open it and you see in the ïnstructions"the Hari history and futher Snowheads story and sketch.

https://github.com/xSnowHeadx/HariFun_166_Morphing_Clock is his web-site.
Now is the funny part of it that the Snowhead sketch is working excellent with Px1.7.0 and mux-delay 1.1.1.0.0 as the only one !!!!!!!!!
This goes beyond my knowledge !

[HariFun_166_Morphing_Clock-master(3).zip](https://github.com/2dom/PxMatrix/files/3821469/HariFun_166_Morphing_Clock-master.3.zip)

tobobus, 2019-11-19:
Via the "discussions" from the site of Morphing Digital Clock by HariFun i tested a suggestion from "timz3818 " to disable  "#define double_buffer " and voila this works.
No black panel-screen after PX 1.7.0  and ESP 8266.
It works for Hari's clock the remix and further variations.

I have no ideer what double buffer does in those scetches, but disabling works somehow.

chonglar, 2019-11-23:
Wow @tobobus thank you so much for that comment, you helped me make a troubleshooting breakthrough!

Last problem: My 64x32 panel is only showing red pixels, but at least it's readable now and not just black! 👍 

Anyone have ideas for fixing a "red pixels only" problem?  I'm going to recheck my wiring again...

I was using the PxMatrix pixeltime example, and I had uncommented this line:
// #define PxMATRIX_double_buffer false

Details of my setup and what I've tried below: 

Hardware:
  LED matrix is the 64x32 with FM6124 chips.  It says p3-6432-2121-16s-d1.0 on the back.  
  The one from AZERONE on Amazon https://www.amazon.com/gp/product/B079JSKF21

  ESP8266 is "IZOKEE D1 Mini NodeMcu ... Base on ESP8266 ESP-12F for Arduino, 100% Compatible with WeMos D1 Mini"
  From Amazon: https://www.amazon.com/gp/product/B076F53B6S
  The Wifi chip says ESP8266MOD on it.

Arduino IDE: Tools->Board = "LOLOIN (WEMOS) D1 R2 & mini" -- I got the blinking sketch to make the little blue LED blink on and off, so I 'think' I've got the right board configured, but not sure if some other board definition matches it better...
    
Lines I changed in the pixeltime example to make it work:

#define matrix_width 64 
#define matrix_height 32
PxMATRIX display(64,32,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E);
display.begin(16); // instead of 8
display.setDriverChip(FM6124);

I am actually not sure if I needed to call setDriverChip or not ... I've tried with and without the setDriverChip call and it seems to work either way.  Perhaps once the panel receives the init sequence it will work on subsequent reloads?

Some things I've tried tweaking upside down and backwards but seem to have no effect
// display.setMuxDelay(1,1,1,1,1); // No effect
// display.setColorOrder(BBGGRR); // No effect
// display.setFastUpdate(true);

Some things that I verified only make it worse:
- setScanPattern(ZIGZAG) caused everything to be scrambled, so I figured it's unrelated
- setMuxPattern(STRAIGHT) caused garbled output so I don't think that's the right path either
- changing CPU Frequency to 160Mhz didn't do anything about the red color

chonglar, 2019-11-23:
Wow @tobobus thank you so much for that comment, you helped me make a troubleshooting breakthrough!

Last problem: My 64x32 panel is only showing red pixels, but at least it's readable now and not just black! 👍

Anyone have ideas for fixing a "red pixels only" problem? I'm going to recheck my wiring again...

I was using the PxMatrix pixeltime example, and I had uncommented this line:
// #define PxMATRIX_double_buffer false

Details of my setup and what I've tried below:

Hardware:
LED matrix is the 64x32 with FM6124 chips. It says p3-6432-2121-16s-d1.0 on the back.
The one from AZERONE on Amazon https://www.amazon.com/gp/product/B079JSKF21

ESP8266 is "IZOKEE D1 Mini NodeMcu ... Base on ESP8266 ESP-12F for Arduino, 100% Compatible with WeMos D1 Mini"
From Amazon: https://www.amazon.com/gp/product/B076F53B6S
The Wifi chip says ESP8266MOD on it.

Arduino IDE: Tools->Board = "LOLOIN (WEMOS) D1 R2 & mini" -- I got the blinking sketch to make the little blue LED blink on and off, so I 'think' I've got the right board configured, but not sure if some other board definition matches it better...

Lines I changed in the pixeltime example to make it work:

#define matrix_width 64
#define matrix_height 32
PxMATRIX display(64,32,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E);
display.begin(16); // instead of 8
display.setDriverChip(FM6124);

I am actually not sure if I needed to call setDriverChip or not ... I've tried with and without the setDriverChip call and it seems to work either way. Perhaps once the panel receives the init sequence it will work on subsequent reloads?

Some things I've tried tweaking upside down and backwards but seem to have no effect
// display.setMuxDelay(1,1,1,1,1); // No effect
// display.setColorOrder(BBGGRR); // No effect
// display.setFastUpdate(true);

Some things that I verified only make it worse:

setScanPattern(ZIGZAG) caused everything to be scrambled, so I figured it's unrelated
setMuxPattern(STRAIGHT) caused garbled output so I don't think that's the right path either
changing CPU Frequency to 160Mhz didn't do anything about the red color

tobobus, 2019-11-24:
@chonglar , glad to hear it is working now.
I wonder one thing, you were using the matrix with chips FM 6124 and that is what we call the old panel.This should has work even with Px matrix 1.3.0, Json 5.13.3, board manager2.5.0 and NTP ClientLib 2.05.

The problems with most of us have where with the panels with chips ICM 2037 BP.
Eventualy thanks to 2DOM  Px1.7.0 Hari's clock works if you disable "double buffer"
About the power supply i use 0ne with max 2.5 A. Also the Tetris clock is ESP32 so be carefull with the pinout.

chonglar, 2019-11-25:
@tobobus , thanks!  Yes, I think I have the old panel.  Just to double check that the double_buffer thing was really the cause of the blackness (since I tried so many things and I end up forgetting sometimes) I recompiled and uploaded it with the double_buffer on and it just gives a blank screen.  Here's the back of my board -- most of the chips are the FM6124 and the other thinner chips say SM5166PC on them.  

https://imgur.com/a/NHX4xJq

I can't find the letters ICM 2037 BP anywhere on my chips, but I haven't pulled out a magnifying glass to see the tiniest.

I did manage to get MorphClock to work great -- but I haven't done the Tetris clock yet because I haven't gotten my ESP32s in the mail yet.  

This Reddit post made me decide to get an LED matrix to play with, and now that I got it working, I'm trying to determine if I can really push pixels to it in the kind of color & refresh rate that was shown in that post.  I have some ESP32 boards coming today so I can see if the performance is a lot better than the ESP8266.

https://www.reddit.com/r/arduino/comments/duk0ej/a_64x32_matrix_with_2048_leds_driven_at_72fps_by/

The color quality seems pretty bad on mine -- I display my little avatar icon on the screen ( this guy, from the Wardner arcade game https://piskel-imgstore-b.appspot.com/img/ce0e7085-0ed7-11ea-be1f-b3c1e72105d4.gif ) but it looks more like a 4 color image (tan, white, brown, blue, and red) ... none of the shading seems to be there in the arms and hair.  I'm guessing it's because I have the flicker problem still and I haven't gotten that quite sorted out.  When I film it with my phone camera, I get a horrible kind of flashing color effect where the whole sprite changes from one bright color to the next... 

Is there another repository of cool demos like Aurora that work with PxMatrix?  I don't have a teensy, so I can't try driving my LED matrix with the regular SmartMatrix library but it looks like when I get the ESP32 I can give it a try since there's a port for SmartMatrix to the ESP32.

https://www.youtube.com/watch?v=WrSl1rnEOdA

tobobus, 2019-11-25:
@chonglar , try this one its the most colorfull one i have and the colors are realy bright.

https://github.com/kolle86/MorphingClockRemixRemix

chonglar, 2019-11-26:
Thanks @tobobus I'll try that one -- I just got the ESP32 and I ran my sprite code on it and NO flicker at all (I did have flicker at first, but then tried changing #define PxMATRIX_COLOR_DEPTH from 16 to 8 and then all the flicker was gone.  I tried changing it to 12 and no flicker still, but it seems like the color still isn't working great.  

I wonder now if the ESP8266 flicker was because of that define?  But I don't want to reconnect all the pins to the 8266 again now to find out -- the ESP32 seems to be working really well.  Maybe the connections are just better, but I hardly see any ghosting on the top pixels at all now.

tobobus, 2019-11-26:
Hmmm, thats a difficult one, i can not answer that , because my knowlegde is not enough .
For my boards and panels i have never had any problems with flicker. 
I don't have a teensy either but a will take a look at it.
Mostly flicker and goosting is from long cable, bad contacts and so on, but a asume that you know that.

Picolo12, 2020-03-02:
hello everyone i also have the problem with black display
compile is successful. double buffer on off also tested
I use the matrix    P4   Drive IC: ICN2037 ICN2038S SMD16126 MBI5124 MBI5153   https://de.aliexpress.com/item/33049102240.html?spm=a2g0s.9042311.0.0.27424c4dmxamB4.

Please help me, I really want the watch.

tobobus, 2020-03-03:
@Picolo12 , The difference is possible the P4 where i used a P3.
What sketch are you using ?
Did you follow all the issues with the problems like 
and Tested with IDE 1.8.10. Latest libraries except Json 5.13.3, Ntp clientlib 2.5.1. Used mux delay from 1.1.1.0.0 till 5.5.5.0.0 and everething in between.
I used Morphing RGB Hari Clock and get ,see picture , with Px 1.3.0 and nothing of pixels with 1.7.0 +mux-delay.
All the ESP 8266 scetches gives the same result except one. See below
The picture is from the original Hari-clock . See above

closed 2020-03-28 by 2dom