2dom/PxMatrix #141: 1/4 Scan invalid position

59160922, 2019-10-31:
I've problem,I purchased a P10 1/4 scan matrix.  I've tried combinations of ZAGZIG and BINARY can display character
but display only Lower half and Invalid position.

https://drive.google.com/open?id=12d_1gO5CRDWSMkX7psPAS3ix97VdmEDL
display.setCursor(0,0);

kasperv79, 2019-10-31:
that is the qexact same problem I have with mine P8 display, see issue 138

59160922, 2019-11-01:
I tried replacing GND with D wire and change 
PxMATRIX display(32,16,P_LAT, P_OE,P_A,P_B,P_C);
to 
PxMATRIX display(32,16,P_LAT, P_OE,P_A,P_B,P_C,P_D);
 but result are the same as before.
How do you solve this problem? Please advise me.

2dom, 2019-11-01:
Can you run the pattern test and post the resulting gif/video?

59160922, 2019-11-02:
The pattern test
https://drive.google.com/file/d/1P4KZV3cphD3iDa34n-U25CHkHZ89nSRm/view

2dom, 2019-11-02:
There might be a bug in the offset calculation - can you try the latest master and give feedback?

2dom, 2019-11-03:
Seems to work now ... check an re-open if still present

closed 2019-11-03 by 2dom

59160922, 2019-11-06:
Thank you very much

onlineeuropean, 2020-03-24:
I am having the exactly same problem with P8 screen 1/4 scan. Have tried all scan patterns during pattern test but getting the same result:
https://drive.google.com/file/d/1P4KZV3cphD3iDa34n-U25CHkHZ89nSRm/view

I have downloaded your code yesterday so it should be the newest one.

onlineeuropean, 2020-03-24:
#include <PxMatrix.h>

#ifdef ESP32

#define P_LAT 22
#define P_A 19
#define P_B 23
#define P_C 18
#define P_D 5
#define P_E 15
#define P_OE 2
hw_timer_t * timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

#endif

#ifdef ESP8266

#include <Ticker.h>
Ticker display_ticker;
#define P_LAT 16
#define P_A 5
#define P_B 4
#define P_C 15
#define P_D 12
#define P_E 0
#define P_OE 2

#endif
// Pins for LED MATRIX

PxMATRIX display(40,20,P_LAT, P_OE,P_A,P_B,P_C);
//PxMATRIX display(40,20,P_LAT, P_OE,P_A,P_B,P_C,P_D);
//PxMATRIX display(40,20,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E);

#ifdef ESP8266
// ISR for display refresh
void display_updater()
{
  display.displayTestPattern(70);
  // display.displayTestPixel(70);
  //display.display(70);
}
#endif

#ifdef ESP32
void IRAM_ATTR display_updater(){
  // Increment the counter and set the time of ISR
  portENTER_CRITICAL_ISR(&timerMux);
  //isplay.display(70);
  display.displayTestPattern(70);
  portEXIT_CRITICAL_ISR(&timerMux);
}
#endif

uint16_t myCYAN = display.color565(0, 255, 255);
void setup() {
  // put your setup code here, to run once:
Serial.begin(9600);
  display.begin(4);
  display.setScanPattern(ZAGZIG);
  display.flushDisplay();
  display.setTextColor(myCYAN);
  display.setCursor(2,0);
  display.print("Pixel");
  Serial.println("hello");

  #ifdef ESP8266
    display_ticker.attach(0.002, display_updater);
  #endif

  #ifdef ESP32
    timer = timerBegin(0, 80, true);
    timerAttachInterrupt(timer, &display_updater, true);
    timerAlarmWrite(timer, 2000, true);
    timerAlarmEnable(timer);
  #endif

  delay(1000);
}

void loop() {

 delay(100);

}

2dom, 2020-03-24:
Die you pull the latest Master from GitHub?

onlineeuropean, 2020-03-24:
Yes, if the latest file is 5 months old. I started using your library this month, so do not have any old one.

onlineeuropean, 2020-03-26:
> Die you pull the latest Master from GitHub?

Any chance you can help me?

2dom, 2020-03-28:
@onlineeuropean can you post a picture of the problem ... pattern test seems fine

onlineeuropean, 2020-03-31:
#define double_buffer
#include <PxMatrix.h>

#ifdef ESP32

#define P_LAT 22
#define P_A 19
#define P_B 23
#define P_C 18
#define P_D 5
#define P_E 15
#define P_OE 2
hw_timer_t * timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

#endif

#ifdef ESP8266

#include <Ticker.h>
Ticker display_ticker;
#define P_LAT 16
#define P_A 5
#define P_B 4
#define P_C 15
#define P_D 12
#define P_E 0
#define P_OE 2

#endif
// Pins for LED MATRIX

//PxMATRIX display(32,16,P_LAT, P_OE,P_A,P_B,P_C);
PxMATRIX display(40,20,P_LAT, P_OE,P_A,P_B,P_C,P_D);
//PxMATRIX display(64,64,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E);

#ifdef ESP8266
// ISR for display refresh
void display_updater()
{
  display.display(70);
}
#endif

#ifdef ESP32
void IRAM_ATTR display_updater(){
  // Increment the counter and set the time of ISR
  portENTER_CRITICAL_ISR(&timerMux);
  display.display(70);
  portEXIT_CRITICAL_ISR(&timerMux);
}
#endif

struct Text {
	char *text;
	uint16_t width, height;
	int16_t x, y;
	int16_t dx, dy;
} text = {"1", 0, 0, 0, 0, 1, 1};

uint16_t textColor = display.color565(0, 0, 255);
uint16_t myBLACK = display.color565(0, 0, 0);
uint16_t lineColor = display.color565(255, 0, 0);
uint16_t backgroundColor = display.color565(0, 255, 0);

void setup() {
  // put your setup code here, to run once:
Serial.begin(9600);
  display.begin(4);
  display.flushDisplay();
  display.setTextWrap(false);

  #ifdef ESP8266
    display_ticker.attach(0.002, display_updater);
  #endif

  #ifdef ESP32
    timer = timerBegin(0, 80, true);
    timerAttachInterrupt(timer, &display_updater, true);
    timerAlarmWrite(timer, 2000, true);
    timerAlarmEnable(timer);
  #endif

  int16_t x1 = 0, y1 = 0;
  display.getTextBounds(text.text, 0, 0, &x1, &y1, &text.width, &text.height);
  text.width-=2;
  text.height-=2;
}

int16_t x=0, dx=1;

void loop() {
//  display.clearDisplay();
  display.fillScreen(myBLACK);

  if(x+dx>=display.width() || x+dx<0)
  	dx=-dx;
  x+=dx;
  display.drawLine(x,0, display.width()-x-1, display.height()-1, lineColor);

  if(text.x+text.dx+text.width>=display.width() || text.x+text.dx<0)
  	text.dx=-text.dx;
  if(text.y+text.dy+text.height>=display.height() || text.y+text.dy<0)
  	text.dy=-text.dy;
  text.x+=text.dx;
  text.y+=text.dy;
  display.setTextColor(textColor);
  display.setCursor(text.x, text.y);
  display.print(text.text);

  display.showBuffer();
  delay(20);
}

onlineeuropean, 2020-03-31:
Video of double buffer
https://youtu.be/TFr-ONNwOm4

onlineeuropean, 2020-03-31:
And something better with zaggiz scan.
https://youtu.be/QcgtJjT8DQc

onlineeuropean, 2020-04-01:
As I understood led panel is driven by this https://datasheet.lcsc.com/szlcsc/1809031521_MBI-MBI5120GM-B_C261129.pdf

onlineeuropean, 2020-04-01:
Any chance to modify and make it working? Would have a lot of creative time making some examples during quarantine period :)

onlineeuropean, 2020-04-07:
The wiring is column 75. Seems the same as in your example
[image]

onlineeuropean, 2020-04-14:
Hi! Can I expect your comment?

2dom, 2020-04-14:
Sorry man ... Very busy atm and no time into supporting new chip sets right now. Will look into it as soon as I can

onlineeuropean, 2020-05-04:
> Sorry man ... Very busy atm and no time into supporting new chip sets right now. Will look into it as soon as I can

Hi! Any way to update it?