2dom/PxMatrix #144: Double pixels on AB panel

jaroslavas, 2019-11-03:
I have a panel which I cannot get working properly. The panel has only A and B connection, other two pins are not connected (NC).
[image]

After trying all the combinations of settings, I've came up with something that almost works, but it glows two pixels instead of one. When I set pixel in any even row, the next pixel on odd row lights up too (for example if I turn on pixel at (0, 0), two pixels turn on: (0, 0) and (0, 1) ) and vice versa, when I turn on pixel in odd row, it lights it and the pixel over it in even row. I've made a simple program that iterates all pixels one by one. You can see the result in [this video](https://youtu.be/LcPrwdmJIMg).

Parts of code:
```
display.begin(2);
display.setScanPattern(ZAGGIZ);
display.setMuxPattern(BINARY);
display.setDriverChip(SHIFT);

...

for (int y = 0; y < 16; y++) {
    for (int x = 0; x < 32; x++) {
      Serial.print(x);
      Serial.print(";");
      Serial.println(y);
      display.clearDisplay();
      display.drawPixel(x, y, myYELLOW);
      display.showBuffer();
      delay(100);
    }    
  }

```

[Here you can see](https://youtu.be/3nG2Z0C0U18) your example test running with above settings.

I'm using the code from master that I've got an hour ago. 
ESP32 controller (DevC).

Any help setting it up or tips what to try, are appreciated.

xsrf, 2020-04-13:
The test pattern shows that when line 0 is activated (A and B low on BINARY), both lines are lit.
When line 1 is activated (A high, B low), only odd lines are lit.
When line 2 is activated (A low, B high), only even lines are lit.
-> You have to use `setMuxPattern(STRAIGHT)`

snow-leopard-climbing, 2020-06-05:
@jaroslavas, hello. Please tell me if you could make the panel work only with A, B pins using Arduino. I have the same problem and cannot find a solution for 2 months ...

jaroslavas, 2020-06-05:
First of all I'm using esp32, but to make it work I've switched to another library: https://github.com/pixelmatix/SmartMatrix