2dom/PxMatrix #146: No issue, just question about wiring.

kasperv79, 2019-11-06:
Hi.

About wiring Input and output together:

from the description: 
"When driving a long chain of LED modules in a row, parallel color data lines make a lot of sense since it reduces the data rate. But since we are only driving a single module here, we really don't need that. We can therefore use jumper wires between input connector (PI) and output connector (PO) to chain all shift registers together and create one big shift register. This has two advantages: it reduces the number of required GPIO pins on the microcontroller and we can use the hardware SPI interface to drive it."

How big a change is it to have this library to work without chaining input to output and use the extra io pins that is required?  For me it would make sense to use the ribbon cable directly to a shield.

2dom, 2019-11-06:
Would be a complete rewrite...there are libraries out there that do this already...

kasperv79, 2019-11-06:
> Would be a complete rewrite...there are libraries out there that do this already...

Yes there’s other lib. But only with yours lib I got the right pattern after you added support for AVR.

And thanks for that :)

closed 2019-11-07 by kasperv79