2dom/PxMatrix #151: Working with more than 4 panels

SimaoRenato, 2019-11-19:
Does anyone have used this lib with more than 3 panels?

I'm using a p10 32x16 4row scan/ zzagg pattern / straight mux.

The pattern scan works fine, but pixel time, or even drawing a single white pixel is not working.

If anyone did it before, I appreciate the help.

metingul, 2019-11-28:
I work with 6 panels (192X16) with older version of PxMatrix library, but current version does not work properly with 4 panels, the RED color is fine but other colors does not work properly.

SimaoRenato, 2019-11-28:
Whitch version are you using?

SimaoRenato, 2019-11-28:
Sorry, saw the version rigth after I've hit Comment.

I'll try 1.3.0 to see if works.

for now I'm developing my own library, but not with all resources as PxMatrix has.

2dom, 2019-11-29:
We discovered that chaining multiple panels may lead to pixel color shift due to the propagation delay of all the registers in the loop-back signal path

2dom, 2019-11-29:
Do you have this problem https://twitter.com/MLE_Online/status/1192265470117695488?s=19 ?

EvilArtist, 2019-12-16:
> We discovered that chaining multiple panels may lead to pixel color shift due to the propagation delay of all the registers in the loop-back signal path

I have the same problem! Did you have any solution?
Could you share to me. 

Thanks.

SimaoRenato, 2019-12-16:
I did have to build my own library for the project I did.  It is not as complete as the PxMatrix, but I was able to work with 12 panels in row.

vinzzzzz24, 2020-02-12:
> I did have to build my own library for the project I did. It is not as complete as the PxMatrix, but I was able to work with 12 panels in row.

Can you share your solved problem here please :( thanks

closed 2020-02-19 by 2dom