2dom/PxMatrix #152: 64*32 P5 panel missing lines and printig some line several times

Porokoura, 2019-11-19:
Hello!
I can't get my LED panel to work as it should. I did the pattern test you see in the video: https://www.youtube.com/watch?v=a8-nj3kZqoM&t=10s
I also run Pixel time:
 https: //www.youtube.com/watch? V = UvT3FkPRUh4
Chips on the board are: TC5020A, RUC7258 and 74HC245C.
My panel is this: 
https: //www.aliexpress.com/item/32850389208.html? Spm = a2g0s.9042311.0.0.50074c4dVlgPqw
I'm using library version 1.7.0
Micro controller is ESP 32 DOIT kit V1
The E-pin of the Hub75 connector is labeled GND and connected to the same point as other GND pins. I use a level shifter, without the panel it won't work at all.
The panel works with Arduino Mega, but there are problems with ESP32.
The settings for display are:
#define matrix_width 64
#define matrix_height 32
PxMATRIX display(64,32,P_LAT, P_OE,P_A,P_B,P_C,P_D);
display.begin(16);

What could be done about this problem?

2dom, 2019-11-19:
Looks like your A pin connection might be bad ?

Porokoura, 2019-11-21:
I tested the contacts continuity meter, no problems.
I tried with another similar panel, same result. I changed the panel to a 64 * 32 P3 panel with a different chipset (ICN 2037) and this worked properly.

2dom, 2019-11-22:
Did you try adding some mux delay?

Porokoura, 2019-11-22:
I added a mux delay. I tried values of 1-5 microseconds. I set a delay for all channels. Without successTo make sure there is no problem with I / O I defined the different I / O for A-pin, no effect.

Porokoura, 2020-02-01:
Hello! I solved the problem by ordering a panel where RUC 7258 has been replaced by ICN 2012. Such a change to the panel was made to me without compensation at https://szlightalldisplay.aliexpress.com/
I also own one 64 * 32 P3 and one 64 * 64 P3 panel. These work without level shifters.64 * 32 My P5 panels do not. The level shifter selection also seems to be significant. Simple and cheap do not work, use an integrated circuit model.For example: https: //www.digikey.com/catalog/en/partgroup/8-channel-bi-directional-logic-level-converter/49463
Greetings,Porokoura.

DanielRTRD, 2020-02-03:
@Porokoura how did you wire it up with the level shifter?

Porokoura, 2020-02-04:
> @Porokoura how did you wire it up with the level shifter?
Hi!
A level shifter is placed between the esp32 and the led panel input connector. The Esp32 is connected to the low voltage terminals and the display to the high voltage terminals. When using 64 * 32 panels you need an 8 channel unit. Hope the picture helps.
https://images.app.goo.gl/s4VMeWqAFTn1YR2o9

2dom, 2020-02-19:
Glad it works

closed 2020-02-19 by 2dom