2dom/PxMatrix #155: What is wrong?

HiveParking, 2019-11-27:
// This is how many color levels the display shows - the more the slower the update
//#define PxMATRIX_COLOR_DEPTH 8

// Defines the buffer height / the maximum height of the matrix
#define PxMATRIX_MAX_HEIGHT 32

// Defines the buffer width / the maximum width of the matrix
#define PxMATRIX_MAX_WIDTH 32

// Defines how long we display things by default
//#define PxMATRIX_DEFAULT_SHOWTIME 30

// Defines the speed of the SPI bus (reducing this may help if you experience noisy images)
//#define PxMATRIX_SPI_FREQEUNCY 20000000

// Creates a second buffer for backround drawing (doubles the required RAM)
//#define PxMATRIX_double_buffer true

#include <PxMatrix.h>

// Pins for LED MATRIX
#ifdef ESP32

#define P_LAT 22
#define P_A 19
#define P_B 23
#define P_C 18
#define P_D 5
#define P_E 15
#define P_OE 2
hw_timer_t * timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

#endif

#ifdef ESP8266

#include <Ticker.h>
Ticker display_ticker;
#define P_LAT 16
#define P_A 5
#define P_B 4
#define P_C 15
#define P_D 12
#define P_E 0
#define P_OE 2

#endif

#define matrix_width 32
#define matrix_height 32

// This defines the 'on' time of the display is us. The larger this number,
// the brighter the display. If too large the ESP will crash
uint8_t display_draw_time=20; //10-50 is usually fine

//PxMATRIX display(32,32,P_LAT, P_OE,P_A,P_B,P_C);
PxMATRIX display(32,32,P_LAT, P_OE,P_A,P_B,P_C,P_D);
//PxMATRIX display(64,64,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E);

// Some standard colors
uint16_t myRED = display.color565(255, 0, 0);
uint16_t myGREEN = display.color565(0, 255, 0);
uint16_t myBLUE = display.color565(0, 0, 255);
uint16_t myWHITE = display.color565(255, 255, 255);
uint16_t myYELLOW = display.color565(255, 255, 0);
uint16_t myCYAN = display.color565(0, 255, 255);
uint16_t myMAGENTA = display.color565(255, 0, 255);
uint16_t myBLACK = display.color565(0, 0, 0);

[... 174 lines cut ...]
  {
    display.drawLine(xx+16,0,xx+16,5,display.color565(xx*16,0,0));
    display.drawLine(xx+16,6,xx+16,10,display.color565(0,xx*16,0));
    display.drawLine(xx+16,11,xx+16,15,display.color565(0,0,xx*16));
  }
  delay(1000);
  for (uint8_t dimm=255; dimm>0; dimm--)
  {
    display.setBrightness(dimm);
    delay(5);
  }
  for (uint8_t dimm=0; dimm<255; dimm++)
  {
    display.setBrightness(dimm);
    delay(5);
  }

}

[image]

[image]

 latest PxMatrix 
 esp8266

yefeng0, 2019-11-27:
> // This is how many color levels the display shows - the more the slower the update
> //#define PxMATRIX_COLOR_DEPTH 8
>
> // Defines the buffer height / the maximum height of the matrix
> #define [...]

for (int xx=0; xx<16;xx++)
{
display.drawLine(xx+16,0,xx+16,5,display.color565(xx16,0,0));
display.drawLine(xx+16,6,xx+16,10,display.color565(0,xx16,0));
display.drawLine(xx+16,11,xx+16,15,display.color565(0,0,xx*16));
}
 this area miss the "*" behind xx

HiveParking, 2019-11-27:
Hi, thanks for the help, It turns out I needed to connect ground for it to work. 

Can you help figuring out how to center the gif? thanks
[image]

closed 2019-11-29 by HiveParking