2dom/PxMatrix #157: issues with p2.5 64x64 160mm panels with ICN2038S drivers HUB74E connections

Jay-esp, 2019-11-28:
Having a couple of issues.
Connected to esp32 dev1 doit.
Pinout for reference according to supplier.
[image]

When setting the driver to anything other than SHIFT I get this, parts are dim, red lines. This is fixed when I power cycle the display without powering down the esp so it must be an initialization issue.

[image]

Now if I change the driver type to SHIFT, I don't have that issue but another one, vertical is offset and for example line 31 appears on line 30 and line 31 is on 33.

[image]

Any idea?
Other point, the esp crashes very often and reboots
./home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1446 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x400890ef on core 1

Using esp32 plus rtc module and wifi webserver.

xsrf, 2020-04-13:
Guess it's the same issue as with https://github.com/2dom/PxMatrix/issues/101
Using `SHIFT` should be correct... Can you try with `setFastUpdate(false)`?

Jay-esp, 2020-04-22:
have not worked on this project for some months, will check once I take it on again, I suspect the issue might be that I don't use level shifters and hi-lo logic levels are marginal, I solved some of the issues by reducing the voltage to the lcd panel

Nikitok122, 2020-05-17:
Hi!! I have the same panel, but cant make it to work. The issue is doubling the screen. can u share your code from this example??

Nikitok122, 2020-05-17:
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/img/text-ghosting.jpg about your problem

Jay-esp, 2020-05-20:
Hi, I ended up using other libraries but had similar issues, solved for now by reducing the 5V voltage to the panel a little bit to get the 3.3V signal from the esp in range of proper 0 and 1 detection by the panel, you might want to recheck the wiring first

closed 2020-06-12 by 2dom