2dom/PxMatrix #158: RE: Ghosting Issue

jagadeeshpinter, 2019-11-30:
I 'm using NOde (ESP-12E) module  and arudino Program..I have Ghosting Issue on LED display

Below code reference..

`#include <Ticker.h>
#include <PxMatrix.h>
#include <string.h>

#ifdef ESP8266

Ticker display_ticker;
#define P_LAT 16          /*Define A,B,C,LATCH,OUTPUT ENABLE PINS*/
#define P_A 5
#define P_B 4
#define P_C 12 
#define P_OE 15

#endif

PxMATRIX display(64,32,P_LAT, P_OE,P_A,P_B,P_C);      /*Create Matrix Display Object*/
#ifdef ESP8266
void display_updater()                                /*Display refresh rate - do not change*/
{
  display.display(50);
}
#endif

void display_update_enable(bool is_enable)
{

#ifdef ESP8266                                        /*Display latch time - do not change*/
  if (is_enable)
    display_ticker.attach(0.001, display_updater);
  else
    display_ticker.detach();
#endif
}
//uint16_t Red = display.color565(0, 0, 255);           /*Display Color Definitions*/
uint16_t Red = display.color565(255, 0, 0);
uint16_t Green = display.color565(0, 255, 0);
uint16_t Yellow = display.color565(0, 69,255);
//uint16_t Blue = display.color565(255, 0, 0);
uint16_t Blue = display.color565(0, 0, 255); 
uint16_t Black = display.color565(0, 0, 0);
uint16_t White = display.color565(255,255,255);

void COLOR(char c1){                                 /*Display color switcher function*/
 if(c1=='0')display.setTextColor(Red);
 else if(c1=='1')display.setTextColor(Green);
 else if(c1=='2')display.setTextColor(Yellow);
 else display.setTextColor(Blue);
  }

int xi;short int garb=7;                             /*Operating variables*/
int ti;int key=1;int saavi=0;int thiravu=0;
int itr=0;int buff_ptr=0;
char r1i,r1[25],r2[25],r3[25],c1[25],c2[25],c3[25],buff[300];
int sthas,endhas;
char welcm[9]=" Welcome";

//#################################CLEAR BUFFER#####################################
void clear_buffer(){
        for(int i=0;i<300;i++)buff[i]=0;
        r1i=0;
        for(int i=0;i<25;i++)r1[i]=0;
        for(int i=0;i<25;i++)c1[i]=0;
        for(int i=0;i<25;i++)r2[i]=0;
        for(int i=0;i<25;i++)c2[i]=0;
        for(int i=0;i<25;i++)r3[i]=0;
        for(int i=0;i<25;i++)c3[i]=0;
[... 304 lines cut ...]

void loop() {
  if(digitalRead(10)==LOW)
  {
    get_uart();
    saavi=1;
    thiravu=1;
  }
  else
  {
    if(saavi==1)
    {
      sort_uart();
      while(Serial.available())
        Serial.read();
      saavi=0;
      }
    if(thiravu==1){
      generic_print(r1i,r1,r2,r3,c1,c2,c3);
      clear_buffer();
      thiravu=0;
      itr=0; 
    }
    }
}`

2dom, 2019-11-30:
Did you red the readme and tried all remedies for ghosting?

jagadeeshpinter, 2019-12-06:
Yes we tried ground un used pins,but still problem is there.We checked with multimeter ,it carries .33amp on some led blinks.

stephenkamin, 2020-03-30:
My ghosting issue was wrong setting for....
display.begin(8);
{4,8,16,32} is the scanning layout options.

2dom, 2020-06-12:
OK ... closing

closed 2020-06-12 by 2dom