2dom/PxMatrix #164: Connect 2 16x32 matrix

theaddies, 2020-01-01:
I am terribly sorry to bother you with this question.  First of all, thanks very much for your wonderful work.

I am trying to connect 2 16x32 displays side by side.  The display functions but the image I get is off in a weird way.  It is like the top of the one matrix is a duplicate of the bottom of the other.  This means that they don’t overlap and it looks like gibberish.

I have been working with the Pixel_time_example sketch to get things to work.  The sketch works fine on a single 32x16 display and things only go awry when I connect the second.  I have changed the settings you outlined on github.
#include <PxMatrix.h>
PxMATRIX display(96,16,...);
...
[]
...
void setup(){
display.begin(4);
display.setPanelsWidth(3);

The confusing thing for me is, I think, how to wire up the display.  In the github example you say that we can chain multiple displays together by simply connecting the ribbon cable and make the software change above.  What I don’t understand is that in your wiring diagram for the single display you have connected the jumper wires from PI to PO.

Connect PI and PO as follows:
PI            PO
R2           R1
G1          R2
G2          G1
B1           G2
B2           B1

If I were to make those connections then I cannot use the ribbon cable obviously.  Attached is a picture of my arrangement where I have wired up Brian Lough’s hat to the board.  I soldered jumper wires between the PI and PO connections for both boards as you indicated.  When I connect up the ribbon cable I get the display shown in the second picture.  I am using an ESP32 but with Brian’s ESP8266 adapter.  I went in to the libraries and changed the pins around to make it work.  As I said everything is fine for the single 32x16 display.

I would be most grateful for any insight you might be able to provide.
Thanks
Andrew
[image]
[image]
[image]

Porokoura, 2020-01-04:
Hi,theaddies! I would try following: Switch "PxMATRIX display(96,16,...);" to PxMATRIX display(64,16,...);
And also "display.setPanelsWidth(3);" to display.setPanelsWidth(2);

theaddies, 2020-01-12:
I have done all of that.  I didn't blindly copy the settings provided.

`#define matrix_width 32
#define matrix_height 16

// This defines the 'on' time of the display is us. The larger this number,
// the brighter the display. If too large the ESP will crash
uint8_t display_draw_time=20; //10-50 is usually fine

PxMATRIX display(64,16,P_LAT, P_OE,P_A,P_B,P_C);
...........
display.begin(8);
  //display.begin(8, 14, 13, 12, 4);
  // Set the number of panels that make up the display area width (default is 1)
  display.setPanelsWidth(2);
`

2dom, 2020-01-18:
if you connect multiple panels you loop back from the last one.
Also, 

#define matrix_width 64
#define matrix_height 16

theaddies, 2020-01-18:
Thanks very much for the reply.  What do you mean loop back from the last one?  Is this wiring?

2dom, 2020-01-18:
Yes...

theaddies, 2020-01-18:
Sorry I don't understand.  I only have 2 panels side by side, 2 16 x 32 panels.  I don't understand where the wiring would loop back to?

closed 2020-03-28 by 2dom

theaddies, 2020-03-28:
Why close?  I still want an answer.  I still want to figure this out.

 

Closed #164 <https://github.com/2dom/PxMatrix/issues/164> .