2dom/PxMatrix #166: P10 chaning 2x (32x16) dosen't work

thbl, 2020-01-09:
When trying to chain 2 displays, it looks strange....
[image]
```

#include <Arduino.h>
#include <PxMatrix.h>

#include <Ticker.h>
Ticker display_ticker;
#define P_LAT 16
#define P_A 5
#define P_B 4
#define P_C 15
#define P_D 12
#define P_E 0
#define P_OE 2

#define matrix_width 64
#define matrix_height 16

// This defines the 'on' time of the display is us. The larger this number,
// the brighter the display. If too large the ESP will crash
uint8_t display_draw_time=5; //10-50 is usually fine
//PxMATRIX display(matrix_width,matrix_height,P_LAT, P_OE,P_A,P_B,P_C,P_D);
PxMATRIX display(matrix_width,matrix_height,P_LAT, P_OE,P_A,P_B,P_C);
//PxMATRIX display(matrix_width,matrix_height,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E);

void display_updater()
{
  display.display(display_draw_time);
}

void display_update_enable(bool is_enable)
{

    display_ticker.attach(0.002, display_updater);

}

void setup() {
 Serial.begin(9600);
  // Define your display layout here, e.g. 1/8 step
  display.begin(2);
  display.setPanelsWidth(2);
  // Define your scan pattern here {LINE, ZIGZAG, ZAGGIZ, WZAGZIG, VZAG} (default is LINE)
  display.setScanPattern(ZAGGIZ);
  //display.setMuxDelay(1,1,1,1,1);
  //display.setMuxPattern(STRAIGHT); 
  display.setFastUpdate(true);
  display_update_enable(true);
  delay(3000);

}

void loop() {
    display.setBrightness(255);
    display.setTextSize(1);     
    display.setTextColor(display.color565(0, 255, 0));
  for (int i = 0; i <= 4; i++) {
      display.clearDisplay();
      display.setCursor(0,2);
      display.println(1600+i);
      delay(2000);
  }

}
```

With a signle display it works fine, what could be the issue ?

2dom, 2020-01-18:
You could try running the Pattern Test and see what happens?

theaddies, 2020-01-19:
I am having exactly the same issue with strange things displaying.  I ran the pattern_test_io program.  It seems to be working on a 32x16 until I get to blue.  Pixels in rows 1 and then 9 light one at a time from right to left in red.  It then repeats in green.  When it comes to blue the display increments to row 2, shows one blue pixel, it then turns to red and the pattern continues over the entire display.  Code is below.  I would really appreciate the help.  I am trying to get 2 32x16 displays to work side by side, but I figure if one doesn't work I have no chance.  I have tried looking through the code, but it is beyond my capabilities.  I am using the D1 mini matrix breakout.  I am using PxMatrix32 because I redefined the pins to accommodate an ESP32 instead of ESP8266.
My pin definitions in PxMatrix32.h are below and then the codes.
`// HW SPI PINS
#define SPI_BUS_CLK 18
#define SPI_BUS_MOSI 23
#define SPI_BUS_MISO 19
#define SPI_BUS_SS 21`

`#include <PxMatrix32.h>

#ifdef ESP32

#define P_LAT 26
#define P_A 22
#define P_B 21
#define P_C 5
//#define P_D 5
//#define P_E 15
#define P_OE 16

//#define P_LAT 22
//#define P_A 19
//#define P_B 23
//#define P_C 18
//#define P_D 5
//#define P_E 15
//#define P_OE 2
hw_timer_t * timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

#endif

#ifdef ESP8266

#include <Ticker.h>
Ticker display_ticker;
#define P_LAT 26
#define P_A 22
#define P_B 21
#define P_C 5
//#define P_D 12
//#define P_E 0
#define P_OE 16

#endif
// Pins for LED MATRIX

#define matrix_width 32
#define matrix_height 16

PxMATRIX display(64,16,P_LAT, P_OE,P_A,P_B,P_C);
//PxMATRIX display(64,32,P_LAT, P_OE,P_A,P_B,P_C,P_D);
//PxMATRIX display(64,64,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E);

#ifdef ESP8266
// ISR for display refresh
void display_updater()
{
//  display.displayTestPattern(70);
  display.displayTestPixel(70);
  //display.display(70);
}
#endif

#ifdef ESP32
void IRAM_ATTR display_updater(){
  // Increment the counter and set the time of ISR
  portENTER_CRITICAL_ISR(&timerMux);
  //isplay.display(70);
//  display.displayTestPattern(70);
  display.displayTestPixel(70);
  portEXIT_CRITICAL_ISR(&timerMux);
}
#endif

uint16_t myCYAN = display.color565(0, 255, 255);
void setup() {
  // put your setup code here, to run once:
Serial.begin(9600);
  display.begin(8);
   //display.begin(4, 8, 16, 32);

  // Define multiplex implemention here {BINARY, STRAIGHT} (default is BINARY)
  display.setMuxPattern(BINARY);  //TRIED BINARY LINE AND 8

  // Set the multiplex pattern {LINE, ZIGZAG,ZZAGG, ZAGGIZ, WZAGZIG, VZAG, ZAGZIG} (default is LINE)
  //display.setScanPattern(LINE);

    // Rotate display
  //display.setRotate(true);

  // Flip display
  //display.setFlip(true);

  // Helps to reduce display update latency on larger displays
  //display.setFastUpdate(true);

  // Control the minimum color values that result in an active pixel
  //display.setColorOffset(5, 5,5);

  // Set the multiplex implemention {BINARY, STRAIGHT} (default is BINARY)
  //display.setMuxPattern(BINARY);

  // Set the color order {RRGGBB, RRBBGG, GGRRBB, GGBBRR, BBRRGG, BBGGRR} (default is RRGGBB)
  //display.setColorOrder(RRGGBB);

  // Set the time in microseconds that we pause after selecting each mux channel
  // (May help if some rows are missing / the mux chip is too slow)
  //display.setMuxDelay(0,1,0,0,0);

  // Set the number of panels that make up the display area width (default is 1)
  display.setPanelsWidth(1);

  // Set the brightness of the panels (default is 255)
  //display.setBrightness(50);

  // Set driver chip type
  //display.setDriverChip(FM6124);
  
  display.flushDisplay();
  display.setTextColor(myCYAN);
  display.setCursor(2,0);
  display.print("Pixel");
  Serial.println("hello");

  #ifdef ESP8266
    display_ticker.attach(0.002, display_updater);
  #endif

  #ifdef ESP32
    timer = timerBegin(0, 80, true);
    timerAttachInterrupt(timer, &display_updater, true);
    timerAlarmWrite(timer, 2000, true);
    timerAlarmEnable(timer);
  #endif

  delay(1000);
}
void loop() {

 delay(50);

}`

stephenkamin, 2020-03-30:
theaddies,
Your math is wrong here...

#define matrix_width 32     ---S/B 64? or match the 32 below???
#define matrix_height 16
PxMATRIX display(64,16,P_LAT, P_OE,P_A,P_B,P_C);
                                 ^^
and are the boards chained together?
you should have your board plugged into the in on the #1 board
there should be a jumper from the out of #1 to the in on #2 
THEN the out from your board plugs into the out on #2 
[image]

2dom, 2020-06-12:
Thanks everybody ... closing this due to OP inactivity

closed 2020-06-12 by 2dom