2dom/PxMatrix #169: Half of the P3 RGB matrix is yellow

Nurees, 2020-01-28:
Hi,

I created a board for the P3 RGB matrix with an ESP8266F module. In the test with a finished ESP board everything worked. 
with my own board is the half of the matrix yellow. What could be the reason for this ? 
 Greetings

Nurees, 2020-01-28:
Addendum: It is not the complete half of the matrix yellow, but only what I display. If I write e.g. hello on the matrix, only half of the text is yellow, the rest is normal.

2dom, 2020-01-29:
Well..pictures, code and schematics might will help to figure outs what's going on

Nurees, 2020-02-01:
> Well..pictures, code and schematics might will help to figure outs what's going on

Hi,

Thanks for the answer. Sorry, I did not expect an answer so quickly. Yes More information would be helpful of course sorry...

I took two pictures and attached the schematic.  What you see is yellow (not very visible on the picture) but should be white. What I noticed is that the color blue is nowhere to be found. That's probably why the yellow is there. 

I must have done something wrong with the schematic.

The code is on my laptop but I forgot it at work so I can't give it to you until Monday.
[image]
[image]
[image]

Nurees, 2020-02-02:
If you still want the code I will comment it in English, unless you can speak German :D

2dom, 2020-02-19:
Could you try the pattern test an post the result?