2dom/PxMatrix #170: 32x32 P6 pattern_test results

nygma2004, 2020-02-02:
First of all thanks for this code. I have purchased a 32x32 P6 modul with 1/8 scan from Ali. I am running the pattern_test using the following definition: PxMATRIX display(32,32,P_LAT, P_OE,P_A,P_B,P_C,P_D);
I checked and I have not E pin, it is connected to ground, so I am only using A to D. PxMatrix is 1.7.0. Wemos D1 mini is set to 160MHz.
My test pattern first comes blue, cyan and white. Draws like in your video, Line 9, Line 1, and Line 16 and Line 24. And after white it starts again with one line down.

Did I wire the loop connections incorrectly that I am getting wrong colors? I don't know if it makes a difference, both on the board, it is labelled as HUB75D, not just HUB75.

2dom, 2020-02-02:
Try playing with setColorOrder (see header file)

nygma2004, 2020-02-02:
> Try playing with setColorOrder (see header file)

You mean in the PxMatrix.h, right? I can't find any such function there.

2dom, 2020-02-02:
You will have to pull the latest master

nygma2004, 2020-02-02:
And it works :) Thanks a lot. I am so happy. It looks like most people go for rectangular displays not square like me.
Let me leave the details here, in case somebody else also uses this display. I got mine from [Aliexpress](https://www.aliexpress.com/item/32658820147.html?spm=a2g0s.9042311.0.0.59944c4dQiQpUc).
I have followed [this Instructable](https://www.instructables.com/id/RGB-LED-Matrix-With-an-ESP8266/) for wiring, with the exception that line E does not need to be connected. That is the yellow wire between the ESP and the display.
I used the following in the definition:
`PxMATRIX display(32,32,P_LAT, P_OE,P_A,P_B,P_C,P_D);`
And the following lines in setup:
`  display.begin(8);`
`  display.setFastUpdate(true);`
`  display.setScanPattern(WZAGZIG);`
`  display.setColorOrder(BBGGRR);`
I still have a very faint ghosting under and right to the pixel being drawn, but will address that as described in the readme.
I hope others also have fun which these P6 displays.

2dom, 2020-02-19:
Glad that it work :) Happy coding!

closed 2020-02-19 by 2dom

arahasya, 2020-05-17:
As the panel is 1/8 you don't need the D line also

A,B, C does the job