2dom/PxMatrix #175: Ghost :-( Yes I read previous issues, after many try, no way...

pbenaim, 2020-02-10:
I have some bloody ghosts :-( and yes I read issues of others guys...

I have one P3 64x32
That I done:
I use Wemos D1 R1, butI tried with one NodeMCU too ( like in differents tutorials )
I changed my power supply ( Now : 5V / 10A, it's vey large...)
I tried some combinasons of display.setMuxDelay(x....)
I tried too, add capacitors missing, but it's same... Grrrr

To reduce ghosts, I reduced the brightness with setBrightness(4) but theys are always  here :-(

And I have ghosts, in particular in green.

I asked to the manufacturer, and the answer is : 
_We don't have a way to solve the problem because dont know what controler you use. If you use  Linsn brand, Novastar, Huidu, Xixun etc , you can adjust the refresh rate and the scan clock on the software._

So, I tried, again, some combinasons of display.setMuxDelay(....)

Now, I don't where looking for, maybe order an other one new display...

But before, if somebody have, some very goods, ideas, it's welcome before buy another one device...

Thanks to the all community :-)

2dom, 2020-02-10:
Did you check out the readme and tried everything it says regarding ghosting ?

pbenaim, 2020-02-10:
> Did you check out the readme and tried everything it says regarding ghosting ?

Yes, many times, and on many forums too...

awyork, 2020-02-22:
Have you tried to slow down your SPI bus?  At 20Mhz you will start to see issues regarding issues with loops, crosstalk, ect.  Try what solved the issue here: https://github.com/2dom/PxMatrix/issues/84

The issue there was either bad contacts in the bread board or signal wires being too far away from their ground wires.  Sorry can't remember the correct terms right now, but you want your ground wires as close to you signal wires as possible.

pbenaim, 2020-02-23:
Already tried (10 Mhz / 5 Mhz ) :-) Same Issues :-(

For information I have this set to drive the panel:
https://www.electrodragon.com/product/rgb-dot-matrix-display-drive-wifi-board-esp8266/

So I just order one level converter to drive the panel in 5 Volts level:
https://fr.aliexpress.com/item/33015881084.html

I will give you one feedback when is done...

awyork, 2020-02-23:
I was able to see on the output side (of the panel) clock pin, that 3v3 was being read successfully.  I was getting out a 5v clock signal on the output pin.

 

https://www.electrodragon.com/product/rgb-dot-matrix-display-drive-wifi-board-esp8266/

https://fr.aliexpress.com/item/33015881084.html

awyork, 2020-02-24:
FYI.... 

I opened #177 because setting the frequency does not seem to work.  I set it to 5 but still got 20 on my scope.

closed 2020-06-12 by 2dom