2dom/PxMatrix #177: pixletime example has bad #define

awyork, 2020-02-24:
https://github.com/2dom/PxMatrix/blob/master/examples/pixeltime/pixeltime.ino

Line 14:
//#define PxMATRIX_SPI_FREQEUNCY 20000000

Should be:
//#define PxMATRIX_SPI_FREQUENCY 20000000
  
"E" and "U" are transposed in the word frequency

awyork, 2020-02-24:
Also, setting the frequency does not seem to work.

I changed line 14 in pixeltime as below but I still see 20Mhz on the 'scope.

// Defines the speed of the SPI bus (reducing this may help if you experience noisy images)
#define PxMATRIX_SPI_FREQUENCY 5000000

2dom, 2020-03-28:
OK .. fixed the example. Still strange that it does not effect the actual speed. Any ideas?

awyork, 2020-03-30:
> OK .. fixed the example. Still strange that it does not effect the actual speed. Any ideas?

Sorry, I accidentally fried my eval board.  I've been waiting on a shipment from China.  I'll get back to you.

2dom, 2020-06-12:
OK ... re-open when the issue persists...

closed 2020-06-12 by 2dom