2dom/PxMatrix #178: Cant get red blue and green to line up

slowsauce, 2020-02-25:
Im using pixel time example and getting wrong colors. The red blue and green channel of rbg are split apart. Ex white will have a pixel on as red then 2 over green then 2 over blue. 
[image]

## Rules
1. Post your code 
2. Post a picture of the problem and describe what you expect to see
3. Run the pattern test and post the result as gif/video (need to have a bit more than one full RED/YELLOW/WHITE cycle)
4. State what version of PxMatrix you are running and what MicroController you use

2dom, 2020-02-26:
Code?

slowsauce, 2020-02-26:
I posted this from phone and am not near computer. It's the example code pixeltime with screen size configured as 64x32 with d and e connected though in testing d seems to not change anything. Also after watching some videos on these types of display I feel I may have 2 duds because they seem to only work at 3.3v not 5. At 5v the blue and green led stop putting out light.

2dom, 2020-02-26:
You could try setting the horizontal resolution to 65 ...

slowsauce, 2020-03-01:
found problem, it was a faulty cable between output of esp and display, the cable wires were super high resistance on only some pins, i had 3 of the cables and each had issue.

closed 2020-03-01 by slowsauce