2dom/PxMatrix #180: Using Arduino Uno as a Microcontroller

wazosky, 2020-03-10:
Hello, i'm a newbie of Arduino's world, i found this library and tutorial form youtube. I'm confused what am i going to do with this code. I'm using Arduino uno as my microcontroller. I have a project for my final project. I was using Example>PxMatrix LED Matrix Library>pattern_test and error. How to make this code to be using Arduino Uno? 
[pattern_test.txt](https://github.com/2dom/PxMatrix/files/4312768/pattern_test.txt)

2dom, 2020-03-10:
Sorry...have you read the Readme of PxMatrix? Also ... You may want to start with something a bit more simple if you are new to Arduino

wazosky, 2020-03-18:
Sorry i'm just reply this issue, yeah i have read Readme,i have successful to turn on this panel, and now i have a problem about, how to adjust the brightness of this panel? i have read another article, is just 0-7 number for the reds for example? sorry for my bad english.

2dom, 2020-03-28:
brightness is 0-255

wazosky, 2020-04-07:
okay, thanks for your help dom..

closed 2020-04-07 by wazosky