2dom/PxMatrix #182: Can't download code when panel is connected

lg007, 2020-03-16:
Hello,

First of all, thanks for this wonderful library.

I am using a Wemos Lolin32 Lite board to drive a P5 32 x 64 matrix. After several misunderstandings, I have got it it to work, everything is fine. But I am not able to download code if the panel is connected.   

```
Connecting........_____....._____....._____....._____....._____....._____.....____An error occurred while uploading the sketch
_

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
```
If I switch off the 5V power supply of the matrix, or if I disconnect the connector to the matrix, the code downloads perfectly ("E" line is not connected, because it is GND on the matrix).  I know that I am missing something trivial, but I don't know what...

Any ideas?

Thanks,

slowsauce, 2020-03-19:
i get same problem on ESP8266, i have to depower the panel to allow it to upload.
im suspecting a hardware issue with the panel and the required pin selections for booting / flashing

2dom, 2020-03-28:
Just an idea ... entering programming mode on the ESP8266 requires 
GPIO 2 HIGH
GPIO 0 HIGH 
GPIO 15 LOW

maybe your particular matrix has some internal pull-up/pull-down resistor. I would try disconnecting those pins one after the other and check if programming works. If it does swap some pins where it does not matter if the panel pulls them low or high.

slowsauce, 2020-03-28:
I figured as much. I actually had to swap d line on 15 to another pin as the d line on panel would be high when the panel was powered. Which actually fixed another problem where if the panel was powered and the esp crashed it got stuck in a boot loop.

2dom, 2020-03-28:
So it works ? Close issued?

lg007, 2020-03-28:
Yes, thanks to all of you!

closed 2020-03-28 by lg007