2dom/PxMatrix #185: Twinkling pixels

pyzimmer, 2020-03-27:
Hello,

There is some twinkling pixels on my panel : 
https://photos.app.goo.gl/ECdXQgufkL1dPgGm8
 
The less text displayed, the more flickering pixels there are...

When I move the ground cable away (see photo) there are fewer pixels, when I move it closer to another cable, there are more :
[image]

I try with another power supply, there are less flickering pixels.

I try to add capacitors 10pF, 1nF, , 10nF,  1uF, 10uF, 1000 uF, no way...

Thank you

2dom, 2020-03-28:
You could try reducing the SPI speed in the library. Place

PxMATRIX_SPI_FREQUENCY 10000000

before #include <PxMatrix.h>

pyzimmer, 2020-03-29:
Thanks, I Try with 15 000 000 and 12 000 000, it seems not better. At 10 000 000 EPS32 crash because of resolution (I think)

pyzimmer, 2020-04-03:
I found a solution to reduce the issue : group all wire and keep all away from the ground wire.
Before :
[image]

After :
[image]

closed 2020-04-03 by pyzimmer

2dom, 2020-04-04:
Cool...impressive that it works with 3 panels

xsrf, 2020-04-13:
> The less text displayed, the more flickering pixels there are...

This strongly indicates that it's an issue with the 3.3V levels driving 5V logic.
When more pixels are lit, more current is drawn and your supply voltage may drop slightly... 
Lowering the supply voltage is and should be the proper solution, you actually don't need 5V. If you can, go down to ~4V. (or buy proper and fast level converters)
If you have a power supply with variable voltage, you can easily confirm this. My panel also starts twinkling at 5.05V when only few pixels are lit with an SPI frequency of 16MHz.

pyzimmer, 2020-04-15:
Thank you for your expertise. You are right : I have 2 power supply of fixed voltage 5v. But one of them have thinner wire who warm up, then the voltage is about 4.5v and I get less twinkling pixels.

pyzimmer, 2025-03-14:
Hello, I found another cause: the output-input link cables. To solve the problem, you need to add a ground cable between each signal cable on the ribbon. And connect these cables to ground at each end on the HUB75 socket. eg:

PI | PO
-- | --
R1 | R0
GND | GND
G0 | R1
GND | GND
G1 | G0
GND | GND
B0 | G1
GND | GND
B1 | B0