2dom/PxMatrix #186: Multiple Examples Core 1 Panic (IntegerDivideByZero) at higher pixel counts (ESP32)

culbeda, 2020-04-13:
When attempting to run several of the examples on a 128x64 panel, I receive something like the following:  (This one from LED-Matrix-Double_Buffer)

Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d0f29  PS      : 0x00060530  A0      : 0x800d0fe5  A1      : 0x3ffb1e70  
A2      : 0x3ffc0150  A3      : 0x0000007c  A4      : 0x00000011  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x0000000f  A8      : 0x3ffc6250  A9      : 0x00000020  
A10     : 0x00000000  A11     : 0x3ffc6150  A12     : 0x000006bf  A13     : 0x3ffc61c4  
A14     : 0x00000610  A15     : 0x00000000  SAR     : 0x0000001c  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400d22c0  LEND    : 0x400d2322  LCOUNT  : 0x00000002  

Only changes to the stock code are:
PxMATRIX display(128,64,P_LAT, P_OE,P_A,P_B,P_C,P_D,P_E);
(Or without P_E)
display.begin(32);
(same with display.begin(16)

The examples I've been able to get to work without completely crashing is pixeltime_AVR.j  (pixeltime displays something, but it's garbled and eventually reboots).  All others go into a panic loop almost immediately.

anderscallertun, 2020-04-17:
There seems to be a hard coded limit of max 64x64 pixels, so you have to do a
`#define PxMATRIX_MAX_WIDTH 128`
before the
`#include "PxMatrix.h"`
to make it work with a 128x64 panel.

culbeda, 2020-04-21:
Thanks,

I added the following to the top of the Double-Buffer and Mario sketches to match PixelTime AVR:

#define PxMATRIX_COLOR_DEPTH 1
#define PxMATRIX_MAX_HEIGHT 64
#define PxMATRIX_MAX_WIDTH 128

Mario worked, but it confirmed what I've suspected, that my blue and green are swapped.  Still trying to figure out how to fix that.

Double Buffer is still PANICing with the new lines.

2dom, 2020-04-21:
Swapping colors you can do with setColorOrder in the latest master

closed 2020-06-12 by 2dom