2dom/PxMatrix #188: Display corruption on P4 (16S 32x64) Display - pixeltime

alexberry, 2020-04-19:
# Display corruption on P4 (16S 32x64) Display - pixeltime

_The body of this ticket is taken from [markdown in my repo](https://github.com/alexberry/trainboard/blob/master/bugreport/README.md)_

Hi there. I've been having somewhat of an issue with using PxMatrix with a P4 display. I have got the point where I can reliably cause the code to work on a 1/8 Scan 16x32 display, but if I move the cables to the larger display and just reconfigure the code for 32x64 it simply doesn't work properly.

## Videos

### Pixeltime

I have recorded the resultant issue and have uploaded it to youtube [here](https://www.youtube.com/watch?v=kKVjqLSDeao). In the video I reset the esp immediately and then wait until it scrolls to the icon / colour-block test.

### Pattern_test

I have also recorded the patter_test and uploaded it [here](https://www.youtube.com/watch?v=oXgAmYN9Rmg). From the video, things look ok, however I have noticed that it is not symmetrical - there are 7 rows below the bottom line and 8 rows above the top line, suggesting an alignment issue perhaps?

## Code differences

Below are the diffs of the two files I used when switching from 8S to 16S:

```C++
50,51c50,51
< #define matrix_width 32
< #define matrix_height 16
---
> #define matrix_width 64
> #define matrix_height 32
57,58c57,58
< PxMATRIX display(32,16,P_LAT, P_OE,P_A,P_B,P_C);
< //PxMATRIX display(64,32,P_LAT, P_OE,P_A,P_B,P_C,P_D);
---
> //PxMATRIX display(32,16,P_LAT, P_OE,P_A,P_B,P_C);
> PxMATRIX display(64,32,P_LAT, P_OE,P_A,P_B,P_C,P_D);
135c135
<   display.begin(8);
---
>   display.begin(16);
```

## Source

And here they are on github:

### Pixeltime
[1/8 Scan 16x32 pixeltime](https://github.com/alexberry/trainboard/blob/master/examples/pixeltime-32x64/pixeltime-32x64.ino)

[1/16 Scan 32x64 pixeltime](https://github.com/alexberry/trainboard/blob/master/examples/pixeltime-32x64/pixeltime-32x64.ino)

### Pattern_test
[1/16 Scan 16x32 pattern_test](https://github.com/alexberry/trainboard/blob/master/examples/pattern_test-16x32/pattern_test-16x32.ino)

[1/16 Scan 32x64 pattern_test](https://github.com/alexberry/trainboard/blob/master/examples/pattern_test-32x64/pattern_test-32x64.ino)

## Library Versions

Here's a list of all libraries & software I've been using, _note I have tried both master and the latest tag for PxMatrix_:

Library | Version
--- | ---
PXMatrix (2nd attempt)| [No tag - f1fa1bf](https://github.com/2dom/PxMatrix/commit/f1fa1bfce4ccf04c6a086f48208e602e0d4a01e4)
PXMatrix (1st attempt) | [v1.7.0](https://github.com/2dom/PxMatrix/tree/v1.7.0)
Ticker | Built-in / 1.0.0
ESP8266 Board Manager | [2.6.3](https://github.com/esp8266/Arduino/releases/tag/2.6.3)
Arduino IDE | 1.8.5

## Wiring

I made sure wiring was identical, with the addition of connecting D for 1/16 Scan rate. I also attempted to connect not-in-use E to ground on arduino, to no good effect. The display has been powered by both a USB -> Barrel power supply and a dedicated 10A 5v power supply, neither solution made a difference. _All cables used on 8S were re-used on 16S, including cross-patches between IN and OUT_.

I have made my own wiring table for this, viewable [here](https://github.com/alexberry/trainboard#wiring-guide)

Also, here's a photo:

[image].

## Hardware

This has been tested with 3 different ESP8266 chips and 2 identical RGB panels, all present the same issue.

## Troubleshooting

I have made efforts to follow [your troubleshooting guide](https://github.com/2dom/PxMatrix#troubleshooting) and have double-checked:

* Grounding unused A,B,C,D,E pins (namely E in this instance)
* Testing different multiplex patterns (LINE, ZIGZAG,ZZAGG, ZAGGIZ, WZAGZIG, VZAG, ZAGZIG)
* Disabling setFastUpdate
* Attempting to slow it down with setMuxDelay
* Testing alternative drivers with setDriverChip
* Continuity tested all cabling
* Tested using patterntest.ino.
* Pared-down to a simple [helloworld](https://github.com/alexberry/trainboard/blob/master/examples/helloworld-32x64/helloworld-32x64.ino) version of the code, confirmed as working at 8S and still not at 16S

All to no avail. Any assistance would be greatly appreciated, I'm at a loss as to where to go next to resolve this.

Thanks,

Alex

2dom, 2020-04-19:
It looks like the muxing ist not working properly. So setMuxDelay would be the a good start. What values did you try?

alexberry, 2020-04-19:
I tried setting 0,0,0,0 1,1,1,1 10,10,10,10 and 100,100,100,100 (for lack of a better understanding of what these mean) but they seemed to have no effect.

alexberry, 2020-04-19:
To be doubly sure, I retried the following settings (not all at once, editing the same line and then recompling each time):
```c++
display.setMuxDelay(0,1,0,0,0);
display.setMuxDelay(0,10,0,0,0);
display.setMuxDelay(0,100,0,0,0);
display.setMuxDelay(0,0,0,0,0);
display.setMuxDelay(1,1,1,1,1);
display.setMuxDelay(10,10,10,10,10);
display.setMuxDelay(100,100,100,100,100);
```
Unfortunately this doesn't seem to have had any effect except, on the last (and perhaps some of the earlier), to cause it to hang on the first frame a long while. The resultant image, when it appeared, is the same though. Can you give me some advice as to some better settings? I'm not sure what numbers would be good, or valid, to test with.

alexberry, 2020-04-19:
So looking at your code https://github.com/2dom/PxMatrix/blob/master/PxMatrix.h#L203 It looks like only the first 4 ints mean anything in my scenario, as I'm using A-D channels and not E, is that about right?

xsrf, 2020-04-19:
The video of the pattern_test actually showed that there might be an issue with row selection.
I just hacked together a piece of code that will try different row selection patterns, just to make sure the right one is used and works as expected. Can you run that and post a video?
You only need to change the pins if yours are different.

If you get 8 pixels going down line by line, it is BINARY.
If you get 4 pixels going down line by line, it is STRAIGHT.
If you get 2 pixels going down line by line, you may have a Shift Register for row selection.

On your panel I'd expect something like this:
https://www.youtube.com/watch?v=gMbrnnBO890

<details><summary>Code</summary>

```
#include <Arduino.h>
#include <SPI.h>

#define PIN_NC 255

// Matrix Dimensions
#define MATRIX_WIDTH  64
#define MATRIX_HEIGHT 32

// PIN Wiring, set to PIN_NC if not connected
#define P_LAT  16 // D0
#define P_A     5 // D1
#define P_B     4 // D2
#define P_C    15 // D8
#define P_D    12 // D6 (Only for 1/16 or 1/32 panels)
#define P_E     0 // D3 (Only for 1/32 panles)
#define P_OE    2 // D4 (Documentation only, this Pin is hardwired and cannot be changed!)
#define P_CLK  14 // D5 (Documentation only, this Pin is hardwired and cannot be changed!)
#define P_DATA 13 // D7 (Documentation only, this Pin is hardwired and cannot be changed!)

enum mux_patterns {BINARY, STRAIGHT, SHIFTREG_ABC};

void setup() {
  Serial.begin(115200);
  Serial.println("HUB75 Matrix Debug");
  // Init SPI
  SPI.begin();
  SPI.setHwCs(false);
  SPI.setFrequency(1e6);
  // Enable Outputs
  pinMode(P_OE, OUTPUT);
  digitalWrite(P_OE, HIGH);
  pinMode(P_LAT, OUTPUT);
  if(P_A != PIN_NC) pinMode(P_A, OUTPUT);
  if(P_B != PIN_NC) pinMode(P_B, OUTPUT);
  if(P_C != PIN_NC) pinMode(P_C, OUTPUT);
  if(P_D != PIN_NC) pinMode(P_D, OUTPUT);
  if(P_E != PIN_NC) pinMode(P_E, OUTPUT);
}

void selectRow(uint8_t row, uint8_t mux_pattern) {
  if(mux_pattern==BINARY) {
    if(P_A != PIN_NC) digitalWrite(P_A, (row>>0)&1 );
    if(P_B != PIN_NC) digitalWrite(P_B, (row>>1)&1 );
    if(P_C != PIN_NC) digitalWrite(P_C, (row>>2)&1 );
    if(P_D != PIN_NC) digitalWrite(P_D, (row>>3)&1 );
    if(P_E != PIN_NC) digitalWrite(P_E, (row>>4)&1 );
  }

  if(mux_pattern==STRAIGHT) {
    if(P_A != PIN_NC) digitalWrite(P_A, (row==0) );
    if(P_B != PIN_NC) digitalWrite(P_B, (row==1) );
    if(P_C != PIN_NC) digitalWrite(P_C, (row==2) );
    if(P_D != PIN_NC) digitalWrite(P_D, (row==3) );
    if(P_E != PIN_NC) digitalWrite(P_E, (row==4) );
  }

  if(mux_pattern==SHIFTREG_ABC) {
    if((P_A == PIN_NC) || (P_B == PIN_NC) || (P_C == PIN_NC)) return;
    // A,B,C are connected to a shift register Clock, Latch, Data
    uint32_t rowmask;
    rowmask = (1<<row);
    digitalWrite(P_A,LOW);
    digitalWrite(P_B,LOW);
    digitalWrite(P_C,LOW);
    shiftOut(P_C,P_A,MSBFIRST,rowmask>>24);
    shiftOut(P_C,P_A,MSBFIRST,rowmask>>16);
    shiftOut(P_C,P_A,MSBFIRST,rowmask>>8);
    shiftOut(P_C,P_A,MSBFIRST,rowmask);
    // Latch
    digitalWrite(P_B,HIGH);
    digitalWrite(P_B,LOW);
  }
}

void strobe(uint32_t t_ms) {
  uint32_t _s;
  _s = micros();
  while ( (micros()-_s)<t_ms*1e3 ) {
    digitalWrite(P_OE, LOW);
    delayMicroseconds(10);
    digitalWrite(P_OE, HIGH);
    delayMicroseconds(100);
  }  
  yield();
}

void latch() {
  digitalWrite(P_LAT, HIGH);
  digitalWrite(P_LAT, LOW);
}

void fillBuffer(uint16_t bits_high, uint16_t bits_low) {
  bits_low -= (bits_low+bits_high)%8;
  while (bits_low >= 8) {
    SPI.write(0x00);
    bits_low -= 8;
  }  
  if(bits_high%8 > 0) {
    uint8_t b=0;
    uint8_t bhr = bits_high%8;
    while (bhr > 0) {
      b |= (1<<(bhr-1));
      bhr--;
    }    
    SPI.write(b);
    bits_high -= bits_high%8;
  }
  while (bits_high >= 8) {
    SPI.write(0xFF);
    bits_high -= 8;
  }
}

void loop() {
  uint32_t bpad = (MATRIX_WIDTH*MATRIX_HEIGHT*3)/2;

  Serial.println("Testing: BINARY");
  fillBuffer(8,bpad);
  latch();
  for(uint8_t r = 0; r < 32; r++) {
    selectRow(r, BINARY);
    strobe(100);
  }

  Serial.println("Testing: STRAIGHT");
  fillBuffer(4,bpad);
  latch();
  for(uint8_t r = 0; r < 15; r++) {
    selectRow(r%5, STRAIGHT);
    strobe(200);
  }

  Serial.println("Testing: SHIFTREG_ABC");
  fillBuffer(2,bpad);
  latch();
  for(uint8_t r = 0; r < 64; r++) {
    selectRow(r, SHIFTREG_ABC);
    strobe(100);
  }
  delay(2e3);
}
```

</summary>

alexberry, 2020-04-19:
@xsrf Thanks for the diagnostic code! Mine ended up looking like this

```c++
#include <Arduino.h>
#include <SPI.h>

#define PIN_NC 255

// Matrix Dimensions
#define MATRIX_WIDTH  64
#define MATRIX_HEIGHT 32

// PIN Wiring, set to PIN_NC if not connected
#define P_LAT 22
#define P_A 19
#define P_B 23
#define P_C 18
#define P_D 5
#define P_E 15
#define P_OE    2 // D4 (Documentation only, this Pin is hardwired and cannot be changed!)
#define P_CLK  14 // D5 (Documentation only, this Pin is hardwired and cannot be changed!)
#define P_DATA 13 // D7 (Documentation only, this Pin is hardwired and cannot be changed!)

enum mux_patterns {BINARY, STRAIGHT, SHIFTREG_ABC};

void setup() {
  Serial.begin(115200);
  Serial.println("HUB75 Matrix Debug");
  // Init SPI
  SPI.begin();
  SPI.setHwCs(false);
  SPI.setFrequency(1e6);
  // Enable Outputs
  pinMode(P_OE, OUTPUT);
  digitalWrite(P_OE, HIGH);
  pinMode(P_LAT, OUTPUT);
  if(P_A != PIN_NC) pinMode(P_A, OUTPUT);
  if(P_B != PIN_NC) pinMode(P_B, OUTPUT);
  if(P_C != PIN_NC) pinMode(P_C, OUTPUT);
  if(P_D != PIN_NC) pinMode(P_D, OUTPUT);
  if(P_E != PIN_NC) pinMode(P_E, OUTPUT);
}

void selectRow(uint8_t row, uint8_t mux_pattern) {
  if(mux_pattern==BINARY) {
    if(P_A != PIN_NC) digitalWrite(P_A, (row>>0)&1 );
    if(P_B != PIN_NC) digitalWrite(P_B, (row>>1)&1 );
    if(P_C != PIN_NC) digitalWrite(P_C, (row>>2)&1 );
    if(P_D != PIN_NC) digitalWrite(P_D, (row>>3)&1 );
    if(P_E != PIN_NC) digitalWrite(P_E, (row>>4)&1 );
  }

  if(mux_pattern==STRAIGHT) {
    if(P_A != PIN_NC) digitalWrite(P_A, (row==0) );
    if(P_B != PIN_NC) digitalWrite(P_B, (row==1) );
    if(P_C != PIN_NC) digitalWrite(P_C, (row==2) );
    if(P_D != PIN_NC) digitalWrite(P_D, (row==3) );
    if(P_E != PIN_NC) digitalWrite(P_E, (row==4) );
  }

  if(mux_pattern==SHIFTREG_ABC) {
    if((P_A == PIN_NC) || (P_B == PIN_NC) || (P_C == PIN_NC)) return;
    // A,B,C are connected to a shift register Clock, Latch, Data
    uint32_t rowmask;
    rowmask = (1<<row);
    digitalWrite(P_A,LOW);
    digitalWrite(P_B,LOW);
    digitalWrite(P_C,LOW);
    shiftOut(P_C,P_A,MSBFIRST,rowmask>>24);
    shiftOut(P_C,P_A,MSBFIRST,rowmask>>16);
    shiftOut(P_C,P_A,MSBFIRST,rowmask>>8);
    shiftOut(P_C,P_A,MSBFIRST,rowmask);
    // Latch
    digitalWrite(P_B,HIGH);
    digitalWrite(P_B,LOW);
  }
}

void strobe(uint32_t t_ms) {
  uint32_t _s;
  _s = micros();
  while ( (micros()-_s)<t_ms*1e3 ) {
    digitalWrite(P_OE, LOW);
    delayMicroseconds(10);
    digitalWrite(P_OE, HIGH);
    delayMicroseconds(100);
  }  
  yield();
}

void latch() {
  digitalWrite(P_LAT, HIGH);
  digitalWrite(P_LAT, LOW);
}

void fillBuffer(uint16_t bits_high, uint16_t bits_low) {
  bits_low -= (bits_low+bits_high)%8;
  while (bits_low >= 8) {
    SPI.write(0x00);
    bits_low -= 8;
  }  
  if(bits_high%8 > 0) {
    uint8_t b=0;
    uint8_t bhr = bits_high%8;
    while (bhr > 0) {
      b |= (1<<(bhr-1));
      bhr--;
    }    
    SPI.write(b);
    bits_high -= bits_high%8;
  }
  while (bits_high >= 8) {
    SPI.write(0xFF);
    bits_high -= 8;
  }
}

void loop() {
  uint32_t bpad = (MATRIX_WIDTH*MATRIX_HEIGHT*3)/2;

  Serial.println("Testing: BINARY");
  fillBuffer(8,bpad);
  latch();
  for(uint8_t r = 0; r < 32; r++) {
    selectRow(r, BINARY);
    strobe(100);
  }

  Serial.println("Testing: STRAIGHT");
  fillBuffer(4,bpad);
  latch();
  for(uint8_t r = 0; r < 15; r++) {
    selectRow(r%5, STRAIGHT);
    strobe(200);
  }

  Serial.println("Testing: SHIFTREG_ABC");
  fillBuffer(2,bpad);
  latch();
  for(uint8_t r = 0; r < 64; r++) {
    selectRow(r, SHIFTREG_ABC);
    strobe(100);
  }
  delay(2e3);
}
```
As I just took my pin assignments from my example code [here](https://github.com/alexberry/trainboard/blob/master/examples/pixeltime-32x64/pixeltime-32x64.ino#L24-L30). I recorded the result and uploaded [here](https://youtu.be/lZybAIkwPeg).

So, given your diagnostics suggest it is a shift, I have just tried setting the driver chip to shift and uploading that, but it does not seemed to have changed anything. The code I just tested, with the modified line referenced, is [here](https://github.com/alexberry/trainboard/blob/diag/examples/pixeltime-32x64/pixeltime-32x64.ino#L173).

xsrf, 2020-04-19:
Hmm, not sure why you only get 2 pixels for all three tests, you also should see 8 for the BINARY test and 4 for the STRAIGHT test ... I guess that didn't work at all. But I only tested on ESP8266. Just realized you have an ESP32 connected.

You are right, the standard is SHIFT with BINARY multiplexing and I think that is okay for your panel. I was just curious because the pattern test didn't actually select the right line on your panel.

alexberry, 2020-04-19:
I'm using an esp8266 too, mine is a nodemcu.

xsrf, 2020-04-19:
An ESP8266 has no GPIO 22 or 23 🤔 Can you show your board in detail?

alexberry, 2020-04-19:
Hm, that could explain things if that's the case D: Mind you, with the same pin definitions, this works on my 8S 16x32 boards (I just move the other ends of the cables, and exclude D, then change the display settings in code - pin layouts stay the same). Either way here's my controller board:

[image]

xsrf, 2020-04-19:
Actually, the pinout from my example sketch should work for your board:
```
#define P_LAT  16 // D0
#define P_A     5 // D1
#define P_B     4 // D2
#define P_C    15 // D8
#define P_D    12 // D6 (Only for 1/16 or 1/32 panels)
#define P_E     0 // D3 (Only for 1/32 panles)
#define P_OE    2 // D4 (Documentation only, this Pin is hardwired and cannot be changed!)
#define P_CLK  14 // D5 (Documentation only, this Pin is hardwired and cannot be changed!)
#define P_DATA 13 // D7 (Documentation only, this Pin is hardwired and cannot be changed!)
```
On your board they are labeled D0 to D8 on the right side and as shown that corresponds to the GPIO Numbers shown in light blue on your image.

P_LAT is needed, only P_E is not needed in your case.

alexberry, 2020-04-19:
Yep, you beat me to it :) I had blindly copied the pinouts from my code, but copied from within ifdef ESP32, woops! brb with updated video.

alexberry, 2020-04-19:
So, [here](https://github.com/alexberry/trainboard/blob/diag/examples/diag-32x64/diag-32x64.ino#L11-L16)'s the updated code I just pushed to it, and [here](https://youtu.be/eNunm-nlniU)'s the video, which looks rather wacky when trying to pick rows...

xsrf, 2020-04-19:
Okay, so with BINARY the order is: 0,1,**0**,3,**0**,5,6,7,8,9,**8**,11,**8**,13,14,15

```
X000
X001
X010->0000
X011
X100->0000
X101
X110
X111
```
In STRAIGHT, which should trigger 1,2,4,8 on a BINARY panel it actually triggers 1,0,4,8.
So basically there is something wrong with the row selection. 
I can't explain this with a lose connection or a swapped wire... Not sure why this happens.

alexberry, 2020-04-19:
Hm. As part of all this, and the reason for me [documenting](https://github.com/alexberry/trainboard) was to be 100% sure I got the wiring right, and given I've tried on a 16x32 I got working and multiple 32x64 (same brand) that don't, as per the video, I still think my wiring is probably right.

Nevertheless I will check again tomorrow, and try to provide some sort of wiring diagram or table to show how I've got it hooked up. In theory it's wired as per [these tables](https://github.com/alexberry/trainboard#wiring-guide).

xsrf, 2020-04-20:
I also think the wiring is right, because if PIN_B wasn't properly, selection of line 3 would result in line 1 being selected etc..

Do you have the ability to power your panel from ~4V instead of 5V and test again?

alexberry, 2020-04-20:
I don't have a 4v psu, nor a way to divide the voltage at the present time, so would have to purchase something to accomplish this - may I ask why you want it tested at 4V when they're rated at 5V? Are you worried my psu is iffy? I can test it is 5V, and have other psus at that voltage to use instead.

xsrf, 2020-04-20:
The panels use 5V logic and are designed to work with 5V logic levels for HIGH signals. The ESP only outputs its Vcc (3.3V) as logic HIGH.
A typical 5V logic IC specifies the logic level to be 0.7*Vcc to be registered as HIGH. If it is below, the level is unspecified and the logic may behave unpredictable.
For 5V Vcc on the logic this level is 3.5V. So by driving the panel with 3.3V you are constantly in the unpredictable range.

To get into the specified range, you can either rise the Voltage of the ESP (3.5V should be fine) or lower the Voltage of the panel logic, which usually is easier / safer.
For 4.5V Vcc on the logiv the required level is only 3.15V

A lower Vcc on the panel would help to rule out this issue as the cause for the strange behavior. A proper solution would require a TTL compatible buffer IC as level converter.

If you have other 5V PSUs, find the one with the lowest voltage ;)

alexberry, 2020-04-20:
I have tested a psu I had that some amount of adjustability, but only went down to 4.95v and didn't help - I'm going to get an 8 channel [logic level converter](https://www.ebay.co.uk/itm/TXS0108E-8-Channel-8-Bit-Logic-Level-Bi-directional-Converter-Module-High-Speed/401266874380?hash=item5d6d5e980c:g:Gc0AAOSwmHdbO4~V) and see if that helps, as it amounts to the same thing, as far as I understand?

xsrf, 2020-04-20:
sure, using one of these is always a good idea and will definitively rule out issues caused by weak or to low signal levels. But that's really just a wild guess...

Just to make sure, you could also run my sketch with just all A-D set to GND. Just use this loop instead of the one from the code above.

```
void loop() {
  uint32_t bpad = (MATRIX_WIDTH*MATRIX_HEIGHT*3)/2;
  fillBuffer(8,bpad);
  latch();
  selectRow(0, BINARY);
  strobe(100);
  delay(100);
}
```

This should flash just the first row on the panel.

Then you should disconnect PIN B from the ESP and instead connect PIN B of the Panel to 5V manually. 
If that shifts the flashing row to the third line, the 3.3V logic actually is the cause and the level-shifter will solve that.
If that doesn't change anything (still first line flashing), the level-shifter won't help either...

alexberry, 2020-04-20:
You want me to do this before the level shifter?

Not really sure I understand fully - I'm very much beginning in hobbyist electronics. 

So you want me to ground channels A-D, ie from panel to GND on esp, testing above loop when connected like that, and then unplug pin b from GND and connect it... Where? To a 5V positive feed?

xsrf, 2020-04-20:
Basically yes, you got it. You can connect A, B, C and D literally to a PIN on the ESP that says GND, but you can also leave them on the Dx PINs of the ESP because `selectRow(0, BINARY);` in the loop will make sure that these Dx PINs actually are always LOW, thus equivalent to GND.

Row 0 (first row) should flash.

You then unplug the cable that is connected to B from the ESP/GND and connect it to +5V of the panel. If you do have a through-hole resistor (everything from 1k to 10k is okay) you may connect it in series by pushing it into the jumper cable and just use the other end of the resistor to poke Vcc pads on the power connector. Using a resistor will prevent magic smoke if you connect something wrong ;)

Row 2 (third row) should flash while you connect B to Vcc.

This can be done now instead of waiting for the level-shifter.

If you don't understand why you should do this, it may also help to watch some YT videos to learn how matrix displays actually work. Or read https://www.sparkfun.com/news/2650 :)
You should get comfortable with doing this.
Basically, A-D are binary inputs of a 4 bit decimal decoder, decoding the 4 bit input to one of 16 output lines.

alexberry, 2020-04-20:
Alright, I'll take a read through that document and give your diagnostic a go. I'll try to figure out why your test will do what you suggest it will from the documentation / your code first.

Thanks

mrweaver, 2020-06-02:
Hi Alex, I've just got a couple of displays that look very similar to yours and am having essentially the same results for the pixeltime and pattern_test examples seen in the videos you posted. I haven't read through this post in extensive detail yet, but am keen to see this issue solved.

Did you happen to resolve this issue? 

Cheers,
Michael.

alexberry, 2020-06-02:
Hi @mrweaver - I haven't managed to resolve it yet. I have recently purchased a logic level converter to test but hit an issue where the esp8266 would not boot when one of the GPIO pins was pulled up.

I've been meaning to move to a different pin and test again, I may try to do that this evening - if I get any further once using a logic level converter I shall post back here.

alexberry, 2020-06-02:
Unfortunately a llc did not work - I tested the llc with a 16x32 and got it working, then with this 32x64 and all I got was noise - I moved back to 3v logic and got back the same issue. I think perhaps the grounding was lacking, as I had to somewhat bodge it to provide 5v to the llc, but I got it working _enough_ on the 16x32 to suggest it wasn't this.

mrweaver, 2020-06-03:
Hey @alexberry,
Thanks for getting back to me. I had a muck around today with the issue, trying many of the stuff mentioned here in this thread.
Basically, in conducting @xsrf's script test from [above](https://github.com/2dom/PxMatrix/issues/188#issuecomment-616216942). I ran into the same issue as you, with the incrementing of rows being non-sequential. Specifically, I mapped the output in a spreadsheet to see if there was a pattern (nothing really stood out though).

[image]

Here, you can see that at rows 2, 4, 10 and 12, the expected rows are incorrect and identical to what @xsrf [pointed out](https://github.com/2dom/PxMatrix/issues/188#issuecomment-616233924).

At this point I chucked it all on at 3.3V supply and ran @xsrf's [simplified loop code](https://github.com/2dom/PxMatrix/issues/188#issuecomment-616750903) whilst grounding the channels A, B, C and D and incrementally cycling through them. Through a combination of trial and error, luck and caffeine, I realised that rows 2, 4 , 10 and 12 were occasionally triggered, but it was state dependant somehow. When I triggered A+B and then grounded A, B would remain on. I could not simply trigger B independently and have row 2 activate.

Anyway, as a test to mimic this, I modified the iterating test line loop so that it would select the row ahead, and then the desired row in these four cases.

```c++
void loop()
{
  uint32_t bpad = (MATRIX_WIDTH * MATRIX_HEIGHT * 3) / 2;
  Serial.println("Testing: BINARY");
  fillBuffer(8, bpad);
  latch();
  for (uint8_t r = 0; r < 32; r++)
  {
    if (((r & 0b111) == 0b010) | ((r & 0b111) == 0b100))  // select cases 2, 4, 10 & 12
      selectRow(r + 1, BINARY);                       // select the next row temporarily
    selectRow(r, BINARY);
    strobe(100);
  }
  delay(500);
}
```
This process was actually successful in causing the rows to sequentially increment, as desired. I haven't done any further testing for now. I hope this can be used as a starting point though.

I can't explain what is going on here, maybe some who has more than a day's experience with RGB matrices can help. @alexberry let me know if this code has any effect in your situation.

By the way my boards are as follows:
Supplier: [AliExpress LYSONLED Store](https://www.aliexpress.com/item/32636193207.html)
RGB 1/16 Scan P5 LED Module 320x160mm 64x32 Pixels

Cheers,
Michael

alexberry, 2020-06-03:
Hm, this is very interesting. At the moment (and from last night) I was doing all this without a breadboard, and so reproducing your code properly without would likely give me unreliable results. I intend to go to my flat tomorrow now lockdown is easing, I will grab a breadboard / take things with me to test then and will feed back.

I know this is quite a stale issue but it would be good to hear @xsrf view on this as he was instrumental in narrowing things down in the first place.

mrweaver, 2020-06-03:
Well, I did the initial experiments using a breadboard when I was messing about with the grounding etc., But the modified code doesn't require a breadboard to run, just wire it in the normal (as described in the repo).  I'm keen to hear how you go mate.

mrweaver, 2020-06-03:
So, I modified the `mux_pattern == BINARY` condition block of the function `PxMATRIX::set_mux()` in `PxMatrix.h` in the following way and it seems to have fixed the issues that I was having, I can run the `pixeltime` example fine now.
```c++
void PxMATRIX::set_mux(uint8_t value)
{

  if (_mux_pattern == BINARY)
  {
    if (value & 0x01)
      digitalWrite(_A_PIN, HIGH);
    else
      digitalWrite(_A_PIN, LOW);
    if (_mux_delay_A)
      delayMicroseconds(_mux_delay_A);

    if (value & 0x02)
    {
      bool lastA = digitalRead(_A_PIN);
      if (!lastA)
        digitalWrite(_A_PIN, HIGH); // set A prior to B
      digitalWrite(_B_PIN, HIGH);
      if (!lastA)
        digitalWrite(_A_PIN, lastA); // revert A to prev state
    }
    else
      digitalWrite(_B_PIN, LOW);
    if (_mux_delay_B)
      delayMicroseconds(_mux_delay_B);

    if (_row_pattern >= 8)
    {

      if (value & 0x04)
      {
        bool lastA = digitalRead(_A_PIN);
        if (!lastA)
          digitalWrite(_A_PIN, HIGH); // set A prior to C
        digitalWrite(_C_PIN, HIGH);
        if (!lastA)
          digitalWrite(_A_PIN, lastA); // revert A
      }
      else
        digitalWrite(_C_PIN, LOW);
      if (_mux_delay_C)
        delayMicroseconds(_mux_delay_C);
    }

    if (_row_pattern >= 16)
    {
      if (value & 0x08)
        digitalWrite(_D_PIN, HIGH);
      else
        digitalWrite(_D_PIN, LOW);
      if (_mux_delay_D)
        delayMicroseconds(_mux_delay_D);
    }

    if (_row_pattern >= 32)
    {
      if (value & 0x10)
        digitalWrite(_E_PIN, HIGH);
      else
        digitalWrite(_E_PIN, LOW);
      if (_mux_delay_E)
        delayMicroseconds(_mux_delay_E);
    }
  }

 // if (_mux_pattern == STRAIGHT) etc. below

```
 I know this is just a dodgy workaround and I have not actually fixed the root of the issue. It seems this is probably a timing issue or some such thing, that I'm not configuring correctly. @alexberry does this also work for you?
Any further input still welcome.

xsrf, 2020-06-03:
@mrweaver thx for your input on this! I'm curious what may cause this issue though... Is it possible for you to actually reverse engineer the schematic of the A-E inputs using a multimeter?
I only have one kind of panel and all I learned about different panels is from the issues here within the last few weeks, so for the moment I'm not able to come up with an explanation of why the input logic actually is that way... Usually the input is just stateless logic (decoders or and/(x)or gates) and it shouldn't matter in what sequence they are triggered...

xsrf, 2020-06-03:
btw. have you tried if it works if you just scan the rows backwards? :)

mrweaver, 2020-06-04:
@xsrf Thanks for getting back to us. I would not have expected to see any state dependant logic in there either, sure it defeats the purpose to some extent! I'll try and reverse engineer it a bit this arvo when I finish up with work. 

I managed to get a little time this morning to test your reverse order idea, and it works correctly :-). That would be a neater solution for the library that does add additional overhead - although it still doesn't help explain anything. I modified the `PxMATRIX::display(uint16_t)` function to reverse the row order and it solved my issues on `pixeltime`, but not the `Aurora_Demo` - I guess because I that for loop wasn't intended to be run backwards!. I have include the relevant code excerpt below anyway:

```c++
#if REVERSE_ROWS == 1
  for (uint8_t i = (_row_pattern - 1); i < _row_pattern ; i--)
#else
  for (uint8_t i = 0; i < _row_pattern; i++)
#endif
  {
    if (_driver_chip == SHIFT)
    {
      if ((_fast_update) && (_brightness == 255))
      {
     // etc...
```

I've never used these displays before, so on a side note, when running complex scenes such as the `Aurora_Demo` is it typical for the display to flicker? Otherwise, perhaps my hacky changes are causing the display to clear/blackout between frames.

I'll report back with any findings from my probing.

mrweaver, 2020-06-04:
Actually, it seems that the flickering also occurs without my modifications. It is only a noticeable issue when a large number of pixels are changing. Is it because the OE is toggled during register loading?

mrweaver, 2020-06-04:
Did a couple of continuity tests between the inputs A..D and the board. They all run to a CHIPONE 74HC245 chip with pins corresponding below:
```
// Input -> 74HC245 Pin
A -> 6
B -> 7 
C -> 8
D -> 9
```
These then run to a poorly documented Chinese chip, DP7268D. Where A, B & C route to inputs A0, A1 & A2, respectively. E is routed to the EN pins, I assume that given there are two chips, and two ENs one is active low, the other active high and, when used in this manner, it essentially doubles the output capacity.

I started drawing up a schematic, but saw that it is essentially the same as [this EasyEDA schematic](https://easyeda.com/editor#id=61b9e05f47824794833f217a96a15bc0) - at least for the relevant part and despite slightly different pin order routings.

This is still stateless logic so I can't explain it yet, unless these chips are dodgy in someway - and who would be surprised to see that?

Cheers,
Michael.

xsrf, 2020-06-04:
Can you check if the ENL/ENH of the DP7268D that is NOT connected to the 74HC245 is properly tied to its EN signal? ENH of the DP7268D for rows 0-7 must be connected to 5V, ENL of the DP7268D for rows 8-15 must be connected to GND.

[20181023173941260.pdf](https://github.com/2dom/PxMatrix/files/4730779/20181023173941260.pdf)

Regarding flicker: `aurora` uses fast_update which I assume will not work properly in reverse row order because it sends the data for the next (+1) row. This must also be adapted.
https://github.com/2dom/PxMatrix/blob/286135f5758dea24c198fda38b4323bc9c65a228/PxMatrix.h#L1126
`pixeltime` does not use fast_update.
Also, the library should never toggle OE while data is loaded into the shift registers.
With fast_update, OE is enabled (LOW) before the data is sent to the shift registers and is disabled (HIGH) afterwards. Without fast_update, OE is toggled after the data was sent.

Are you using an ESP8266 or ESP32?

mrweaver, 2020-06-05:
There are four DP7268D chips on each panel, T1, T2, T3 & T4. I checked all the voltages on the following pins of each of those chips with a scope.  I also checked the VDD and GND pins for completeness - they were all fine.
 Chip | `ENL` [Pin 10] | `ENH` [Pin 7]
---|---|---
T1 | D | H
T2 | H | D
T3 | D | H
T4 | H | D

I guess the next thing to do, observe the behaviour of these chips when running the row select script that you provided above. @xsrf what are your thoughts? At some point in the chain, something unexpected has to occur.

Sorry, I wasn't very clear above, the `aurora` example:
1. displays incorrectly (due to the original issue) and flickers when using the original code.
2. displays correctly and flickers when I implement the [Enable-A](https://github.com/2dom/PxMatrix/issues/188#issuecomment-638139428) modification. 
3. displays incorrectly and flickers when running the [Row-Reversal](https://github.com/2dom/PxMatrix/issues/188#issuecomment-638528646) modification. As you point out, it is clear to see why! :) 
Thanks for clarifying the OE stuff.

I am using an ESP8266, NodeMCU Devkit V1.0. Do you know if there are any performance differences in driving these displays with ESP8266 vs ESP32?

mrweaver, 2020-06-08:
So, I had an ESP32 lying around and I connected it up for a test and it worked. 

Initially, It wasn't really playing nice for me, probably due to LoRa and OLED peripherals on that board, so I changed the pinout mapping and it seems to work (this was after making a level converter board with 2N7000s and experiencing the same results).
I didn't try remapping the ESP8266, yet but if I ever bother doing so, I'll report back here. @alexberry might be worth giving that a crack.

The ESP32 I used was a TTGO v2.0 LoRa board, this Platformio configuration file excerpt describes the mapping and a couple of other things.

<details>
<summary>ESP32 Pin Map</summary>

```
[env:heltec_wifi_lora_32]
platform = espressif32
board = heltec_wifi_lora_32
upload_protocol = esptool
board_build.f_cpu = 240000000L ; change MCU frequency
build_flags =
    -D P_NC=255  ; Not connected
    -D P_A=32    ; 
    -D P_B=33    ; 
    -D P_C=25    ; 
    -D P_D=27    ;  (Only for 1/16 or 1/32 panels)
    -D P_E=P_NC  ;  (Only for 1/32 panels)
    -D P_LAT=22  ; 
    -D P_CLK=14  ; 
    -D P_OE=15   ;  
    -D P_DATA=13 ; 
```
</details>

Thanks for your help along the way. Stay safe guys.

2dom, 2020-06-12:
Thanks ... and you!

closed 2020-06-12 by 2dom