2dom/PxMatrix #194: OE pin should have a pull up resistor

awyork, 2020-05-05:
To protect the panel during programming there should be a 5k or 10k pull up resistor between OE and 3V3.  This will disable the LEDs during programming.  

I've burnt out a few green pixels because the display was stuck on.  I was displaying all white with display(90).

xsrf, 2020-05-05:
Decent panels do have input protection in form of a series capacitor and an input pullup (RC-circuit) that will block any DC / too long LOW pulses on OE. A pullup only will not help if, for some reason, the CPU crashes while OE is driven LOW.

However, this is not something this library can solve... But a warning in the readme may be nice.

2dom, 2020-05-06:
Not sure the pixels supposed to be die if we pull OE to low permanently. I never came across this issue and I have misused all kinds of matrices over the years. This might also be due to some bad panel designs. Will keep this in mind - if it becomes an more common issue we might add something to Readme as @xsrf suggested.

closed 2020-05-06 by 2dom

xsrf, 2020-05-06:
I don't have your experience with multiple panels (I only have one kind of RGB HUB75 and one monochrome one with 5mm LEDs) but keep in mind that typically all panels may be designed to drive the LEDs above their continuous current rating due to the fact that they are never on for 100%.
On a 1/16 panel the maximum duty cycle for an LED cannot exceed 1/16 due to the row scanning. Thus, to achieve a even modest brightness for the panel, the LEDs must be driven with a higher current.
I'd assume almost all panels do this. 
And if they do, keeping LEDs on for to long will damage them.
However, as I've said, I hope most panels do actually have input protection do protect against this kind of fault.